sinara-hw/sinara #410: Sayma RTM schematics cosmetics

hartytp, 2017-12-04:
Very low priority issue, but at some point it would be nice to have a cleaner version of the Sayma RTM schematics:

- [ ] Lots of text is overlapping and hard to read
- [ ] Huge annotation saying "FIducials missing" on first page
- [ ] Once we DNP the ADCs, can they be removed from the schematic PDF posted to the release (removes a few pages and makes it easier to read).

marmeladapk, 2019-01-14:
- [ ] On RTM_IPMI sheet there's overlapping text on I2C port
- [ ] Si5324 sheet cdr_clk_clean3, rst  port has overlapping text, 
- [ ] date out of bounds on most sheets
- [ ] Sheet numbering is weird
- [ ] "Matching between (...)" text goes beyond sheet

Otherwise all other text is readable and not obstructed by anything.

hartytp, 2019-03-06:
I think this is no longer relevant now we've moved to AD, so closing

closed 2019-03-06 by hartytp