sinara-hw/sinara #411: BaseMod remove power detector in next revision?

jbqubit, 2017-12-04:
Moving from https://github.com/m-labs/sinara/issues/269. 

@hartytp asks
> Does anyone object to removing the power detector from BaseMod v1.1? cf #397

@dhslichter responds
> The point of the power detector is to noise-eat at the output of the DACs. What is it costing us?

@hartytp responds

Yes, but only when the switch is off, and only via the 24-bit ADC.

Main cost is space: as @sbouhabib pointed out elsewhere, this is the largest component on BaseMod and fitting it on was actually quite challenging. If we remove it then we can make the layout a bit less tight.

Not a huge deal, but it's not totally cost free so unless you really plan to use it then we should scrap it.
-----------------

My comments.... 

Agreed that space occupied by the power detector design is huge. Especially because it involves impedance controlled traces. Design complexity due to the inclusion of the detector can only be described as very high. 

[image]

It also consists of 27 BOM items. AD8363 is about $7/ea and HMC349 switch is about $5. Component cost is not high. 

> I don't think it's actually that useful because:
> It is read out by the 24-bit ADC which is very slow

The original intent to compensate for long-term drift. Not for fast feedback. As you point out the power can only be monitored when the RF output to the physics target is off (using HMC349 switch). This was never intended for fast feedback.

Agreed that this is something that probably should have been culled from the original design.

hartytp, 2017-12-04:
So, are we happy removing it then? Is that a decision or a discussion?

jbqubit, 2017-12-04:
@hartytp No. This is not a decision. This is moving the discussion to a separate thread for discussion.

hartytp, 2017-12-04:
optimistic ;)

jbqubit, 2017-12-04:
Let's add to errata list #269 only once a decision is made.

hartytp, 2017-12-04:
ack

dhslichter, 2017-12-04:
I don't love it enough to fight for it.

hartytp, 2017-12-04:
> I don't love it enough to fight for it.

IME, that's generally a good indication that it's not useful enough to be worth keeping. I'll keep the pistols ready in case you change your mind though ;)

dhslichter, 2017-12-04:
I think my rationale is that probably I will be rolling my own special-purpose AFEs for the applications we want to be ultra-precise on and so we can always go back to this if we want to use it for design ideas later.

hartytp, 2017-12-04:
Sounds good. Let's get a stripped down basic AFE that works really nicely. People can then add their own pet features to it as they desire.

dtcallcock, 2017-12-04:
The two large red areas @jbqubit highlights on the left of the top image are the rf switches.  These would presumably be retained even if the detector was removed.

I like the power detector because I spend a lot of time in the lab plugging in the rf power meter to debug DDSs or calibrate out changes in output power vs frequency.  The latter is probably not that important for driving AOMs but will make the board much nicer for broadband applications.

hartytp, 2017-12-04:
But, won't you still end up doing that even with the power detector, since you can only calibrate up to the switch? The gain flatness of this board should be really quite good, so the amplifiers/filters you put after it will ultimately limit the gain flatness as much as the stuff up to the switch.

hartytp, 2017-12-04:
Anyway, you have an analog input, so your best bet AFAICT is to just add a small power detector + pickoff near your load. Then everything is calibrated. Plus, if you do it that way you get to use the nice 5MSPS ADC on the mezzanine so you can do it in-loop...

dtcallcock, 2017-12-04:
> But, won't you still end up doing that even with the power detector, since you can only calibrate up to the switch? 

I think having signal sources that output the programmed power regardless of frequency is generally considered a useful thing.

> The gain flatness of this board should be really quite good.

I imagine the DAC output power is the main issue, especially as you start climbing up towards f_clk/2.

> Anyway, you have an analog input, so your best bet AFAICT is to just add a small power detector + pickoff near your load. 

Given this is an option I don't think it's a big real to remove it.  I just like the option of doing this without adding over a dozen little minicircuits boxes and cables to the Sayma card...

hartytp, 2017-12-04:
Well, I think we've heard all possible arguments from each side. Time for a decision. How do we make those  in this project again (do I need to get my CAPSLOCK key out after all)?

hartytp, 2017-12-04:
I've added the "discussion" label to all Sayma and BaseMod issues that might be contentious. For most of these, the various possibilities have been mapped out, as well as the various arguments for and against each. So, now we just need an executive decision to be made. @jbqubit I assume that you will do that? If so, when you get a chance can you review them, make a decision, summarize the action points at the top of the issue, and remove the discussion label? Feel free to delegate some of this work if you'd rather.

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/BaseMod#2

closed 2018-08-18 by gkasprow