sinara-hw/sinara #413: DNP Sayma RTM JSD204 ADC

hartytp, 2017-12-06:
- [ ] DNP JESD204 ADC on Sayma RTM, as well as all associated components (regulators etc). Do not remove from schematic/PCB, so that users can populate them if they desire.
- [ ] If easily possible, remove the JSESD ADC pages from the schematic PDF uploaded to the release (makes it shorter and easier to read).

Reasons for this decision discussed at length elsewhere.

gkasprow, 2018-11-28:
Let's get rid of ADCs. We don't have them characterised. Guys at Creotech did some code to do basic tests but have some issues with synchronisation. So if there are some bugs, they will propagate. If someone wants to have 125MS/s ADCs, can use low cost AFC or AFCK board with FMC ADC boards. There are several open source and proprietary ones available. Let's use space to attach better cooling for DACs. 
If someone wants fast ADCs, we will have RFSOC solution in compatible form factor.

gkasprow, 2018-11-28:
We can also route Rx JESD204B lanes to the analogue mezzanines if someone wants to play with fast ADCs on them.

hartytp, 2018-11-29:
> Let's get rid of ADCs. We don't have them characterised. Guys at Creotech did some code to do basic tests but have some issues with synchronisation. So if there are some bugs, they will propagate. [...]

Yes, let's do that.

> We can also route Rx JESD204B lanes to the analogue mezzanines if someone wants to play with fast ADCs on them.

I'm not bothered about this. I wouldn't bother with this. I don't see anyone putting a fast JESD ADC on the AFE mezzanines. Routing multi-gigabit lanes from the AMC FPGA to the AFE mezzanine with acceptable SI seems like a lot of work.

gkasprow, 2018-12-30:
I removed the ADCs. Optionally one can install such ADC on mezzanine, together with clock distribution IC. JESD204 lanes are shared with LVDS ones and can be routed by changing capacitor assignment.

closed 2018-12-30 by gkasprow