sinara-hw/sinara #414: Sayma AMC to RTM LVDS links

hartytp, 2017-12-06:
Add extra LVDS links between Sayma AMC and RTM. These can be used, e.g. for to provide user IO for LVDS ADCs on the AFE mezzanines.

On Sayma AMC:
- [ ] LVDS0: Route IO_L22P_T3U_N6_DBC_AD0P_66/IO_L22N_T3U_N6_DBC_AD0P_66 to unused pin on AMC<->RTM connector
- [ ] LVDS1: Route IO_L19P_T3L_N0_DBC_AD9P_66/IO_L19N_T3L_N1_DBC_AD9N_66 to unused pin on AMC<->RTM connector
- [ ] LVDS2: Route IO_L12P_T1U_N10_GC_66/IO_L12N_T1U_N10_GC_66 to unused pin on AMC<->RTM connector
- [ ] LVDS3: repurpose FAT_PIPE2_0 (IO_L18P_T2U_N10_AD2P_66/IO_L18N_T2U_N10_AD2P_66) and route to unused pin on AMC<->RTM connector
- [ ] LVDS4: repurpose FAT_PIPE2_1 (IO_L17P_T2U_N8_AD10P_66/IO_L17N_T2U_N8_AD10P_66) and route to unused pin on AMC<->RTM connector
- [ ] LVDS5: repurpose FAT_PIPE2_2 (IO_L16P_T2U_N6_QBC_AD3P_66/IO_L16N_T2U_N7_QBC_AD3N_66) and route to unused pin on AMC<->RTM connector
- [ ] LVDS6: repurpose FAT_PIPE2_3 (9. IO_L15P_T2L_N4_AD11P_66/IO_L15N_T2L_N5_AD11N_66) and route to unused pin on AMC<->RTM connector
- [ ] LVDS7:  repurpose SYNC_OUT11 (IO_L13P_T2L_N0_GC_QBC_66/IO_L13N_T2L_N0_GC_QBC_66) currently NC on Sayma RTM side of AMC<->RTM connector
- [ ] LVDS8:  repurpose SYNC_OUT11 (IO_L10P_T1U_N6_QBC_AD4P_66/IO_L10N_T1U_N7_QBC_AD4N_66) currently NC on Sayma RTM side of AMC<->RTM connector
- [ ] LVDS9: repurpose ADC0_SYNC (solder jumper on Sayma RTM)
- [ ] LVDS10: repurpose ADC1_SYNC (solder jumper on Sayma RTM)
- [ ] LVDS11: repurpose FPGA_ADC_SYSREF (solder jumper on Sayma RTM)

Comments:
- that uses 9 of the 10 currently unused pairs on the RTM<->AMC connector.
- We leave the ADC JESD serial lines in place, since they are multi-gigabit and hence we can't easily use solder jumpers. the ADC_SYNC lines are only latched from a 125MHz clock so SI is less crucial and solder jumpers can easily be used
- FAT_PIPE2 is the link to the second MCH slot in the uTCA rack. This will not be used in ARTIQ.

hartytp, 2017-12-06:
If anyone has any comments/objections to the above then please make them known!

Otherwise, the main remaining question is what to do with all that IO on Sayma RTM? The majority of it should route directly to the AFE mezzanines. Some could route to the RTM FPGA if we want to add in a bit of extra flexibility (although, what we can do here is very limited by available pins on the RTM FPGA).

The above proposal gives us enough for 6 LVDS lines to each AFE if we want.

**My preference would be to route 5 LVDS lines to each AFE mezzanine, and 2 LVDS lines to the FPGA.** e.g. we can then use the FPGA as a flexible fanout to route shared CNV and SCLK to the two AFE mezzanines. We then, in principle, have enough to run a full 5MSPS ADC on each AFE which is nice.

gkasprow, 2018-12-30:
I added fixed 6 LVDS lines to each mezzanine + 3 LVDS lines to RTM FPGA.
Plus another 6 shared LVDS/MGT lines to each mezzanine.

closed 2018-12-30 by gkasprow