sinara-hw/sinara #415: Sayma RTM: size of AFE mezzanines

hartytp, 2017-12-07:
- [ ] Switch from having 4 AFE mezzanines on Sayma, each carrying 2 DAC channels, to having 2 mezzanines, each with 4 DAC channels.

Context:
- Having 4 AFE mezzanines on Sayma provides a more fine-grained control of the AFE than any user actually requires.
- Reducing the number of mezzanines from 4 to 2 frees up quite a bit of space. Both because it reduces the amount of dead-space around the edges of mezzanines and between adjacent mezzanines, and because it allows more sharing of components like regulators etc. This makes the mezzanines less cramped (they are *very* dense atm) and hence easier for users to modify/rework by hand, and should help to resolve some of the mechanical issues we've seen fitting the mezzanines in. It also gives us more ability to add extra functionality to the mezzanines later on.
- Reducing the number of mezzanines allows us to put quad LVDS ADCs with very compact digital interfaces on the mezzanines. We probably don't have enough AMC<->RTM digital IO to do this with double ADCs without some rather nasty hacking. The same argument applies more generally to other components one might want to put on the AFE mezzanines: since there aren't enough free pins on the RTM IO to use it as a flexible router between the AMC FPGA and AFEs, we have to implement a fixed routing between the AMC FPGA and the RTM AFE mezzanines. The more mezzanines there are, the less IO they each get, and the less useful the IO becomes.
- Reducing the number of mezzanines reduces the overall cost of ownership of a working Sayma.

As an alternative, we could just make BaseMod and MixMod double-width mezzanines, occupying two AFE slots each. This is probably a worse option because:
- If we switch to having a large cout-out for each AFE mezzanine then double-width mezzanines would need to have two front panels attached. This is mechanically worse and more expensive than having a single larger front panel
- More expensive: the RF connectors are a significant cost. More connectors is more expensive than few connectors with more pins!
- Having a mezzanine with 4 connectors (2 digital and 2 RF) is mechanically worse than having a mezzanine with 2 connectors, each with more pins
- Having more DAC channels per mezzanine allows us to use connectors with fewer pins, which is mechanically better and cheaper. e.g. we only need a single I2C bus per AFE, and we only need a single LO line per AFE.
- It makes the gateware easier to write/maintain if we know that all mezzanines have the same number of DAC channels. Supporting both single-and double-width mezzanines will get a bit messy.

Risks/work:
- Needs a bit of work to get right. However, since we need to more-or-less completely redo the AFE mezzanine mechanics anyway (new RF connectors, adjust spacings, new front panels, etc etc) it's not clear how much additional work this would involve. Having fewer mezzanines will make some things easier AFAICT.

dhslichter, 2017-12-07:
I am totally fine with two mezzanines.  The main reason for the 4-mezzanine design was to allow more granularity for inputs and outputs so that we wouldn't be wasting money by having many channels go unused on a given board.  However, it seems that breaking it up as 4 channels per mezzanine is fine, especially since now that we are letting the ADCs be a property of the mezzanines themselves, and the ADCs are slow, there are fewer possible configurations.

dhslichter, 2018-03-09:
Make AFEs longer to use space vacated by removing SATA connectors and clock mezzanine. 
 See https://github.com/m-labs/sinara/issues/386#issuecomment-371579489