sinara-hw/sinara #416: stress from MNC/SMA (need nuts)

a-shafir, 2017-12-07:
There is some problem with the BNC/SMA connectors mounted to the panels:

1. PCB 3U BNC rev 1.1
Certainly need nuts for the BNC. The panel LEDs adding some constrains. But at least the top (io2,3,6,7) BNC nuts can be placed that will improve a lot. Also need few thin washers from the connectors side to prevent stressful bending of the PCB.

2. 3U DDS rev 1.0
Needs metal/plastic hard washers from inside or adjust the pcb mounting. In any case currently the panel nuts can't be secured without stressing the pcb.

3. 3U SMA rev 1.0
Currently no nuts. It is quite constrained because of the LEDs but in fact the leds can be moved further from the PCB (need "tall" LEDs). Also there is limited space for the nuts from the inner side of the panels and some difference in height between the connectors. It can be solved by moving the connectors/panel screw mounts/LEDs.

I think all the BNC/SMA panel connectors must have nuts for the lab environment and this needs some changes in the further revisions of the boards. Also the panel design affected (the nuts will cover some drawing items and/or LEDs)

jordens, 2017-12-07:
I'd also like to get the lock screws and washers for the isolated BNC connector. Does Technosystem still have them? IIRC they came with the connectors.

gkasprow, 2017-12-11:
@jordens the connectors were not equipped with them, so they need to be purchased separately.

gkasprow, 2017-12-11:
I will add them to the BOM for Urukul and DIO-BNC

jbqubit, 2017-12-12:
@gkasprow, @a-shafir please assign Issue to somebody.

gkasprow, 2017-12-12:
@jbqubit I added TODO things to corresponding issues related with Urukul/Sampler and BNC-DIO boards so you can close it.

jordens, 2017-12-13:
I would advise that you leave individual issues open until they are fixed with a commit. Transferring items into summary issues with long check lists is a bad idea because (among other reasons) there is no tracking of the changes to those summaries. But I'll let you choose.

closed 2017-12-13 by jordens