sinara-hw/sinara #417: 3U Panel design style unification

a-shafir, 2017-12-07:
There is some inconsistency in the panel designs. For instance comparing Urukul Quad 1Gs/s DDS rev 1.0 and EEM OCTAL BNC IO rev 1.1 panels the fonts are different, the capitalization ("Quad" vs "OCTAL") are different the "rev" label alignment etc is inconsistent. 

Since the modules will be installed in a one rack and any misalignment/inconsistent thins will be quite obvious.

It makes sense for the further revisions of the panels follow some standards of the design as possible.

hartytp, 2017-12-07:
Yes, this is a minor bug bear that's been around for a while.

The "rev" inconsistencies have been pointed out before and are, I think, all already noted on the minor errata issue for each 3U PCB. That should be sorted in the next revision.

I'm less fussed about the quad versus octal thing. Maybe switch to just saying `4x`, `8x` etc?

hartytp, 2017-12-07:
Anyway, probably the best way to deal with this is to go to the minor errata thread for each PCB and make a recommendation of what change you'd like to see made.

jordens, 2017-12-07:
It needs to be consistent. Just putting random capitalization, wording, and changing fonts needs to stop.
As long as all panels are in a single file, it is just fine IMHO to keep a single issue that applies to all panels.

hartytp, 2017-12-07:
@jordens Sure. But, my feeling is that this will go faster if there is a complete list of suggested changes to the panels, rather than a general "please make this consistent" issue. Based on past experience, having that kind of general issue results in multiple iterations and multiple PDFs being uploaded and reviewed; Greg's idea of consistent may not match yours!

jordens, 2017-12-07:
As long as there is a single PDF for all panels, I don't see how that would happen.
If you expect that to be split issues, then there are multiple things to be resolved in order.

1. Split the panels
2. File issues for each board that deviates from the desired style

jordens, 2017-12-07:
And if Greg doesn't agree with my idea of consistency, I'd like to know and hear his rationale.

hartytp, 2017-12-07:
> deviates from the desired style

Time to define a desired style for front panels? Add something to the style guide on the Wiki (or other suitable location) to make your expectations/wishes clear for this?

Also, tangentially related, but do you have a proposed naming convention for the DIO boards? Happy to tidy that all up if you specify what you want...

gkasprow, 2017-12-07:
All panels are multi-PCB project where each panel is a PCB file. I generate common pdf and separate dxf files for CNC machine.
It does not matter how you form issues. You can post common issue which applies for all panels and then once I close it, you have a look and if you find something else to fix, will post another, dedicated issue.

gkasprow, 2017-12-07:
Anyway, I'd like to finally cleanup all schematics, panels, silkscreens and use common naming convention. It's optimal to apply these changes at once to all designs.

jordens, 2017-12-07:
The first panels Greg made for the DIO boards (apart from what we already have in the errata) are just fine IMHO. Let that define the style. And please file an issue and assign me to the style guide for the panels if you support that path. But only if Greg agrees to implement the style guide. Same (on both fronts) for the names.

hartytp, 2017-12-07:
> Same (on both fronts) for the names.

To check though: this isn't going to reopen the can of worms with Russian names, right? The names will be something moderately close to the DIO_BNC that we currently have, right? IIRC, we agreed that for these boards a descriptive name is fine.

jordens, 2017-12-07:
Don't get to excited. My preferred names are DIO-BNC, DIO-SMA, DIO-RJ45 with dashes instead of the underscores. No EEM, no PCB, no 3U.

hartytp, 2017-12-07:
:+1: To that. Best news I've had all day.

I'll take care of renaming those DIO boards later in the week when I have time, and will assign style guide etc issues.

gkasprow, 2018-01-19:
@jordens @hartytp please check the front panels. It's quite possible I overlooked sth.

hartytp, 2018-01-19:
Which ones need checking? Did you look at @jordens guidelines on panels [here](https://github.com/m-labs/sinara/wiki/EEM)?

hartytp, 2018-01-19:
@gkasprow assuming you want nit picks (quotes are from the style-guide on the Wiki):

- Personally, I'd like a dash between the name and description (or, italicise the name or something) for improved readability (or, just put the description on a separate line to the name, just be consistent about what you do!). @jordens can feel free to overrule me on this, I don't feel strongly about it.

> Place a label with the board name (Urukul) and a minimal description (4x 1GS/s DDS) right below the upper PCB angle bracket sunk mounting screw. Center it horizontally on the panel. Use proper English capitalization: Urukul 4x 1GS/s DDS.

Currently, the capitalisation for names like "ZOTINO" is not right.

> Place labels below the item they refer to where possible

Currently, most labels are *above* the item they refer to. I do think it's important to be consistent about this, otherwise it can definitely lead to confusion! (Although, in some places it's not possible, so we'll have to live with minor inconsistencies here).

- On sampler, the second EEM light is labelled "EEM2". That doesn't make sense IMHO since the board only has "EEM0" and "EEM1" on it. I'd prefer to label is "Mode" and just document that the mode LED being on means "fast mode" with two EEMs.

- On the DIO-BNC board are the LED holes missing? Also, did we name this "DIO-BNC" or "DIO_BNC"? Just copy what you did on the PCB silk.

- Is the power connector hole missing on the VHDCI carrier?

- Do the washers/nuts around connectors interfere with the text?

- Should the LEDs on Urukul have a label? For Urukul, I'd probably scrap the boxes around the channels (the LEDs aren't necessarily associated with a channel are they?). 

- Urukul SMAs should be labelled "RF0" etc, not "CH0"

- Urukuk: if there is room, label "CLK IN" "EXT CLK IN" to match the schematic (makes life easier for users to have the two match)

- I'd just call "OCTAL IDC-BNC adapter" "IDC-BNC adapter" to match the schematics/project name.

hartytp, 2018-01-19:
Anyway, those are minor nits. Basically, they all look really nice to me! Great job @gkasprow

jordens, 2018-01-21:
Thanks @gkasprow 

* No strong opinion about the stylistic distinction between the name and the function.
* Ack and support the items @hartytp  raised.

We'll probably need to do a couple front panels with ink in milled shapes on powder coated panels (like Schaeffer/Frontpanel-Express). Has anybody already converted the DXFs or did designs from scratch for Sinara panels in HPGL or the Schaeffer tool's native format? I started and it's looking like a major pain.

klickverbot, 2018-01-21:
As for Schaeffer, [as per their website](https://www.schaeffer-ag.de/en/support/faq/#c965), they accept DXF at an extra charge. Has anyone used that service yet?

gkasprow, 2018-01-21:
Once we finally make sure that panels appearance won't change, I will order batch of ones from Schroff directly, with cutouts, anodized, EMC brackets and proper silk-screen markings. I order panels from them regularly and they accept DXF files.

gkasprow, 2018-01-22:
It seems I fixed all issues, @jordens please have a look.
I can place version number at the very top, above ARTIQ logo. It is covered by the handle and visible only when one looks at it when it's placed at the height of ones eyes.

hartytp, 2018-01-22:
@gkasprow isn't there room for it on the bottom left, just above the handle? Otherwise, at the very top is okay for me (I like to have the version number visible so I can easily spot problems that arise due to people accidentally using the wrong hardware version).

Otherwise, all looks good to me, thank you!

Not sure if it's actually worth changing, but final thoughts:
- For consistency it might be worth labelling the clock input on Kasli "CLK" rather than "Clock"
- If there is room without making things too cramped, it might be helpful to label the clock inputs on Kasli and Urukul with the maximum power ("?dBm MAX"). I didn't see this specified on the SI5324 datasheet, but it's usually something like 2Vpp single-ended, so +13dBm if we use the baluns (although, one would need to read the datasheet carefully to confirm that this is actually okay with the supply voltage we're using).

jordens, 2018-01-22:
* Either `BIT` or `Bit`. I'd prefer `Bit`. But be consistent. (Zotino vs. Sampler). (minor)
* On Urukul the LED grouping with the SMAs is pretty fixed (PLL lock/error and RF switch per channel). I'd put the boxes around them. (minor)
* I'm OK with using the dash in `DIO-BNC` if there is strong preference, but it needs to be consistent (rename the files in the directory). My rationale for the underscore was that there is an unambiguous 1:1 mapping to python module names. But we can also systematically translate dashes to underscores and hope for no collisions.
* Having the revision that low on the panel is fine with me. The handle does not obstruct that.

jordens, 2018-01-22:
* My other assumption for some of the placements (designators below the connectors e.g.) is that these devices tend to be at or above eye level and less frequently below eye level as bench top devices would be. Ergo the placement.

jordens, 2018-01-22:
* Also @hartytp please can we go for 8HP Kasli panels? Have you tried mounting either the VHDCI carrier or Kasli complete with 8 EEM ribbons 4HP left of another panel? Please do and tell me how to properly solve that problem.

hartytp, 2018-01-22:
I'm not overly fussed about this. Do whatever you think is best

jordens, 2018-01-22:
@gkasprow Can we do 8HP Kasli panels? (https://github.com/m-labs/sinara/issues/450)

gkasprow, 2018-01-22:
OK, all issues fixed, Kasli and VHDCI are 8HP.

closed 2018-01-22 by gkasprow