sinara-hw/sinara #418: Novo v2.0 termination & electrolytic capacitors

a-shafir, 2017-12-07:
1. I think it makes sense to change the termination resistors R4 to 100R. It both better matches the balanced BNC layout and increases thermally safe DC differential input range.

2. I see there are many 10uF 16V 3216 capacitors. As well as larger count of 22uF 16V 0805 ceramic parts.

Is there any specific reason not to use only the ceramic capacitors? 

I think from the parasitic resonances point of view some of them (that are currently parallel to 22uF like C49, C28 etc can be replaced in BOM to 47nF-470nF 1206. The other that are only capacitors in the net (like C118, C35...) can be replaced for 10uF 16V 1206 ceramics. In both cases the reliability and the cost will be improved.

gkasprow, 2017-12-07:
Ceramic capacitors have high Q and they cannot be used with LC circuit. That's why I add tantalium in parallel to damp oscillations. Moreover they are noisy due to piezo effect and pickup vibrations.
Another effect is that their capacitance varies strongly with voltage. With 10V 10u 0805 cap connected to 10V you get something like 2..3uF of real capacitance.
Yet another reason is that LDOs generally don't like low ESR,  high capacity ceramics at their outputs and tend to oscillate.

jordens, 2017-12-07:
I'd also consider going to 100R termination. We can't handle 10V**2/50R*8ch and I suspect that there are neither noise advantages nor drivers that do drive it.

hartytp, 2017-12-07:
> I think it makes sense to change the termination resistors R4 to 100R. It both better matches the balanced BNC layout and increases thermally safe DC differential input range.

You are welcome to do this for any boards you make, but I don't want to do it for the default Sampler population option.

Re balancing: I can't think of any cases in the lab where one would want a 100Ohm termination. This isn't a termination for digital differential signals like LVDS. The intended use case for this is things like digitising the IF of a mixer near DC, for example for phase locks. That requires a 50Ohm termination at the IF to make the mixer behave properly. We can't use a 2:1 balun for this because it needs to work down to DC.

The BNCs balancing you refer to is more about avoiding ground loops that introduce low-frequency noise than about using 100R differential signals like that.

TBH, this use-case is pretty uncommon, so I wouldn't object too much to just DNPing the termination resistors. One can always add an external 50Ohm resistor and T-piece at these frequencies if needed.

Re power consumption: yes, users have to be aware of that, but we're using pretty beefy resistors and most lab sources can't drive enough current to damage them. Even with 100R you can still break it if you try.

hartytp, 2017-12-07:
Re capacitors: all the things Greg said.

hartytp, 2017-12-07:
> I suspect that there are neither noise advantages nor drivers that do drive it.

No noise advantages, and no drivers that can or should drive 10V into 50Ohms!

The purpose is about terminating loads that require it such as mixers. The output from these will generally be 100mV or less, so power dissipation isn't an issue. Note that the physicists practice of terminating quasi-DC +-10V outputs from photodiode amplifiers is strongly discouraged and not the point of this terminator; reflections aren't generally an issue at 100kHz unless you really spend a lot of money buying drums of BNC cable.

closed 2017-12-07 by hartytp

jordens, 2017-12-07:
What I meant is that I don't think you get better SNR out of that very mixer application with that INA and 50R than you would get with 100R.

jordens, 2017-12-07:
If you are OK with DNPing the resistors, we could as well DNP the switches.

hartytp, 2017-12-07:
> What I meant is that I don't think you get better SNR out of that very mixer application with that INA and 50R than you would get with 100R.

No, what I mean is that the bias currents from the IF port have to flow through that resistor. The mixers are optimized for 50R loads. 100R loads give a bigger signal (higher phase to voltage gain for phase detector applications), but they tend to degrade the mixer's performance in other ways (worse offsets etc).

There are some nice app notes I can dig out for you if you're interested. I also have pages of lab book notes characterizing these affects for various mixers. 

> If you are OK with DNPing the resistors, we could as well DNP the switches.

If we DNPd the resistors then, yes, we would also DNP the switches. We might also consider DNP ing the termination LEDs and removing the cutouts on the front panel. 

For the time being, I don't want to DNP anything, I just want to send the boards to manufacture.

FWIW, the cost of these components is so low, and they do have some uses, so I probably wouldn't want to DNP them in any of the boards we buy. We can stick a warning on the schematic/docs reminding users not to try to drive silly amounts of current through these resistors. But, as I said before, most signal generators/photodiodes/etc won't be able to source/sink nearly enough current to actually cause damage.

jordens, 2017-12-07:
I know those app notes. But from those exact app notes, you will also see that just dumping it into 50R is not the right one for SNR.

hartytp, 2017-12-07:
> I know those app notes. But from those exact app notes, you will also see that just dumping it into 50R is not the right one for SNR.

Depends on what you're trying to optimise for.

Anyway, this is getting way too details ;)

I don't think those components do any hard to the design, and they don't add a significant cost. As they do have some use-cases, I'm not inclined to muck around with the design/population options at this stage.

jordens, 2017-12-07:
The biggest problem I see here (and equivalently for the entire class of DIP switches, direction-settings on I2C, termination options, solder bridges, mux settings) is that at least 10%-20% of the users will have one problem or another with the correct usage, the constraints when they work and when they break, the level of software support, implications on power usage and limits, the interdependencies among them, and the various other technical side-effects and side-considerations. I can already smell the broken boards because people run their 8 current transducers into those 50R terminations. They look like they are meant for exactly that purpose. The transducers will happily drive 10V...

hartytp, 2017-12-07:
> The biggest problem I see here (and equivalently for the entire class of DIP switches, direction-settings on I2C, termination options, solder bridges, mux settings) is that at least 10%-20% of the [...]

Firstly, small grumble: if you felt strongly about this then it would have helped to have raised the issue earlier (e.g. v1.0 of the board) rather than waiting until they go to production.

> the level of software support

These switches have no software support at all. They're switches.

> the interdependencies among them

There is no interdependence that I can see.

> I can already smell the broken boards because people run their 8 current transducers into those 50R terminations. 

As you've pointed out elsewhere. There are plenty of ways to break all of this hardware. If the max current is clearly specified on the schematic and docs then I have no issue with this.

jordens, 2017-12-07:
Pretty sure I can come up with a dozen instances where we pushed against those "options" way before any production. I'd think that DNPing something on the current Novo v2.0 run can still be done.
On the switches I was referring to cases like the DIO direction switches. The interdependence will already start to appear when we try to write an API for Novo that uses SI units.

hartytp, 2017-12-07:
> On the switches I was referring to cases like the DIO direction switches. The interdependence will already start to appear when we try to write an API for Novo that uses SI units.
> Pretty sure I can come up with a dozen instances where we pushed against those "options" way before any production.

Let's not mix up the general issue of DIP/direction switches in Sinara with these particular termination switches.

>  I'd think that DNPing something on the current Novo v2.0 run can still be done.

I don't have the bandwidth to deal with this right now. Sorry. Let's just go to prototype production with the boards we have. We can think about DNPin these before the full production run.

jordens, 2017-12-07:
To me this is a general topic that has occurred multiple times in a similar pattern.

Sure. I am not insisting on these changes. It's a suggestion and we only want to improve these devices.

hartytp, 2017-12-07:
> To me this is a general topic that has occurred multiple times in a similar pattern.

hartytp, 2017-12-08:
Okay, thinking about this, my preferred option would be to keep the termination resistor and DIP switch, but to change the termination impedance to 240R.

- That's about smallest resistor value that keeps the power dissipation below the rated 0.5W for the resistors used on sampler, assuming max +-10V inputs.
- It's a standard termination resistor for 4-20mA transducers (this is the max termination that works with +5V compliance transducers). Having sampler be able to digitize current loop sensors directly is a nice feature IMHO.
- It's a low enough resistor value to provide an adequate mixer IF termination for most cases.
- At a pinch, it will also allow us to digitize things like uA signals from unamplified photodiodes (1uA*G=1k*240R=240mV, which should be fine with a bit of averaging).

Everyone happy with that proposal?

> To me this is a general topic that has occurred multiple times in a similar pattern.

I don't personally see it that way. For things like direction dip switches, etc I can see an argument in terms of complexity. But, in this case:
- The switch has a simple function, which is clearly indicated by an FP LED. It's clear from the schematic (and soon from the Wiki) what it does and how it's supposed to be used. So, I don't see any issue with user confusion or complexity.
- There is no real interaction with software. IMHO, the software should always report the voltage at the BNC in SI units, regardless of the termination. If users want to use the termination resistor as a V-I converter or with a signal with non-negligible source impedance then they need to do a trivial conversion. But that's always the case (e.g. if they add an external resistor).
- If we up the resistor value to 240R then this become very hard to break (very low impedance driving source with >10V compliance is not that common in the lab).
- These components do not significantly impact the complexity/cost/power consumption of sampler, so the don't do any real harm. And, the V-I termination option can be quite useful (I have a few 4-20mA sensors in my lab so its nice to be able to connect them directly to sampler).
- We could do population options but, given the above, I don't think it's worth it.

jordens, 2017-12-10:
240 R sounds good to me.