sinara-hw/sinara #421: Clocker v1 errata and tests

gkasprow, 2017-12-12:
Clocker v1 arrived.
There are some issues:
- [ ] voltage setting resistor should be 33k instead of 3.3k
- [ ] front panel SMA assignment is wrong - clock input is on bottom, not top. It was in initial project and changes were not transferred to the panel design.

Otherwise it works fine.
Today I will ship it to @hartytp 

[image]

hartytp, 2017-12-12:
Great, thanks!

jbqubit, 2017-12-12:
Looks great!

hartytp, 2018-02-01:
Had a look at clocker today. The error on the FP threw me for a while other than that, it looks good...

Source for all measurements is a Wenzel 100MHz oscillator (+13dBm) driving clocker directly via the FP SMA.

hartytp, 2018-02-01:
Rise time is something like a few hundred ps ignoring the reflections. The ADCLK itself should provide more like 100ps, but once the balun and the scope BW (4GHz) are taken into account, this is roughly what one expects.

Pk to pk signal is about 800mV. Expect 16.Vpp out of the ADCLK, that's 1.1Vpp into 50R, and 900mV once the 2dB (ish) balun loss at 100MHz is taken into account.

SI looks generally good on the scope. So this all checks out...
[image]
[image]

hartytp, 2018-02-01:
Phase noise looks excellent (plots to follow later).
- noise floor at about -145dBc/Hz
  - NB this is not expected to follow the 20*log(f) scaling with the frequency of the carrier.
  - limit to white noise is probably the fact that I'm doing this at 10MHz. dV/dt is: 2xpixAxf= 2xpix(sqrt(2)x1.4V)x100MHz=1V/ns. Looking at figure 12 of the datasheet, we're definitely still in the regime where the jitter is limited by the input slew rate, so this makes sense.

hartytp, 2018-02-01:
I did see a few spurs, particularly on the AM spectrum, which look like they come from the SMPS. Probably worth adding an LC filter (or ferrite) between the SMPS and LDO. If we do that, make sure to add a tantalum cap or something to prevent ringing in the LC filter.

hartytp, 2018-02-01:
Anyway, great work @gkasprow on another design that worked more or less perfectly on the first attempt! It's a very low priority compared with Sayma/Booster, but at some point I'd like to order some more of these with the bug fixes applied.

gkasprow, 2018-02-01:
I have a few pieces still in my lab. I will test them in a few days.

hartytp, 2018-02-01:
@gkasprow There are no more tests that I want to see done; as far as I'm concerned this board works and nothing else is needed.

I'd prefer to buy v1.1 boards with the fixes applied (e.g. a new front panel!).

gkasprow, 2018-02-01:
new front panel is already attached to all shipped boards. We can ship you correct one.
All the fixes are already applied to existing boards, so the only difference would be this ferrite bead.
You wanted 4 clocker prototypes, @jbqubit  wanted 2, we did 10, so there are still 4 free ...
Actually, I can mount these ferrite beads on existing 4 pcs so they won't differ from v1.1.

hartytp, 2018-02-01:
@gkasprow okay, np. If you apply the fixes above then that's fine by me.

gkasprow, 2018-02-01:
Here is a fix:
100R/1A ferrite bead + 6.8uF tantalium cap
[image]

hartytp, 2018-02-01:
> 100R/1A ferrite bead + 6.8uF tantalium cap

What P/N?

gkasprow, 2018-02-01:
well, I took one from a drawer. It is 100R@100MHz, 150mOhm, 1A ferrite bead. i.e. [this one](http://pl.farnell.com/tdk/mpz1608d101bt/ferrite-bead-100r-1a-0603/dp/2800770)
6.8uF/10V SMDA, ESR is roughly 3R, one can use [this](http://pl.farnell.com/avx/taja685k010rnj/cap-tant-6-8uf-10v-case-a/dp/2283842?st=6.8uF%20SMDA%2010V)

gkasprow, 2018-02-01:
I modified and tested remaining boards

hartytp, 2018-02-01:
That won't do much for frequencies <=100kHz will it? The cap ESR is comparable to the bead's impedance isn't it? At that frequency, the FB is just a small inductor IIRC.

gkasprow, 2018-02-01:
The tantalium cap is just to lower the Q of the LC circuit. There is 10uF cap at the LDO input.

gkasprow, 2018-02-01:
edit 4.7uF ceramic cap

hartytp, 2018-02-01:
Still not convinced that's enough. At 100kHz the 4.7uF cap is about 0.3Ohm. Hard to tell from the FB datasheet, but the impedance at 100kHz looks <1Ohm there as well.

gkasprow, 2018-02-01:
what we want to achieve is to improve LDO PSRR at higher frequency. Isn't LDO performing good enough at 100kHz? I can place small inductor there, let's say 10uH

hartytp, 2018-02-01:
Most of the noise I saw on the output was below 1MHz. So, yes, an inductor and tantalum would probably be the best solution. Let's use the same inductor we already have on the SMSP to avoid introducing an additional BOM line.

gkasprow, 2018-02-01:
I used 6.8uH 2.2A inductor I have in lab:
[image]

hartytp, 2018-02-01:
Nice, that should do it.

hartytp, 2018-02-01:
Tried increasing the SMSPs output voltage to 4V to improve the PSRR of the LDO. Didn't seem to make any difference...

jbqubit, 2018-02-02:
@hartytp Will be interested in seeing your phase noise plot. Please also post that PN for your reference oscillator (or give part number). Do you have a scope that generates eye diagrams?

hartytp, 2018-02-02:
Plots look essentially the same as the ones I posted in the issue about the LTC6957 (or whatever the PN is); they're basically what the ADCLK948 datasheet gives. But, yes, I'll post them when I have time.

Will post the oscillator PN. FWIW, we always measure and plot the phase noise of the reference at the same time. That shows the overall noise floor of the measurement.

Eye diagrams aren't useful in this context and any measurement would just be the noise floor of the scope. The phase noise spectrum has everything you need.

hartytp, 2018-02-02:
@gkasprow Shall we make the tantalum after the inductor 100uF (assuming that's cheap and readily available in an appropriate package) to really kill the SMPS crud in the 10kHz regime?

hartytp, 2018-02-03:
> It would be interesting to get a number for the isolation of the clock mux. With one or both inputs being sinusoidal.

jordens, 2018-02-03:
What is the "datasheet value"? It's unclear (to me at least) what that 90 fs rms value means. I'd expect that to depend dramatically on at least signal slew rate.
ACK the rest. I.e. 5 dBm output.
Just out of curiosity: how much power is in the even harmonics?

hartytp, 2018-02-03:
> What is the "datasheet value"? It's unclear (to me at least) what that 90 fs rms value means. I'd expect that to depend dramatically on at least signal slew rate.

ACK, I thought it was better specified than that, but I may be remembering a different datasheet. Will measure once my cables arrive.

> Just out of curiosity: how much power is in the even harmonics?

Will do.

jbqubit, 2018-02-05:
As a reference for a clock distribution amp see [SDI HPDA-100RM](http://www.spectradynamics.com/products/hpda-100rm-high-performance-distribution-amplifier-80-20mhz/).

hartytp, 2018-02-05:
NB about that kind of thing:
- They don't perform sine-square conversion, which is usually where a lot of the noise comes from for lower frequency references. i.e. if you clock a PLL/DDS off the output from that at 100MHz then don't expect the performance to be as good as the datasheet value.
- IIRC (conversations with the manufacturer) the propagation-delay temp co is about 1ps/K compared with 0.1ps/K for the ADCLK, so AFAICT (not characterised carefully) if you really care about phase stability the ADCLK is the way to go.
- AFAICT, the noise from the ADCLK is low enough to not be a limit in any of the applications we have in mind (see the phase noise models Weida's posted).

That aside, yes, those are very nice buffer amps...

gkasprow, 2018-03-05:
@hartytp I just realised that we have already good LC filter after the buck converter. The converter has embedded inductor. So at its output there is already 6.8hD inductor + 22uF tantalium + 22uF ceramic and then LDO.
I had some brain fade and didn't realize this :D

hartytp, 2018-03-06:
@gkasprow ack. In that case, we can leave it as is.

Before I forget, can you send me another two boards with the other fixes applied, please? (Very low priority).

gkasprow, 2018-03-06:
I handled them to Technosysem some time ago...

hartytp, 2018-03-06:
Okay, so I order from them? Will they apply the fixes above?

gkasprow, 2018-03-06:
yes

marmeladapk, 2018-06-13:
_Migrated to sinara-hw/Clocker#2_

closed 2018-06-13 by marmeladapk