sinara-hw/sinara #422: Urukul LVDS receivers mode (Type 2) problem

a-shafir, 2017-12-12:
There was a very strange looking transition process on all the input LVDS lines of Urukul:
- when a line driven by default (inactive) "1" it works ok
- when a line driven by default (inactive) "0" there are many random pulses, looks like some settling process for ~100ns and than (when there are many "1") works.

Looks like something gets charging. Possible it is related to FMC DIO transmitters and other passives connected to the test system.

On the 3U_BNC I/O board it was no such issue.

After turning the LVDS transceivers into "Type 1" by soldering jumpers on R1A/B & R14A/B it fixed.

Note: Type 2 is unbalanced, see: http://www.ti.com/lit/an/slla108a/slla108a.pdf page 8

Need to verify if we indeed need this Type 2 mode and if yes possible need to connect all cfg together and route it to the sw or jumper. 

Or need to verify why FMC DIO is incompatible with Type 2 etc.

dhslichter, 2017-12-12:
It's important to match Type 1 and Type 2 transceivers on both ends of an LVDS link.  Type 2 has nice properties for disconnection, in that a floating input to a receiver gives a definite logic state (low).

jordens, 2017-12-12:
And are the Type-2 receivers really the root cause here or just the trigger?

a-shafir, 2017-12-12:
It is certainly not 3-state issue. 
The output is the KC705 with FMC DIO. The test code sets all the FMC i/o once and than sends SPI data every ~100ms. So the FMC DIO buffers there and Urukul acts as receiver on that lines. I am not sure how FMC DIO assembled are there any DC/AC issues on the signals.

How you tested Urukul, with FMC DIO as well?

gkasprow, 2017-12-12:
I tested it with [DIO-Tester ](https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/EEMs/Tester) and did not observe any issues.
The FMC LVDS board does not need to have the SPI updated every 100ms. It's enough to load the configuration once. Maybe there is some issue and the buffers have their direction changed periodically.

a-shafir, 2017-12-12:
Yes, the FMC configuration is once, of course. 

Every 100ms the DDS driver test is running just to see all needed on the scope.

DIO-Tester is using different tranceivers than FMC-DIO and certainty has no capacitors etc. 
I am pretty sure that the issue is combination of FMC DIO and Type 2 receivers.

Any way, the best is to have switching between Type 1 and 2 without soldering since both have some advantages. I think using a SW will be the best. The changes on the layout are minor.

Can you add it to the next revision to do list?

hartytp, 2017-12-12:
> Any way, the best is to have switching between Type 1 and 2 without soldering since both have some advantages. I think using a SW will be the best. The changes on the layout are minor.

> Can you add it to the next revision to do list?

Do we really need this?

IMO, we already have too many switches and jumpers on these boards, let's not add more unless we really need to. Just adds more things to document/test more places for users to get confused, etc.

a-shafir, 2017-12-12:
>Do we really need this?

I  heard there was some discussion about Type 2 advantages.  What i know:
- Type 2 has some advantages comparing to Type 1.
- Type 2 works with DIO-Tester And it shall work with Kasli. Type 1 shall be ok too.
- Only Type 1  works with KC705->FMC DIO (v1.2)->VHCI(v1.1)
- 3U_BNC (v1.1) receivers are configured for Type 1
- there is a block of 4 switches on Urukul already and one of then not in use. It must be set correctly.
- cfg signal can be routed to sw

So i see at least on Urukul IMHO it makes sense to use the spare sw.

gkasprow, 2017-12-12:
I don't see any other mechanism that would cause such behaviour.
Can you look at the LVDS signal with scope please? Both P and N lines using 2 probes. This would clarify the things. FMC DIO has ordinary LVDS transceivers that produce either negative or positive outputs. They can also produce Hi-Z when set to input mode.
The situation where selection of Type1/2 mode makes sense is HiZ - only in this case such feature of the LVDS receiver would cause such effect. In practice this introduces some little offset at the LVDS input causing so induced noise is not able to trigger the LVDS input.

a-shafir, 2017-12-12:
I am not sure what's wrong. It looks like some strange oscillation over the receiver. Possible the internal pull-up FSEN is too weak? When i connect a probe it pulses continuously on the high freq during 1st "0" pulse. Possible it is related to the long lvds cable (to on and from VHCI board). Possible there is something on all channels of the FMC DIO that charges/discharges and settles somehow. With R1A etc populated it works fine.

hartytp, 2017-12-12:
>there is a block of 4 switches on Urukul already and one of then not in use. It must be set correctly.

Right,
1. The fact that we already have more configuration switches than we would like isn't an argument to add another one
2. The plan is to cut down the number of switches in the next hardware revision. For example, SW[0] is currently used to select between AD9910/AD9912 this will be replaced with a fixed pull-up in the next version, so won't be needed any more.

If this is really needed, we can add it, but let's try to get to the bottom of it before doing that.

gkasprow, 2017-12-12:
@a-shafir please test one more thing.
Take the multimeter and measure if there is GND connection between the Urukul ground and KC705 ground.

a-shafir, 2017-12-12:
hm. There is connection of GND (somehow) but yes, the VHCI cable itself does not have ground lines.

The shield can be used and on the FMC DIO it seems to be connected. But on VHCI breakout PCB the double connector installed is not attached to the shield.

And I see no nice way to connect the ground to VHCI 1.1 and also to KC705 (exept using one of the BNC). 

This can be the problem. What you suggest to do? Shall we also update VHCI breakout to use the shield or need a label on it about the grounding?

a-shafir, 2017-12-12:
@hartytp 

I agree. Personally i see no reason to mess with Type 1/2 at least for Kasli. It is not hot-plug device by the design etc.

At least we need to move this Type config to one signal and make it easy to solder if need (currently there are 4x 0402 spread on the board with no labels etc). Also there need a pull-up resistor just to make sure that it ok. This is the must.

And we need to verify is there any real advantage of type 1 (or 2) and unify it on all the boards IMHO.

gkasprow, 2017-12-12:
@a-shafir what is the resistance of the connection?
It's good to have the shield connected but it's not a must. There is dedicated GND line within LVDS pairs.
LVDS are balanced so they don't produce much return current.
Can you post an oscilloscope screenshot with two probes connected to the P and N lines of problematic pair with DC coupling on and 0.5V/div scale?

a-shafir, 2017-12-12:
@gkasprow ok, i see the FMC DIO board you sent has GND fix. So yes, there is GND (one pin) via VHCI cable pin 18.

I think possible there is some issue related to the FMC R244, R245 C140/C141 or other components. The FMC board is populated not the same as the sch shows.

Please check.

gkasprow, 2017-12-12:
@a-shafir these components are foreseen for AC-coupled applications in stacked silicon detectors. So they should be not mounted. In case of C140/C140 it does not matter since they are shorted by 0R. I tested all these boards with BNC-IO and worked fine.
So give me a favour and send me the scope screenshots.

a-shafir, 2017-12-12:
@gkasprow try BNC-IO in Type 2. Currently it assembled in Type 1. Type 1 works on urukul as well.

gkasprow, 2017-12-12:
@a-shafir I believe it works. But we are not looking for temporary solution but for a physics behind the issue you observe. It may work now, but may stop working under some circumstances. That's why I asked you for these measurements.

a-shafir, 2017-12-12:
@gkasprow i already modified to Type 1 and have no time to solder it back since there another issues.

You tested with 3U_BNC. According the SCH i see there is R138 100R and R152 10K. I.e. FSEN is low and it works in Type 1 mode.

I will try to find time and will check Type 2 mode on 3U BNC. We have only one DDS board and i would like to avoid changes there just for this test.

For the scope. It looks like the output rises (0-> than, after 20ns or so some random pules for ~50ns and than it gets stable 1). After that (when there is no long 0) it works fine.

gkasprow, 2017-12-12:
@a-shafir Again, publish the scope screenshots. I'm interested in levels, not timing :)

gkasprow, 2017-12-12:
LVDS is centered around 1.25V. So in L state the P line should be ~1.5V,  N line should be ~1V.
In H state the levels should be opposite. I have a suspicion that in your case in the inactive state both lines have similar level, close to 1.25V, or at least the voltage between them is close to 0. Such condition should never take place. And in this condition the case you describe has chance to happen.

gkasprow, 2017-12-12:
It does not matter what Type you select. Just do the measurements with P and N lines !

a-shafir, 2017-12-12:
What minds "inactive state"?

gkasprow, 2017-12-12:
Inactive state means logic level that does nothing, i.e. CS=H.
The FMC DIO board has assembly variants, so yo should have a look at valid one. 
Your board seems to be assembled according to the schematic variant here:
[FMC_DIO_32ch_lvds_a.PDF](https://github.com/m-labs/sinara/files/1553526/FMC_DIO_32ch_lvds_a.PDF)

a-shafir, 2017-12-12:
the pins configured for output permanently. I have not change the config of DIO but just moved from Type 1 to Type 2 to make it working.

Yes, it looks like this variant plus the ground pin on VHCI.18
What was the idea, to have the inactive state voltages equal?

gkasprow, 2017-12-12:
This is the question I ask. They should never be equal. If there is 0 between them there is something wrong with the circuit.

gkasprow, 2017-12-12:
how do you configre the DIO directions?

a-shafir, 2017-12-12:
Ok, i just checked on the not used channel (not terminated with 100R). _P ~1.08V _N ~1.38V

For DIO configuration I am using zotino code on ARTIQ. I checked it certainly configured all the pins correctly.

gkasprow, 2017-12-12:
Is this channel causing problems and you observed issues on it?
Please check it on a channel that caused problems.
Just measure voltages in H and L state, on the P and N lines.

a-shafir, 2017-12-12:
all channels was problematic. all. before i changed on the receiver to Type 1.

gkasprow, 2017-12-12:
OK, so please measure voltage in opposite state.
When you changed to Type 1 you fixed the problem but not the reason behind it.
In normal operation receiver Type does not matter. In matters only in failure mode. And this suggest there is a problem.

a-shafir, 2017-12-12:
[image]

a-shafir, 2017-12-12:
This is one of the channel with no load and termination. The signals are exactly the same but 180deg phase.

gkasprow, 2017-12-12:
ok, but why there is no termination? I 'd like to see a channel that is causing troubles with configuration where you experience such troubles. Receiver type does not matter.

a-shafir, 2017-12-12:
I just unable now to work with the channels connected to the DDS. 
The DDS channels are terminated 100R. I checked. 
Again, for some reason with R1A/B, R14A/B on DDS board open it goes to this oscillation condition.
Attaching the probes to dds makes the condition only worst.
With R1A... 0R it works.
This what i know.

gkasprow, 2017-12-12:
Could you make a screenshot with this condition? I'm trying to help you solving the problem.
connect two probes to N, P and possibly third to the place you observe these oscillations.

a-shafir, 2017-12-12:
Again, i will need to modify the board. I have only the board and need to test something also. I will do this later.

gkasprow, 2017-12-12:
OK, I will try to recreate the problem with KC705, FMC DIO and DDS but I'll need your codes to do so.

a-shafir, 2017-12-12:
The code is not really important. This condition appeared when a line (for instance SCLK) stays 0 long time (100ms etc) and than there are pulses or transition (like SPI). The after the very first rise it oscillates a while. When it opposite (stays 1 and pulses to 0) it does not oscillate. All the signals (i tested SCLK, CS0-2 etc) behave nearly the same. When attaching a probe to input pit it getting worst (btw it reacting differently for P and N inputs load). It was pretty persistent.
You can generate that pulses with any code.

Note: I am using a standard setup (see above). The flat cable between VHCI and DDS is about 20cm.

gkasprow, 2017-12-12:
OK, I will try to recreate the problem.

gkasprow, 2017-12-15:
I looked into that problem.
The LVDS buffers used on Tested board which I utilise to test DDSs uses high current that guarantees valid LVDS levels with 50Ohm diff termination which is a case in MLVDS bus. It is able to deliver 0.9V pp at 50Ohm. And in our case we have 50Ohm because LVDS line is terminated on both sides. This is equivalent to 18mA of output current
The LVDS drivers used on FMC DIO board are FIN1101K8X which are able to deliver 250..450mV at 100R load which gives 2.5..4.5mA of output current.
So the same current flowing through 50R load gives 125...225mV of diff voltage..
Type-1 receivers have thresholds centered about zero with 25 mV of hysteresis to prevent output oscillations with loss of input; Type-2 receivers implement a failsafe by using an offset threshold.
So if we assume that we get lowest voltage level that drivers on the FMC can deliver , we end up with +/-125 mV.

[SN65MLVDS040](http://www.ti.com/lit/ds/symlink/sn65mlvd040.pdf), page 3, specifies that with Type 2 selected, 65 mV ≤ Vin ≤ 135 mV gives undefined output, and this seem to be the case. But open circuit gives L state.
With Type 1 receiver, the undefined input range is –35 mV ≤ V ID ≤ 35 mV. But open circuit gives undefined value.

So @a-shafir was right - while operating with low current LVDS drivers, Urukul should have Type 1 receiver enabled. 
This should be not the case while operating with Kasli.

a-shafir, 2017-12-15:
OK. This is clear.

@hartytp I think Urukul board can be used in the other applications than Kasli. In this case it makes sense to have the type1/2 modes easier configurable. At least it makes sense if not connect it to the dip switch but have only 2 "option control" resistors (per each EEM) easy to configure - 0603 types placed/labeled better.

As i understand the most principal reason of using Type 2 is the EEM1 option. When it is not plugged it can affect some of the signals. But it can be solved on the configuration level. As one of the option an cpld logic can be used to mask that signals.

BTW: i think it makes sense to upgrade the CPLD to XC2C256-6TQG144C that is pin to pin since the current one is close to the limit.

gkasprow, 2017-12-15:
Since Urukuls were already produced in qty of 10, it would be good to replace these existing CPLDs with bigger ones not to maintain 2 code versions.
We already have one switch that selects working mode, so it can be used to control receiver type as well. We can also stick with Type 1 permanently and ignore unused inputs by disconnecting them in the CPLD.

a-shafir, 2017-12-15:
@gkasprow yes, it sounds like a good idea. It allows adding more config pins to CPLD logic.
Currently i have tested code (it is not in git yet) that allows partial write to the CPLD cfg register. This feature can be used to add more config options but to keep it fast on the principal functions. Also the NU-mode can be upgraded. But at the moment it is almost impossible to add any new features to the CPLD because of the limits.

hartytp, 2017-12-15:
> The LVDS buffers used on Tested board which I utilise to test DDSs uses high current that guarantees > valid LVDS levels with 50Ohm diff termination which is a case in MLVDS bus. It is able to [...]

This doesn't sound like a bug in Urukul to me. The interface is LVDS, which is well-defined. In this case, the board doesn't work because the driver (FMC DIO) doesn't produce decent signal levels. That sounds like a problem with the FMC DIO board, or at least the way it's being used to me, and not a problem with Urukul.

Switching to type 1 to make it easier to drive this from marginal sources sounds like a bad idea IMHO.

gkasprow, 2017-12-15:
@hartytp that's true. FMC DIO should simply have stronger drivers. With weak drivers it may have troubles with long cables.
Take into account that Type 2 is not LVDS. It is MLVDS :)
And the margin is not that low - we have 125mV which fully conforms with LVDS requirements.

hartytp, 2017-12-15:
> @hartytp that's true. FMC DIO should simply have stronger drivers. With weak drivers it may have troubles with long cables.

Let's fix that and leave Urukul as is.

hartytp, 2017-12-15:
> And the margin is not that low - we have 125mV which fully conforms with LVDS requirements.

I thought LVDS usually specifies +-200mV with a common-mode of 1.2V.

gkasprow, 2017-12-15:
Actually the standard says about +/- 350mV at the output, I mean what is acceptable at the input.

gkasprow, 2017-12-15:
As a quick fix one can remove terminating resistors at the FMC DIO which are used as outputs :)

hartytp, 2017-12-15:
Can we close this then?

Changing the FMC DIO driver needs a separate issue.

We could add a note to the Urukul schematic about required input levels if people want. Again, that needs a separate issue.

Changing the CPLD should have a separate issue.

closed 2017-12-15 by gkasprow

gkasprow, 2018-01-26:
Type mode is now controlled by CPLD