sinara-hw/sinara #423: Sayma_RTM documentation

jbqubit, 2017-12-12:
@filipswit is working on Sayma_RTM v1.0 documentation. It will be structured similar to Sayma_AMC v1.0 documentation discussed in #133.

jbqubit, 2018-01-11:
@filipswit How's the documentation coming?

jbqubit, 2018-02-07:
@filipswit How's the RTM documentation coming?

filipswit, 2018-02-08:
Currently it looks like this:
[SaymaRTM.pdf](https://github.com/m-labs/sinara/files/1708297/SaymaRTM.pdf)

There is still some placeholders, sectioning is quite finall. After your AMC review, some of your comments affect to RTM so first I'd like to have 'good' version of AMC and then coppy and finish RTM. 
Summarising it: It's still not worth reading;)

jbqubit, 2018-02-14:
@filipswit This is looking good. :) Let me know when you're ready for a careful reading.

jbqubit, 2018-02-27:
@filipswit Is this ready to review?

filipswit, 2018-03-04:
Here is some update for RTM. 
[SaymaRTM.pdf](https://github.com/m-labs/sinara/files/1779090/SaymaRTM.pdf)