sinara-hw/sinara #424: intermittent KU FPGA bringup failure

sbourdeauducq, 2017-12-14:
Sometimes right after the Sayma board is powered up, the KU FPGA is inoperative with one of the symptoms being a 0xffffffff IDCODE read by OpenOCD (the same symptom exists when the 1V8 power supply fails after the board has been powered for a few hours). This persists until the board is power-cycled.

gkasprow, 2017-12-14:
@sbourdeauducq The question is whether JTAG chain breaks or the FPGA JTAG state machine does not respond due to missing supply rail.
Does it happen on all boards you have?

sbourdeauducq, 2017-12-14:
Happens on at least two of them. Looks like it's the 1V8 rail failing shortly after power-up. Seems more frequent when the flash is programmed.

jbqubit, 2018-01-18:
I don't see this error on my boards. Do you see it on yours @enjoy-digital?

jbqubit, 2018-01-29:
@sbourdeauducq If you think this is due to https://github.com/m-labs/sinara/issues/358 please close.

sbourdeauducq, 2018-01-30:
I don't know, the power supply problem is not resolved.

closed 2018-05-03 by hartytp