sinara-hw/sinara #436: Sayma manuals as source code

jbqubit, 2017-12-14:
It's better if a text-based source for the Sayma documentation is put in git. This permits work by multiple editors and permits easy merging of changes. 

## location

@jordens How does the following location sound? 

sinara/ARTIQ_EE/PCB_Sayma_AMC/manual
sinara/ARTIQ_EE/PCB_Sayma_AMC/manual/images

## building
Give instructions as a README of how to build the documentation to PDF.

jordens, 2017-12-14:
AFAICT This is still the same bad idea as before.

jbqubit, 2017-12-14:
Isn't uploading text-based version of manual better than binary?

jordens, 2017-12-15:
Ask me again once you have looked at the dozens of times where we have given advice on this.

jbqubit, 2017-12-15:
@jordens Please let me know where your advice(s) on this subject is cached.

jbqubit, 2018-01-18:
> AFAICT This is still the same bad idea as before.

I still think it's a good idea to have official documentation of Sayma be in PDF format. But I'm failing to understand why git is not good for keeping the source. @jordens Please let me know where you've commented on this and I'll reread.

jordens, 2018-01-19:
Every discussion where we gave advice about what to put into the repository. If you really need to play that game, take #39, #46, #328 to begin with. I won't engage again, this is too frustrating.

dhslichter, 2018-01-19:
I guess I am a little in the dark here as well @jordens -- as long as one is not keeping any documentation outputs (PDFs, associated temp files, whatnot) in the repository, what is the problem with having a set of, say, `.rst` or `.tex` files with the documentation source ONLY, which allow easy versioning and tracking of edit history?  This is what is done with the ARTIQ manual currently, for example.  As long as one is stringent about NOT having the outputs also in the repository (but only in release packages as described in #328, or hosted on readthedocs or the M-Labs website), I fail to see how this doesn't work.

jbqubit, 2018-01-19:
@jordens The Issues you cite do not pertain to the proposal.

dhslichter, 2018-01-19:
> This repository is poisoned.

??  sorry, unclear on what you mean here @jordens, are you saying that the documentation source should not be hosted in the sinara repository?

jordens, 2018-01-19:
@dhslichter Yes.

gkasprow, 2018-01-19:
Filip is doing all documentation using Tex.

hartytp, 2018-01-19:
Also, having cloned Sinara a few times recently on slow WiFi connections, I think I can guess what RJ means by "poisoned". Sinar repo is needlessly huge due to all the PDFs that have been uploaded.

I'm half tempted to suggest that we just burn the repository and start afresh but with much stricter rules (gitignore all pdfs, "Project output" dirs etc) and use releases or whatever for PDFs (or even just attach them to the relevant GitHub issues).

dhslichter, 2018-01-19:
Ack @jordens, new repository just for Sinara documentation sounds good to me.  I think we can keep the current repository for the design stuff, though, and achieve the "burning" that @hartytp wants by having ONE designated person go through and clean out the Augean stables, then have everyone operate on a much stricter basis going forward regarding generated files etc.  Aggressive use of `.gitignore` is clearly called for.

hartytp, 2018-01-19:
> that @hartytp wants by having ONE designated person go through and clean out the Augean stables 

are you offering? ;)

That's obviously the best way forwards but I can't imagine anyone actually finding the time to do that.

dhslichter, 2018-01-19:
If the government shuts down, perhaps I will have the time :P

marmeladapk, 2018-01-19:
Perhaps whitelist could be easier to implement? I think it's easier to identify which files are needed for project than the other way around.

jordens, 2018-01-19:
Whitelisting is always the default. You have to stage files explicitly before committing.

jbqubit, 2018-01-19:
> My two cents here: the documentation may all be written in Tex, but there will still be a lot of binary items in the sources, such as images.

The size of line drawings (binary) is small. Photographs are bigger (also binary) but few in number and infrequently updated. Latex source will be most frequently updated. 

> Adding all that to Sinara will just further inflate this (already massive) repository even bigger and more unwieldy to work with.

Sinara documentation is a small perturbation of the overall 3 GB footprint. 

> I'm half tempted to suggest that we just burn the repository and start afresh but with much stricter rules (gitignore all pdfs, "Project output" dirs etc) and use releases or whatever for PDFs (or even just attach them to the relevant GitHub issues).

> Also, having cloned Sinara a few times recently on slow WiFi connections, I think I can guess what RJ means by "poisoned". Sinar repo is needlessly huge due to all the PDFs that have been uploaded.

I agree that the repository is bloated but that's outside the scope of this Issue. I created a new Issue to discuss this here #486. For present let's keep eye on documentation for Sinara.

-------

Restatement of Issue topic based on feedback above. 

It's better if a text-based source for the Sayma documentation is housed in the Sinara repository. This permits work by multiple editors and permits easy merging of changes. Proposed location:

- sinara/ARTIQ_EE/PCB_Sayma_AMC/manual
- sinara/ARTIQ_EE/PCB_Sayma_AMC/manual/images

Standard for inclusion of images and other binary files to follow that used by the physics [arXiv](https://arxiv.org/help/bitmap/index).

closed 2018-02-07 by jbqubit