sinara-hw/sinara #441: Sayma MMC serial interface

jbqubit, 2017-12-14:
The Sayma_AMC FTDI chip exposes four serial ports. 
- /dev/ttyUSB1
- /dev/ttyUSB2 :: usually the MiSoC console 
- /dev/ttyUSB3
- /dev/ttyUSB4 

Is the MMC [serial interface](https://github.com/m-labs/sinara/wiki/Sayma#mcc) mapped to one of these? I don't see any characters on the other serial ports.

jbqubit, 2017-12-14:
Looks like it should be the 4th's USB interface from the schematic.

gkasprow, 2017-12-14:
Yes, it is. Baud rate is 115200, 8, n, 1
You should get a message:
```
[00]Started. CPU clock = 80 MHz, SDRAM clock = 40 Mhz
LWIP HTTP Web Server FreeRTOS Demo...
Waiting for TCPIP thread to initialize...
Starting LWIP HTTP server...
Net up
PHY init done
```

jbqubit, 2017-12-15:
And, pressing the front panel button should trigger issuance of additional debug info, right? 

I don't see that on any of the four serial ports....

gkasprow, 2017-12-15:
@jbqubit  I added this feature recently when was debugging Ethernet PHY.
I will add Exar support and ask you to do firmware upgrade.

jbqubit, 2017-12-18:
OK. Please work with @filipswit to document how to flash MMC independent of ARTIQ. https://github.com/m-labs/sinara/issues/428 Let me know when the updated firmware is ready.

hartytp, 2018-01-07:
@jbqubit Can we close this issue now?

AFAICT the documentation component of this is taken care of in a separate issue, and the firmware upgrade is being tracked more accurately in the 1V8 supply issue.

jbqubit, 2018-01-08:
> firmware upgrade is being tracked more accurately in the 1V8 supply issue.

This is an atomic feature that can be delivered and tested independent of the Exar debug info. Let's leave it open.

closed 2018-01-18 by jbqubit