sinara-hw/sinara #443: Sayma_AMC: MMC doesn't init power ICs

jbqubit, 2017-12-15:
Moving discussion to sinara from https://github.com/m-labs/artiq/issues/873

jbqubit, 2017-12-15:
@gkasprow Can you reproduce this behavior? Does the fix you outlined work on your board?

gkasprow, 2017-12-15:
@jbqubit Yes, this is known issue. You can fix it by shorting pins 1 and 2 of T24 which disables MMC reset over USB

hartytp, 2018-01-07:
@jbqubit Did you try Greg's fix? Can we close this issue? It's not clear to me what the actionable point here is...

jbqubit, 2018-01-12:
I've not tried Greg's fix on my board since I know how to avoid the problem -- don't leave serial connection open on PC when cycling board power. @gkasprow if you're confident in whatever fix you applied for Sayma v2 please close.

jbqubit, 2018-01-18:
I'm using PUTTY and Linux. [Flow control is off.](https://github.com/m-labs/artiq/issues/873#issuecomment-351870640) But I still see problem where MMC hangs upon power cycling if a serial terminal is open on MMC UART. 

Trying next suggestion... I've sorted pins 1 and 2 of T24 but still problem where MMC hangs upon power cycling if a serial terminal is open on MMC UART. 

I've also noticed that there's a white wire on T23 pin 2 that's not connected to anything. It's just pointing in the air. @gkasprow Where did you intend for that white wire to go?

gkasprow, 2018-01-24:
that wire was disconnected on purpose to not cause MMC reboots during USB insertion.
Once you connect it, you will be able to FLASH the MMC using serial port, but during usb insertion, MMC will reboot. The wire connects T23 pin 2 with T24 pin 2 . In such case the short between pin 1 and pin 2 of T24 should be removed.

gkasprow, 2018-01-24:
T23/T24 has nothing to do with MMC hanging caused by entering bootloader mode - it may cause an issue with MMC reboot.
These transistors control the reset circuit of the MMC.
So at the moment the only way to solve is to make sure that  the RTS is high during board power cycling.
You can also exit the bootloader mode by commands developed by @sbourdeauducq

sbourdeauducq, 2018-01-25:
@gkasprow btw I don't think DIP switches should be added. The problem is easy to understand and fix in software, but it should be documented. The USB cable is not used for board operation, and the switches would get in the way of people trying to update the MMC.

gkasprow, 2018-01-25:
I already did it in the next revision

closed 2018-01-29 by jbqubit