sinara-hw/sinara #444: Sayma_AMC artiq_flash :: Unknown flash device (ID 0x00000000)

jbqubit, 2017-12-15:
Tracking of Issue https://github.com/m-labs/artiq/issues/872 moved here from ARTIQ.

jbqubit, 2018-01-12:
This Issue keep rearing its head. I've found one USB cable that works about 50% of the time. And two other USB cables that fail 100% of the time. @gkasprow @hartytp do you see this on your boards?

hartytp, 2018-01-14:
@jbqubit Not sure. I had a bunch of problems with this at first, but I believe they were all related to other known issues (problems when unplugging cables with the board powered on, problems when powering the board on with a terminal open). It was a little hard to disentangle the problems and I didn't want to spend too much time on it.

In general though, my feeling is that USB is pretty flaky and cheap cables cause problems with all kinds of hw.

hartytp, 2018-01-17:
@jbqubit If you have USB cables that reliably reproduce this issue, is it worth posting them to @gkasprow so he can investigate and fix the issue? Would be great to have reliable USB JTAG for Sayma v2.0...

jbqubit, 2018-03-12:
This is rearing its head again on my AMC. Same AMC and USB cable worked last week. Thinking this had to do with the modification I made to flash the MMC over USB I reverted that change. 

I removed this: Pin 2  T23 is shorted to Pin 2 of T24
And added short across Pin 1 and Pin 2 of T24. 

But still getting consistent """Error: Unknown flash device (ID 0x00000000)""" from openocd.

gkasprow, 2018-03-12:
This FTDI chip works at 480Mbit so requires cables with quite good quality. I had some issues in the past with poor USB cables.
Other popular serial port chips work usually at 12Mbit so the cable quality is not an issue.
When you were adding short across pin 1 and pin 2 of T24 you could kill the T23 or T24 by ESD. These tiny MOSFETs can be easily broken by finger touch. Please measure voltage between pin 3 of T23 and GND. Should be above 3V. If it is below 1V it means the T23 is broken and you can remove it.

marmeladapk, 2018-03-13:
@gkasprow I also saw some problems with JTAG while using board patched by @wizath (I'm not saying it's because of MMC). It's not the cables, artiq_flash works reliably with another Sayma. Some mix of power cycling and unplugging USB helped, but I can't remember the exact sequence.

jbqubit, 2018-03-13:
> I had some issues in the past with poor USB cables.

In the past tried seven USB cables. Two were not functional. The other five worked. I'm using a cable that is known to be fine. 

> This FTDI chip works at 480Mbit so requires cables with quite good quality.

Looks like the FT4232H supports slower speed. From the spec sheet... "Supports 480 Mbit/s "High Speed" (HS)/ 12 Mbit/s “Full Speed” (FS), FS Only and "Low Speed" (LS). Is there a way of forcing use of FS mode? 

With no USB cable attached, 
T23 pin (1,2,3) to gnd is (0, 0, 3.3) V
T24 pin (1,2,3) to gnd is (0, 0, 0) V  [pin1,2 are shorted by wire]

With USB cable attached to Linux box. 
T23 pin (1,2,3) to gnd is (3.3, 2.9, 3.3) V
T24 pin (1,2,3) to gnd is (0, 0, 3.3) V

> do you have a jumper installed on the JTAG connector?

cf https://github.com/m-labs/sinara/issues/463#issuecomment-358046337 I have jumper installed.

jbqubit, 2018-03-13:
@sbourdeauducq Helped me figure it out. I had been trying out various OpenMMC firmware (.hex) to get Sayma working in crate. Just now I rolled back to an [older version of firmware](https://github.com/m-labs/sinara/issues/520#issuecomment-370523526) and ran Sayma on benchtop.  Now JTAG is working again ("Unknown flash device" message is gone).   

To get back to MMC flashing configuration I did the following. 

I installed this: Pin 2 T23 is shorted to Pin 2 of T24
I removed this: short of Pin 1 and Pin 2 of T24.

closed 2018-05-03 by hartytp