sinara-hw/sinara #446: Zotino v1.1

hartytp, 2017-12-15:
- [ ] Use two output resistors in series/parallel to improve power handling (200mW short-circuit)
- [ ] reverse C143, C145, C147, C84
- [ ] mirrored SOT23 diode footprint due to Altium bug - part of component is on top layer, pads are on bottom layer.
- [ ] add Shottky diodes at the LDO outputs
- [ ] 3mm higher EMC shield on PSU
- [ ] reverse IDC connector on BNC to IDC board
- [ ] Zero-index all channels
- [ ] swap BNC connector GND terminals
- [ ] either add ~2mm plastic washers on BNC barrel **behind** front panel or move BNC connectors 2mm towards FP. This is to enable fixing the BNC connectors to the panel using the screws.
- [ ]  Please remove the annotation "This module connects to Kasli ..." and replace it with something like "EEM connector: IO are LVDS, I2C is 3V3 LVCMOS, P3V3_MP up to 20mA, P12V up to 1A."
- [ ] run PI and thermal simulations
- [ ] Can we run the digital interface from 2V5?
- Temperature controller issues:
  - [ ] Decrease the thermal resistance between the thermal island and the main PCB/increase the thermal mass of the island by at least 1 order of magnitude -- without active stabilization, the DAC should only be 1-2C hotter than the main PCB. This should also improve the electrical ground connection between the main PCB and the thermal island.
  -  [ ] Remove current temperature controller and DAC over temperature shutdown circuitry.
  -  [ ] Add mounting holes/pads for peltier + heat sink controller by external circuit.
  -  [ ] Leave temperature probe on the thermal island, bring it out to an external connector, which should also provide +12V power, ground and I2C (maybe use I2C-controlled GPIO instead? This way we can do bit-banged SPI if we need to).
   - [ ] Assuming the peltier + heatsink mount on the top side of the PCB, do we want to add a stick-on insulation on the bottom side to screen the DAC from external air currents?
- [ ] Silk + schematic + panel should be "v1.x" not "rev1.x"
- [ ] Only have one set of +-12V regulators, not two!
- [ ] Reduce SMPs outputs slightly, since we're only using +-12 regulators

dhslichter, 2017-12-17:
I would vote strongly for removal of the thermal island and instead using vias/ground planes to spread heat from the DAC and make it equilibrate better to the board.  This ought to keep the DAC from running itself to death temp-wise, at least.

hartytp, 2018-01-02:
@gkasprow Happy New Year!

Sorry to add to the huge to do list, but is there any chance we can put this in the pipeline to do asap?

This is very important to us because we don't currently have a version of this board that we can use for testing (the current version kills the DACs too quickly to be testable). So, I'd like to get a version with bug fixes sent to manufacture as soon as possible.

Hopefully these changes are small and will be quite quick to implement.

Obviously, crucial Sayma/RF PA issues (like the 1V8 diagnostics) should take priority.

hartytp, 2018-01-08:
Greg, is there any chance we can get this sent off to manufacture this week? I'd really like to get this done so we can start testing properly.

For the temperature control stuff, here are my thoughts:
- Let's scrap the thermal island
- Add some way for users to mount a TEC. Do this however you think is best, but here is my initial thought (feel free to overrule)
  - Leave some exposed copper on the board's underside along with some mounting holes that can be used to attach a peltier and heat sink (I'm assuming that it's better to temperature control the ground that the DAC and reference connect to than to try to mount the TEC directly on the DAC, but feel free to do what you think is best here)
  - Run a thermal simulation to verify that a TEC which can drive +-3W into the thermal island is able to change the DAC's temperature by +-2C or so (that should be enough to ensure regulation in most lab conditions once the board has warmed up)
  - Users who want the best stability can attach a self-adhesive insulating strip over the DAC + reference on the top side of the board to scree out air currents
  - Add a 10k NTC SMT thermistor on the board's underside as close the the TEC as possible
  - Bring the 10k thermistor to a header on the top of the board. This header should ideally also connect to the TEC (are there any TECs that have SMT connections?). The header can then be used to connect this to an external temperature controller EEM via ribbon cable/IDC (see the Thermostat EEM page on the wiki for an idea of what this could look like)
  - Scrap all temperature controller related components on the PCB (including the LEDs)

How does that sound?

Thanks!

hartytp, 2018-01-10:
Greg, thanks for posting the initial v1.1 schematics! Can you do me a favour, please and tick off the items you've addressed on the list at the top of this post (that will make my design review easier).

hartytp, 2018-01-10:
Greg, I've checked over the schematic you uploaded and it mainly looks good. Apart from the outstanding issues above (I've ticked off the ones that looked fixed to me):

- Did you run thermal + PI simulations yet? In particular, I'd be interested to see how many C/W a peltier would produce here so that we know we can temperature stabilise this with a sensible TEC power.
- To match the peltier to the I/V rating of the proposed Thermistor EEM, I'm assuming the TEC we'll use will be something like [this](http://uk.farnell.com/multicomp/mcpf-031-10-25/peltier-cooler-4-5w/dp/1639756)
  - What is the diameter of the holes you used for the TEC wires? Will they be wide enough for this kind of TEC? (I didn't see that specified on the data sheet)
  - [ ] Please can you add an annotation on the schematic and silk screen for the polarity of the TEC wires ("TEC+" and "TEC-"). Please name the nets TEC+ and TEC- instead of B1 and B2
  - [ ] The TEC wiring should be able to handle at least 2A with no problems (and without too much voltage drop in the wiring harness). Should we use 1 or 2 more pins for the TEC to ensure that's not a problem? Would it be worth alternating TEC+ and TEC- in the ribbon cable to reduce emission from the TEC current? (Or is that overkill?)
  - [ ] Is the exposed copper area large enough to mount a 15mm x 15mm TEC?
- [ ] After thinking about this more, let's remove the P12V0, P3V3_MP and I2C lines from the temperature controller header. They're not really necessary with the proposed temperature controller EEM design. (This presumably means we can remove the I2C extender from the design, which is nice!)
- [ ] Do the cutouts you've left in around the DAC make any difference to the thermal management? Or, is it better to remove them completely?
- [ ] Can you add a hole in the screening can for the TEC + heat sink?
- [ ] would it make sense to design the heat sink mounting holes around some standard commercially available heat sink hole pattern. I'd like it if we don't have to machine the heat sink to make it work. We should also include a suggested heat sink PN in the schematic
- Are there any disadvantages of having the DAC on the underside of the PCB? I guess we already have quite a few components there, so this isn't a problem (won't make the board more expensive or more fragile than it already is).
- Since the DAC is on the underside of the PCB, there isn't room to add insulating foam on top of the DAC (unless one also adds a filler panel). That should be fine since the DAC's thermal paddle means it is very well heat sunk to the PCB, so small air fluctuations won't be too much of an issue (and we're only aiming for +-200mK or so stability, which isn't too tough to meet).

Other than that, looks good to me!

hartytp, 2018-01-10:
Oh, one other thing: let's wait and see what @cjbe's measurements tell us about the noise on Zotino v1.0 before doing anything about this, but...

The SMPS we're currently using has over current protection, but not over voltage or over temperature protection AFAICT. I wonder if it's a little fragile, and if some of the abuse we've given it could have lead to the SMPS failing on some of our boards?

hartytp, 2018-01-11:
@gkasprow IIRC, for cross-talk testing, we were planning to build a trivial board which breaks out a HD68 to 4 x IDCs. Did we ever do that? I can't find it in the repository. If not, is it possible to do that for Zotino v1.1? Thanks!

gkasprow, 2018-01-14:
@hartytp The HD68 to IDC  and IDC to BNC adapters were added to repo

a-shafir, 2018-01-16:
How about separate with beads the P/N 12V for each load (DAC, and each opamp)? There will be some AC leakage via D5A/B capacitance etc.  Also R3 can be at least <100R safely.

I think additional LC filter makes sense to add for the DAC DVCC line at least to reduce the return AC current via C83, C13, C14 to the power supply that can leak to the analog side.

I think it makes sense add additional inductor similar to L8 and power IC1 and IC35 separately so it will reduce cross-interference. 

It makes sense to route the IC3 and related parts ground directly to IC18 AGND pins (star routing) but not to the gnd plane. Also it makes sense to have the ground plane between the front panel side cut under most of the boards and have connected only under the DAC so it will prevent parasitic digital ground noise.

I can check the layout as well. According the DAC d/s it is quite sensitive for the layout and grounding.

hartytp, 2018-01-16:
@a-shafir 

> How about separate with beads the P/N 12V for each load (DAC, and each opamp)?

I don't believe that a ferrite bead on each OpAmp would be helpful; SMT ferrites typically only work for 10s of MHz and above, but this circuit only works at audio frequencies so their affect will be minimal. I also do not believe that a ferrite between the DAC's analog supply rails and the OpAmps is helpful for the same reason.

Note that there is already heavy LC filtering between the SMPS and the LDOs.

All digital circuitry (which has frequencies that could benefit from a ferrite bead) is already run from a different regulator to the analog circuitry (it's 3V3).

> There will be some AC leakage via D5A/B capacitance etc.

Really? The capacitance for those diodes is of order 1pF. That forms a pole at 300MHz with the 500Ohm output impedance. Again, given that this circuit is only designed to operate at audio frequencies, and that the supply rails are very low noise, that will not be a problem. Moreover, the  DAC output capacitor is 2.2nF, a thousand times larger and so will filter off any small noise coupled from the supply rails.

> Also R3 can be at least <100R safely.

It potentially could be, but there should be no need for that; those regulators should be very low noise already. Let's wait until we start making noise measurements and only change this if we find a problem.

> I think additional LC filter makes sense to add for the DAC DVCC line at least to reduce the return AC current via C83, C13, C14 to the power supply that can leak to the analog side.

I don't understand what you mean here. You want to put an inductor between the DAC and the decoupling capacitors to prevent digital currents returning to ground through those caps? That's not a good idea. This supply is absolutely fine IMHO and not going to cause any problems.

Let's leave this as it is for now. We will test the prototypes carefully for noise issues and make changes are required.

> I think it makes sense add additional inductor similar to L8 and power IC1 and IC35 separately so it will reduce cross-interference.

Really not necessary. If a small amount of noise couples from the input of one switch mode supply to the input of another that won't be a problem -- a tiny amount of cross-talk between the supplies won't degrade their overall performance and they are already heavily filtered on their outputs.

> It makes sense to route the IC3 and related parts ground directly to IC18 AGND pins (star routing) but not to the gnd plane. 

No, I'd prefer not to do this. These kinds of ground plane cuts are rarely a good idea. Better to do this with a good layout that keeps the reference near the DAC and ensures that both share a low impedance ground-plane.

> Also it makes sense to have the ground plane between the front panel side cut under most of the boards and have connected only under the DAC so it will prevent parasitic digital ground noise.

No! Again, these kinds of ground cuts are generally not a good idea and usually cause more problems than they solve.

hartytp, 2018-01-16:
@a-shafir I appreciate the enthusiasm for improving these boards. But, for now I don't want to make those kinds of changes. Your approach would overcomplicate these designs by adding components to try to solve things that are unlikely to be issues.

Let's just wait until we have the next prototype. Once we've characterised that thoroughly we can see what the actual performance is and evaluate changes based on that. If the performance is bad then we can consider making a new version with some of these modifications.

a-shafir, 2018-01-16:
Ok, about the ground plane this is exactly type of the circuit where it works perfectly. The digital part of the board is quite noisy and imho there is no reasons to spread the return currents via whole board ground/power planes.
Even this looks like over-caution it certainly will not harm in this board where there is only one A-D junction but might help in some case. It cost nothing including the design. There is no reason to afraid the "analog ground" - it only helps in most of the cases. I am not suggesting a full ground separation with beads etc - this indeed can be dangerous in many cases.

Basically i see only not well under control noise source - the ground. Including the VREF signal path.

hartytp, 2018-01-16:
> Some simple numbers: reactive impedance of a trace 100mm wide 0.5oz copper is ~0.000247Ohm so 1A DC current will be 0.24mV drop. The DAC resolution on the output roughly 10V/2^16 ~= 0.15mV. So it is in the same scale. With AC it worst, roughly 10 times at 77KHz.

Firstly, this statement makes no sense. Reactive impedance of a trace at DC? You mean resistive? Where do you get 1A from? What kind of digital logic has that kind of current? I appreciate the effort to quantify your statements, but this calculation is irrelevant.

> Even this looks like over-caution it certainly will not harm in this board where there is only one A-D junction but might help. 

I disagree with this statement. It's really easy to take a perfectly good board and ruin it by splitting ground planes.

Also, note that there is a pretty aggressive output filter on these boards that will make a large difference to any high-frequency digital noise.

---

As the one funding this project, here is an executive decision on this matter to avoid wasting time on it:
unless @gkasprow says it's necessary, we will not do this for the prototypes. If there is a noise issue (beyond the expected clock feedthrough, LDAC glitches etc) with them then we will reconsider in a future revision

---

> BTW: in case of using peltier cooler it makes sense to have proper cleaning and possible conformal coating in the affected area because the it leads to condensation etc.

If you're getting condensation on the peltier here then you're doing something wrong.

hartytp, 2018-01-17:
@a-shafir re diode capacitance. Note that the regulators we're using on this board [negative LOD data sheet](http://cds.linear.com/docs/en/datasheet/1964fb.pdf) [positive LDO datasheet]() have noise floors below 100nV/rtHz. So, for higher frequencies, that's divided by 1000:1 by the output capacitor, giving a noise floor for the board below 100pV/rtHz.

At lower frequencies the supply noise is larger, but it's shunted more effectively by the 500R output resistor.

So, unless there are some large spurs on the regulator (e.g. leakage due to the SMPS) I do not expect an issue. If we see those then we will definitely add an LC filter/ferrite bead/etc.

hartytp, 2018-01-17:
@gkasprow On this subject: @a-shafir's posts did make me look again at the supply network (so, thanks for that) so I noticed that the negative LDO we're using is the version without the bypass pin. As this is quite a noise sensitive device, we should probably use the bypassable version unless there is a good reason no to (e.g. big cost difference). Why didn't we use it?

NB If we change this on Zotino, we should also change it on all other boards using this regulator (is that just sampler?).

hartytp, 2018-01-17:
@gkasprow I had more of a look at the measurement setup we were using here to look at the noise. Based on my new measurements, I currently believe that the noise @cjbe saw is related to the measurement setup and not to the DAC itself.

So, I don't think there is any need to do further testing on the v1.1 boards.

jordens, 2018-01-17:
@Hartytp Care to elaborate?

hartytp, 2018-01-17:
I think the bulk of what Chris was seeing was pickup/grounding/wiring issues.

I'm now using a small coax pigtail soldered directly to the output of the CMC on Zotino to do my probing (DAC removed, OpAmp input grounded). I'm attaching that to a low noise FFT analyzer. I measure a noise floor of something like 17nV/rtHz with lots of junk on it at higher frequencies (instrument noise floor is around 5nV/rtHz). However, AFAICT, all that high frequency junk is just pickup (I see a similar spectrum, but with at a much larger level with the board unpowered). e.g. the noise floor depends more on whether the PSU is plugged in (output off) than whether the board is actually powered.

But, these are the standard low-noise measurement issues one always sees. Doing this measurement properly isn't that hard, just a bit time consuming (battery powered pre-amps, careful wiring and grounding etc) and I don't have time to do it right now.

My feeling is that we should do this all properly once we have the v1.2 design and not spend any more time on the current boards.

hartytp, 2018-01-17:
@jordens Maybe I'll regret saying this, but I'm not too worried about the filter design, which is the only thing I can test at the moment (e.g. can't test the DACs because they die too fast) so I think it makes sense to move onto v1.2 without further delay.

The filter itself is just an OpAmp circuit. We've simulated it in SPICE and it looks good. But even if there is a problem there, I'm confident that we can make it work by changing component values/OpAmps. So, the only real danger is noise sources external to the filter, such as:
- The DACs/digital stuff: this could be a serious issue, but I can't test it at the moment.
- The SMPSs: from Sampler, we know that these SMPSs can be used in very low noise designs without issue; that design achieved 8nV/rtHz (consistent with the InAmp noise floor) and if we had to, we could live with that for this design since we will generally have a filter by the trap which can provide the final 20dB.
- External pickup: not much we can do about that apart from add external screening (the layout is already pretty good and we do already have CMCs etc).

gkasprow, 2018-01-24:
@hartytp the TEC wires holes are 1.1mm. Should be fine.
IDC contact current is usually rated 1A. I added 2 more pins and bigger IDC connector.
Do we really need 15x15mm TEC? I was thinking about 9x9 one...
The I2C extender is also routed to the DAC reset pin. I'm not sure if anyone is going to use that.
OK, let's remove cutouts and make more space for TEC and mounting
I changed the LDO to LT1964ES5-BYP#TRMPBF
There are already cut-outs around theSMPS to limit the return current

gkasprow, 2018-01-24:
It's not easy to find small heatsink with mounting holes...

hartytp, 2018-01-24:
> It's not easy to find small heatsink with mounting holes...

What do you want to do then? Glue it? Or, are there PCB mount clips we can use to hold it in place? I don't mind what you do here so long as there is some easy way of mounting it.

gkasprow, 2018-01-24:
TEC modules need quite a lot of force to work properly.
Let's use [this](https://pl.mouser.com/ProductDetail/Advanced-Thermal-Solutions/ATS-53170D-C1-R0?qs=sGAEpiMZZMttgyDkZ5WiulqfIpz7MfclvtavMunAIpw%3d) type of thing. I use bigger version of them. The yellow frame will hold Peltier module, I will add PCB-mounted hooks for the springs

hartytp, 2018-01-24:
@gkasprow That looks good. I'd be happy to go for something like that.

Please can you add a suitable TEC and heatsink to the BOM? It would be good to populate this for the test boards so that we can check it all works during prototyping. (For that I would probably use the evaulation board that Thorlabs sell, which hooks up to USB and does PID + peltier driver).

gkasprow, 2018-01-24:
I mean such jumpers that are sold as components and will manage to keep such assembly
[image]

hartytp, 2018-01-24:
@gkasprow That sounds good. I just mean that it would be good to also choose a TEC + heatsink + clip and attach it to the PCB. That checks it all fits together and also allows us to make measurements like temperature stability of the DAC more easily.

gkasprow, 2018-01-24:
sure, let's add it to the BOM. I will also make mechanical models to make sure it fits.

hartytp, 2018-01-24:
great!

gkasprow, 2018-01-25:
@jordens @hartytp we can leave I2C, then add I2C <-> UART [bridge chip ](https://www.nxp.com/docs/en/brochure/75015676.pdf)and control TEC controller from Zotino board directly

gkasprow, 2018-01-25:
@hartytp Such [TEC](https://pl.mouser.com/ProductDetail/CUI/CP60139H?qs=sGAEpiMZZMtAhbGwPnfBjWfeDQfrfxJO0OeY1drMFSAjpdSSNUqxAA%3d%3d) should do the job..

gkasprow, 2018-01-25:
Edit - I mean 10x10 modules, 15x15 are slightly too big.

gkasprow, 2018-01-25:
but 15x15 are much cheaper...

gkasprow, 2018-01-25:
OK, I will fit 15x15 as well. What about 2.6V module?

gkasprow, 2018-01-25:
This [one](https://pl.mouser.com/ProductDetail/Laird/430511-504?qs=sRTx%2ftA2SD9NM5cdxVgMYw%3d%3d) should be fine

klickverbot, 2018-01-25:
(@gkasprow: "This one" – did you mean to add a link?)

hartytp, 2018-01-25:
>but 15x15 are much cheaper...

Yes, AFIACT, 15x15 is standard so it's best to go for that if we can.

> What about 2.6V module?

Let's go for one with closer to 2.6Ohm so it works with the nice [Maxim TEC driver](https://datasheets.maximintegrated.com/en/ds/MAX8520-MAX8521.pdf) (1.5A, 4V)

> @jordens @hartytp we can leave I2C, then add I2C <-> UART bridge chip and control TEC controller from Zotino board directly

Okay, I'm fine to leave the I2C there if it doesn't make much difference to the size of the connector. Even if Thermostat supports ethernet as its main interface, having an I2C interface as well would be nice. That way, it becomes just an extension of Zotino.

I guess we can mount the TEC driver board on a separate front panel adjacent to Zotino.

jordens, 2018-01-25:
I think we can live without reset. If the power on sequence leads to a well defined state. The datasheet does not say much about the sequencing of the reset.
Since we are now looking at a TEC with heating and cooling, we can lower the thermal resistance between the DAC and the board as much as possible. The reason is to guarantee that this board will work well even without the TEC and associated logic. The only limit there is that the TEC needs to be able to heat away the all of the DAC power fluctuations times the ratio of the thermal resistances (ambient-TEC-DAC/DAC-board-ambient). That ratio should probably be above 0.5 given that the TEC can do +-2W and a cautious estimate for DAC power fluctuation of +-1W.

hartytp, 2018-01-25:
> I think we can live without reset. If the power on sequence leads to a well defined state. The datasheet does not say much about the sequencing of the reset.

FWIW, I don't recall ever using the reset pin for these DACs. IIRC, I've only ever used the SPI interface to do everything and that's always been fine.

@jordens Do you need the BUSY pin (or CLR for that matter) on the EEM connector? One other option would be to take BUSY/CLR of the EEM connector and to put RESET directly on the EEM connector (more convenient than having it on I2C anyway).

----

In any case, as @gkasprow points out, it's probably worth keeping the I2C interface there for the temperature controller. If something like Thermostat is used to control Zotino, it's nice if it can be wired as basically being an I2C extension of Zotino (gains and setpoint programmed automatically on startup by Kasli) rather than needing an extra ethernet cable.

hartytp, 2018-01-25:
> Since we are now looking at a TEC with heating and cooling, we can lower the thermal resistance between the DAC and the board as much as possible. The reason is to guarantee that this board will [...]

Agreed, that's the current plan. As @gkasprow said in a post above, he's removed the thermal island completely.

He's also planning to do a thermal simulation to check the C/W of the peltier and check that's it's something sensible (aiming for something like 1C/W).

hartytp, 2018-01-25:
@gkasprow To confirm: in the initial prototypes we're planning to get Technosystems to populate the peltier stack (including heat sinks etc), right?

gkasprow, 2018-01-25:
Yes.
I'm not sure if I manage to do the thermal simulation with TEC. I never did it before :)
I used simplified Hyperlynx tool for PCB level simulations.
For TEC simulation, I have to dig into the FloTherm package. I'll do it once we close issues that cannot wait.

hartytp, 2018-01-25:
> I'm not sure if I manage to do the thermal simulation with TEC. I never did it before :)

Can you do this by adding a 1W resistor connected to ground where the TEC should go? See how much the DAC temperature changes with and without the resistor and that tells you everything you need to know.

gkasprow, 2018-01-25:
This I can do. Maybe it's possible to specify negative power :) I will check

hartytp, 2018-01-25:
You don't need to simulate negative power; the number of C/W will be about the same for positive and negative power. So, just simulate with a resistor delivering positive power and it will be fine! We just need a rough estimate of this to check that we're in the right ball park, we're not looking for something ultra-precise.

jordens, 2018-01-25:
If your tools can't simulate that directly, we need to know two new thermal resistances. "junction-package-TEC" (from the DAC datasheet and a good number for the thermal contact of the TEC) and "TEC-ambient" (including the TEC heat sink). The third resistance "junction-board-ambient" is known and is what Greg has been simulating. Then with the efficiency P_thermal/P_electrical(delta T) of the TEC, whever wants to have a go at this can set up the thermal circuit.

gkasprow, 2018-01-25:
I can add 1W of heat source, no problem.

hartytp, 2018-01-25:
> I can add 1W of heat source, no problem.

Great! That gives enough information to check that this will work.

Remember that we only need to take our DAC's temperature fluctuations from a few degress to a few hundred mK, so this isn't a very demanding application. As a result, I'm happy ignoring things like the DAC-ambient thermal impedance (if this is an issue then we can glue insulating foam over the DAC).

jordens, 2018-01-25:
@gkasprow Just out of curiosity: The four mounting holes in each EEM PCB that have so far been unused, do they by chance fit the slots for nuts and screws in [these heat sinks](http://www.fischerelektronik.de/fileadmin/fischertemplates/download/Katalog/gehaeuse.pdf#page=120)?
[image] 

With a spacer between PCB underside and heat sink to make some clearance for SMD parts that could be a simple solution to cooling certain parts of these EEMs, like Urukul or Zotino.

gkasprow, 2018-01-25:
@jordens these heat sinks have wrong fins direction. So won't work with air flow from bottom to top. These are typical 3U enclosures where board fits inside in the inner slot and the enclosure fits between 3U rails. It needs different panel. The hole distance is 85mm.
I can't find this dimension in the datasheet. But once you mount it to the back of PCB, won't be able to fit into the rack.

gkasprow, 2018-01-25:
@hartytp 15x15mm TEC, 2.5Ohm, 7W + 12.5mm heatsink fit perfectly
[image]

gkasprow, 2018-01-25:
TEC model [CP50441](https://www.mouser.com/ds/2/670/cp50-1153086.pdf)

jordens, 2018-01-25:
Just fine for forced horizontal airflow.
And AFAICT they will fit into the rack perfectly when the pcb is mounted to them. You only need a different front panel and card guide position.

hartytp, 2018-01-25:
@gkasprow TEC and heat sink look good!

gkasprow, 2018-01-27:
layout done, fixing some minor power integrity issues

gkasprow, 2018-01-27:
Themal simulation with parameters:
[image]

[image]

gkasprow, 2018-01-27:
@hartytp DAC temp is 53 degrees without airflow
When I specify it as -0.6W instead of +400mW, the temperature drops to 42.1 degrees.
With airflow 20cm/s:
temp is 31.7 with TEC cooling and airflow and 38.5 without TEC cooling and with airflow.

gkasprow, 2018-01-27:
Temperature distribution with 20cm/s airflow
[image]

gkasprow, 2018-01-27:
Current density for 12V rail. Far below limit.
[image]

And for -12V rail. After optimisation:
[image]

a-shafir, 2018-01-27:
Can be improved:
1. C12.2 is connected so SIGGND but not AGND or ground plane.
2. J2.3 trace (and especially the via) looks weak
3. TEC1- pad clearance to SDO trace looks dangerous, can be short with hand soldering of TEC and burn the DAC.
4. Will TEC be powered from P3V3_MP? One pin of the signal cable IMHO it is not a good source for TEC power. Also it is shared power. Also it can be a long cable in some installations etc. Note: local P3V3 is also only 500mA max.

a-shafir, 2018-01-28:
@gkasprow
Do you have any estimation of the TEC current needed?
With this layout the thermal impedance of the ground plane shall be low. 
I know you was trying to isolate the DAC thermally in V1.0. But not it is not isolated at all. So it looks like the TEC will work cooling via quite low thermal impedance to the hot board.
Can you add the "source of cold" to your simulation?

[62 more replies]

closed 2018-01-28 by hartytp