sinara-hw/sinara #449: Sayma_AMC: SCANSTA dependence on MMC

jbqubit, 2017-12-18:
In https://github.com/m-labs/sinara/issues/448#issuecomment-352375182 @gkasprow pointed out 

> Anyway, the idea of connecting MMC to the JTAG switch (SCANSTA) it controls was not the best one...
It can literally cut the branch on which it sits.

The current state is that a misbehaving MMC can disable updating of MMC via JTAG. Resolution requires attaching an external JTAG cable. Is there a straightforward remedy? For example, using spare IO on FT4232 to control SCANSTA instead of MMC? For example, switching DDBUS to bit-bang mode when using JTAG and activating serial mode (MMC_CONSOLE) only when JTAG programming is done.

jbqubit, 2017-12-18:
@gkasprow said
> you can do it [update MMC flash] via USB with some xmodem protocol

Does that work in the case that the MMC is completely corrupted?

gkasprow, 2017-12-18:
@jbqubit yes, the bootloader is carved in stone and the only thing you need to upgrade the firmware is working serial port.
JTAG is yet another story and SCANSTA can be optionally configured with pin straps. But it's more convenient to use it in MMC controlled way.

sbourdeauducq, 2017-12-19:
> spare IO on FT4232

FTDI GPIO is lousy with bugs and I recommend against it (or use the counterfeit chips, which have the bugs fixed).

gkasprow, 2017-12-19:
The counterfeit chips has such feature that they get permanently locked while connected to Windows machine with recent FTDI drivers :) So FTDI recognizes and kills them :)

jbqubit, 2018-01-12:
> the bootloader is carved in stone and the only thing you need to upgrade the firmware is working serial port.

Sounds like this concern is resolved if it is indeed straightforward to upgrade firmware using serial port.

closed 2018-01-12 by jbqubit