sinara-hw/sinara #450: double front panel height of Kasli and VHDCI carrier

sbourdeauducq, 2017-12-19:
All the IDC cables installed and stacked up exceed the current height of the front panel and bump into the next card.
[image]

gkasprow, 2017-12-19:
You can mitigate it in 2 ways.
- install only lower part of IDC connector.
- add blind front panel.
I wouldn't make the 8HP panel because in many cases this would not be an issue.

dhslichter, 2017-12-19:
It looks to me like even with just the lower part of the IDC connector installed you are pretty close to the maximum width of the front panel.  What are the nominal measurements there, @gkasprow?   Is it really not worth making an alternate front panel design which is 8 HP for those who'd rather have the extra room?

gkasprow, 2017-12-19:
Well, we can do but this means that have to keep 2 version stocked...

jordens, 2017-12-19:
I'd vote for 8hp Kasli. It is usually just one or two Kasli a per rack and 4hp horizontal for wiring etc. would be appreciated by me at least.

dhslichter, 2017-12-19:
I second @jordens here, I think the 4 HP are worth it for having the wiring be a bit more manageable.  I imagine that @hartytp will want everything as narrow as possible, though.

cjbe, 2017-12-19:
My vote is on keeping it 4HP:
If we make it 4HP and one wants more space for cabling, one can add a 4HP blanking plate (which are almost free). If instead we make it 8HP one has to crack out the bandsaw when pushed for space.

As an aside, I have been using the 4HP VHDCI carrier with 8 IDCs populated, and there is enough space if one does not use the strain relief clip (although there is very little excess space...)

dtcallcock, 2017-12-19:
Can you just ship a blanking plate with each Kasli?  It would be cool if the blanking plate had some sick graphics on it.

gkasprow, 2017-12-19:
Sure, we can. There are already COTS ones, but without graphics.

jordens, 2017-12-20:
No. Seven ribbon cables on top of an IDC even without the strain relief dont fit. You'd have to unmount the filler panel each time. That's unneeded. And even with 8hp Kasli, there ist still plenty of space to fit a 32 channel NU Servo into a single crate.

jbqubit, 2018-01-12:
OK. Look like consensus is to use blank 4U panel. Closing.

jordens, 2018-01-12:
Nonsense. That was a 3:3 at best.

dhslichter, 2018-01-16:
8HP

hartytp, 2018-01-16:
Don't care either way is mine.

jordens, 2018-01-22:
Let's err on the side of usability and make Kasli 8 HP wide. The only reason to have Kasli 4 HP wide would be density. I measured the height of various IDC connectors without and even without strain relief Kasli/VHDCI would be usable with 4 EEMs max. After that you are hitting the next panel when inserting/removing. Whether that's a filler panel or not does not matter: you would have to unmount the next panel as well each time you are trying to get at Kasli. Saving space in a system where you are only using half (VHDCI) or a third (Kasli) of the capabilities seems paradoxical. Once density becomes necessary, I'd recommend looking into populating the sub-rack from both sides. Then you can happily fill a single 19"/84 HP/3 U sub-rack with 4 Kaslis, 16 Urukuls, and 8 Samplers, get and get 64 channels of nu-servo while still having 4 HP unused on each side. **And** you would be able to mount/unmount Kasli...

jbqubit, 2018-01-22:
I'm persuaded. 8HP sounds good to me. Does that tip the balance so we can close?

gkasprow, 2018-01-22:
I changed both to 8HP..

closed 2018-01-22 by gkasprow