sinara-hw/sinara #452: Booster RF PA schematic errata

dhslichter, 2017-12-27:
@thomasRF some errata for the schematics (moved from #251):

Preamp protection:
- [ ] Input SMA should appear where it's currently labeled just TP1.
- [ ] Part numbers are invalid/garbled for many ICs. Amps and power detector IC6, IC7, IC8, IC9 all have part numbers that don't seem to correspond to any real parts that Google knows about.  What are the actual part numbers for these?  
- [ ] CP2 needs a part number
- [ ] Notation should be made about the purpose and connectivity of the signals on J4 and J5, since they are routed to a separate board (PA_Channel_SUP_CTRL).
- [ ] Any critical inductors and capacitors in the amplification chain whose properties are important (particular voltage/current ratings, dielectric material for caps, core material for inductors, etc) should be called out.

Power stage:
- [ ] IC3 part number doesn't correspond to a part Google knows about (MMWX1)?
- [ ] Notations should be made for voltage/current/power ratings of passive components as appropriate.
- [ ] J3 should have a notation as to its purpose since it is connected to another board not in this schematic.

Output protection:
- [ ] CP1 needs a part number
- [ ] IC1, IC2 again have an invalid part number
- [ ] Notations need to be made about the purposes of the signals on J1, since they go to another board.

hartytp, 2017-12-27:
Adding to Daniel's points:

Control PCB: 
- [ ] Please add a power budget. This should include the maximum power that each RF amplifier module can consume from each supply rail.
  - What limits the current on the supply rails? The supplies themselves? the d-types? the PCB routing (maximum tolerable voltage drop on the PCB?)? The overall max power dissipated in the chassis?
- [ ] Power sequencing
- [ ] Table or something giving the pin-out for the RF module to control PCB D-sub connectors. This should include a description of how each pin is used each pin (e.g. "Forward power (?V/W, differential)" or whatever), as well as the max current per power rail pin, etc

RF module schematics:
- [ ] Add a power budget

hartytp, 2017-12-27:
- [ ] Pease check that all the part numbers in the Altium file are correct. For example, when I looked at the Atlium file before, it looked like the capacitance labelled on the schematic didn't always agree with the actual component part number when I opened up the component properties in Altium. This looks like a simple copy and paste error, but it's important to get it right. Similarly, *please check that the BOM matches the schematic.*

hartytp, 2017-12-27:
- [ ] Translate schematic annotations to English
- [ ] Tidy up schematic: move overview to the first page; update layout pages (currently they show many unrouted nets; etc... @gkasprow can you make this as beautiful as all the EEM schematics, please?
- [ ] Zero-index all channels

hartytp, 2017-12-27:
Oh, and:
- [ ] Please ensure that all schematic (and other design) files are well-organised in a sensible file structure, with sensible names (use the prefix "Booster")
- [ ] Schematic should have the title "Booster RF Power Amplifier" and a version number

hartytp, 2018-01-06:
- Please can you add a table to the schematic (ideally on the front page) with a brief summary of important info for the RF PA, including:
  - [ ] Gain (e.g. "Gain = 40dB ± 1.5dB")
  - [ ] 3dB small-signal bandwidth
  - [ ] Output power at 1dB compression and output power at saturation
  - [ ] Input power at 1dB saturation, and absolute max input power without damage
  - [ ] DC current draw

hartytp, 2018-04-23:
@gkasprow Please close if this has all been done.

hartytp, 2018-05-26:
@gkasprow Please can you close this issue once this is done?

NB schematic/cosmetic issues aren't crucial and can wait until after the next batch goes to production.

hartytp, 2018-06-06:
@gkasprow Now that booster v2.0 has gone to production, do you have time to finish this issue off? I'd really like to close all these Booster issues down and freeze the design/schematics.

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/Booster#49

closed 2018-08-18 by gkasprow