sinara-hw/sinara #453: Booster RF PA tests

hartytp, 2017-12-27:
Tests to do on RF PA prototypes (@gkasprow feel free to add other tests/suggest changes):
1. [x] Power Efficiency
2. [x] Input/output S-parameters (small signal and at 1dB compression)
3. [ ] Output small-signal gain and P1dB point as a function of frequency
4. [ ] Output IP3 point as a function of frequency
5. [ ] Time domain step response, 20/80% rising/falling time. Small signal and at P1dB use 200MHz reference signal or another convenient frequency.
6. [ ] Time domain step response, overshoot. Small signal and at P1dB, convenient reference frequency.
7. [ ] Only if easy to measure: phase noise for P1dB input signal at a convenient frequency (we can measure this at Oxford if easier). NB this is not part of the specification, but just good to know! 
8. [ ] Check temperature regulation, and thermal gain compensation.
9. [ ] Channel-channel cross-talk as a function of input frequency (again, small signal and at P1dB)
10. [ ] SFDR/non-harmonic spurs
11. [ ] Noise figure\far-from-carrier noise floor (again, not part of the specification, but good to know)

cjbe, 2018-01-10:
Measured cross-talk between two adjacent channels of the 2 channel prototype:
[image]
This is at ~1dB compression, and is very similar for a small-signal input

cjbe, 2018-01-10:
The only significant spurs present on the two channel prototype are at ~9-11 kHz and harmonics at ~-50 dBc. The spur frequency changes with input power. I suspect this is coupling from the switchmode, as it makes considerable acoustic noise that changes frequency with input power.
[image]

cjbe, 2018-01-10:
Phase noise, measured at 100 MHz, -10 dBm, using a Wenzel Citrine low noise source:
[image]

cjbe, 2018-01-10:
The large peak in phase noise at ~10 kHz changes with the power supply noise. Shortly after turning on the RFPA the PSU puts out a clean high-frequency tone, and the phase noise is much lower. After a few seconds the PSU starts emitting a dirty skipping noise (like a hard-disk seeking) - whenever the PSU is making this noise the phase-noise is worse.
[image]
The peak structure is harmonics of 11.7 kHz - the first few harmonics are suppressed by the phase-noise analyser peak filter.
[Here](https://github.com/m-labs/sinara/files/1619973/psu_noise.zip) is a (rather poor) audio recording of the noise going from a dirty tone to a clean tone at ~4s.

cjbe, 2018-01-10:
Further tests on the two channel prototype:

Output IP3:
~38 dBm at 80 MHz, 200 MHz, and 300 MHz

Temperature coefficient:
Negligible change (<0.1 dB) for both small signal and 1dB saturated gain taking the amplifier from 23 deg. C to ~55 deg. C. However I could not get the fans to turn on...?

hartytp, 2018-01-11:
Nice! Maybe thermal compensation isn't as important as originally anticipated...

cjbe, 2018-01-11:
Sometimes when I turn on the RFPA at the wall it gets stuck rebooting repeatedly, see [video](https://github.com/m-labs/sinara/files/1622643/Video.zip). 
Normally on power up the fans pulse once, then the channel LEDs turn on. However sometimes it gets stuck in a loop. Sometimes after a few iterations it starts up properly, other times it has continued for a few minutes before I turned it off.

It seems to do this very frequently if the power is turned off for less than ~8s. It looks like the power LED N8V0 stays lit very dimly for a long time (~15s) after turning off the power.

closed 2018-01-29 by hartytp