sinara-hw/sinara #454: Booster RF PA Cross-talk

hartytp, 2017-12-27:
The cross-talk in the current design is excellent, and far beyond the specification (or our expectations). Well done @gkasprow and @thomasRF! However, can we double check that there are no easy ways of making it a little better (particularly at the higher frequency range where it drops off quite a bit), please?
- Do you know what actually limits the cross-talk? e.g. is it the power supply rails?
- Please can you check that all signals going into or our of the RF PA have suitable filtering (common-mode and differential mode).
- e.g. check that there are CLC (or capacitor-ferrite-capacitor) filters on all power lines going into the RF amplifier modules. Filters should be on both sides of the D-type (the control PCB side and the amplifier side).
- All control/signal lines should have suitable CMCs or other RF filters.
- etc etc

Basically, @gkasprow @thomasRF can you have another look over the schematics and check that we're really doing all the easy things we can do to optimise this. I don't want to add any complex features to improve the cross-talk, just check that an extra capacitor or ferrite won't make a big difference.

Thank you!

gkasprow, 2018-01-26:
Indeed, we are missing RF chokes on the side of themain board. I will add them.

hartytp, 2018-01-27:
@gkasprow good catch!

hartytp, 2018-04-23:
@gkasprow Please close this if you're satisfied that all inputs and outputs (power, control voltages/signals, etc) from the RF amplifier modules now have suitable common-mode and differential mode filters on them, on both the main control PCB (the that the RF modules plug into) and inside the RF modules.

closed 2018-05-02 by gkasprow