sinara-hw/sinara #455: Booster RF PA cost

hartytp, 2017-12-27:
@gkasprow can you have a look over the factors that drive the cost of Booster RF PA. And (as well as letting me know what the cost will be) check that there are no easy ways that we can reduce the cost.

I'd like to keep the cost of Booster as low as reasonably possible without compromising on performance. e.g. going from individual RF modules to a single PCB would be cheaper, but would produce an unacceptable degradation of the cross-talk.

Again, not asking for any major changes here, just a quick check that there are no easy ways to get the cost lower.

Thanks!

gkasprow, 2018-01-26:
we can ged rid of some diagnostics but the cost is negligible.
All mechanics per one module is AFAIR roughly 600EUR at 2 pieces. 
It seems that there is no single component which dictates the price..

hartytp, 2018-01-27:
@gkasprow once you've looked over the design and are happy that no significant cost savings are possible without making major compromises, feel free to close this issue.

> we can ged rid of some diagnostics but the cost is negligible.

If they're helpful and don't alter the cost in a signficant way then keep them.

> All mechanics per one module is AFAIR roughly 600EUR at 2 pieces.

By "all mechanics" do you primarily mean those aluminium enclosures that house the amplifier PCBs? Can you double check the design to make sure that there is nothing we can do to simplify the mechanics and make it cheaper to manufacture without compromising on performance? (e.g. is it cheaper to screw commercially produced heatsinks onto the bottom than to mill our own ones?)

Anyway, hopefully that cost will come down a lot once we manufacture this in volume (e.g. 100 channels).

hartytp, 2018-03-16:
@gkasprow Just wanted to check that this is still on our radar.

- Before we finish the design, please can you double check that there is nothing we can do make a decent reduction in the production costs. e.g. can we simplify the mechanics in any way or replace any custom components with COST ones?
- Please can you confirm roughly what the production costs for a complete 8 channel amplifier will be, assuming the production run is for >100 RF channels? (I've had a lot of other people interested in purchasing this amplifier, but they want to know how much it will cost first!)

hartytp, 2018-03-16:
NB we should be able to use the same setup for testing Urukul as well.

gkasprow, 2018-03-16:
@hartytp such test rig would be useful for acceptance testing. I don't thing you want to manually check hundreds of channels :)

gkasprow, 2018-03-16:
@hartytp the mechanics is the least expensive part of it and it was the most complicated part to design :) We tried to use COTS heatsinks but they would have to be machined anyway so actually would make the cost higher.

hartytp, 2018-03-16:
> @hartytp such test rig would be useful for acceptance testing. I don't thing you want to manually check hundreds of channels :)

Good. Hopefully moving all our Sinara to automated testing at production time will reduce costs. I think it can be done quite easily with ARTIQ (e.g. with rectifier I can write you a very simple ARTIQ experiment to do all the testing you do on Urukul in an automated fashion).

hartytp, 2018-03-16:
> @hartytp  the mechanics is the least expensive part of it and it was the most  complicated part to design :) We tried to use COTS heatsinks but they  would have to be machined anyway so actually would make the cost higher.

Okay, well please do double check what the main factors in the cost are, and try to see if we can do anything to get the costs down. Then let's try to get a rough quote so we know what ballpark production costs will be in batches of >=100 channels.

hartytp, 2018-05-26:
I think the design is now pretty well cost optimised.

From an offline conversation about the cost of the mechanics for booster with @gkasprow:

> I paid for 2 complete sets of chassis and 16 modules roughly 1.2k EUR including tooling cost. So this will drop significantly with higher volume. Any other combinations of 2-part heat-sinks make it more costly or much more complicated in production.

So, the total mechanics cost alone in small quantities is about 600EUR/8 channels = 75EUR per channel. That's pretty good.

The design is now pretty well optimised to minimise labour costs for assembly/tuning.

closed 2018-05-26 by hartytp