sinara-hw/sinara #456: Booster RF PA design files

hartytp, 2017-12-27:
@gkasprow Please can you check that the repository has up to date versions of all the design/CAD files needed to manufacture Booster. e.g.
- Schematic PDFs
- BOM
- Gerbers
- CAD files (and mechanical drawings if you're using them) for the metal work etc
- Assembly instructions etc

gkasprow, 2018-01-26:
Tomasz is working on schematic updates. He does not use git, so I will update them once he finishes.
Mechanics source files take 230MB, but single step file is sufficient to build the enclosure, so I published it. It can also be edited when necessary with any 3D CAD tool. Mechanics was designed using Geomagic CAD tool. It is low cost (~3k EUR per perpetual license) but anyway somebody willing to produce or modify the design will use the step file.

hartytp, 2018-01-29:
@jordens @gkasprow what would you think about moving this to a separate repository (happy for this to still be hosted by M-Labs)? It seems that the CAD files for this could get quite big and I don't want to bloat the Sinara repository further. In any case, this isn't really an ARTIQ specific design...

gkasprow, 2018-01-29:
Good idea.

>

jordens, 2018-01-29:
Sure. We already have https://github.com/m-labs/rfpa-firmware. Do you want that renamed to `booster-firmware`, have `booster-mech` and `booster-circuit` in addition?

hartytp, 2018-01-29:
@jordens yes to all of the above, I think.

hartytp, 2018-04-23:
@gkasprow Ping: can we have the design files for booster, please?

gkasprow, 2018-05-02:
https://github.com/sinara-hw/Booster/releases

closed 2018-05-02 by gkasprow