sinara-hw/sinara #457: OpenOCD freezes intermittently when flashing Sayma

sbourdeauducq, 2018-01-02:
When running artiq_flash:
```
Info : sector 232 took 618 ms
Info : sector 233 took 623 ms
Info : sector 234 took 618 ms
Info : sector 235 took 624 ms
Info : sector 236 took 624 ms
Info : sector 237 took 615 ms
Info : sector 238 took 622 ms
Info : sector 239 took 621 ms
Info : sector 240 took 618 ms
Info : sector 241 took 621 ms
Info : sector 242 took 619 ms
Info : sector 243 took 621 ms
Info : sector 244 took 670 ms
erased sectors 0 through 244 on flash bank 0 in 162.251038s
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
```
...and hangs there.

After hitting Ctrl-C and retrying:
```
Info : sector 26 took 620 ms
Info : sector 27 took 619 ms
Info : sector 28 took 621 ms
Error: timeout, device still busy
Info : sector 29 took 3002 ms
Error: Sector erase failed.
Error: failed erasing sectors 0 to 244
```

sbourdeauducq, 2018-01-02:
Could be the power supply bug...

jbqubit, 2018-01-18:
@sbourdeauducq Do you frequently see this? I've never seen it on either of my AMCs.

jbqubit, 2018-03-30:
@sbourdeauducq I don't see this on my boards. If you've not seen this recently please close.

closed 2018-03-31 by sbourdeauducq