sinara-hw/sinara #459: Booster: Output power ranges

hartytp, 2018-01-11:
@gkasprow Something I wanted to run past you and your RF team:

For booster, the power efficiency is very good near saturation, but not so good for powers at around 1W-2W, which we will often use. So, I was wondering if we can do something like have  
a software option that allows different output power ranges to be set to allow the optimum power efficiency to be achieved at different output powers.

The idea would be to adjust the gate bias voltage (currently controlled by a DAC), to decrease the P1dB and gain while increasing the efficiency at lower output powers. For example, perhaps we could have a software option which selects between 3 different ranges with P1dB={1W, 2W and 4W} (and different overall gains as well). The user can then select the range that works best for their application.

Does that sound reasonable? Are there any hardware changes required to do that, or is it just software? If the temperature compensation doesn't work out (or isn't required) can we do this instead?

Just an idea, let me know your thoughts.

T

gkasprow, 2018-01-26:
@hartytp this is just software that need to be implemented
So maybe it's worth keeping the drain current measurement and gate adjustment.
We can adjust desired current of the output transistor in simple control loop.
All power conversion is done using switching regulators, so power consumption would be proportional. Control electronics consumes negligible amount of power.
The second most power hungry component are fans, but they are fully controlled by SW.

hartytp, 2018-01-27:
> @hartytp this is just software that need to be implemented
> So maybe it's worth keeping the drain current measurement and gate adjustment.

@gkasprow that sounds like a good plan. Let's leave this for now and come back to it later on once we have the hardware up and running. But, let's definitely make sure that we have all the hardware in place (e.g. drain current measurement, gate adjustment, etc)  to do this.

hartytp, 2018-01-27:
If we're going to set the gate voltage with a control loop then let's make sure that the DAC used to create it has a decent resolution and also some aggressive reconstruction filters to avoid noise on the RF output.

hartytp, 2018-01-27:
From #460...

> We can scrap the current monitoring hardware - it will simplify the things a lot. The original idea was to use the output MOSFET gate voltage control to stabilise the working point of amplifier by adjusting the gate negative level and measurement of the drain current. I don't think we will use this feature.
Since it was a prototype I installed plenty of options to test. It's easier to get rid of them then to solder something that was not foreseen originally.

Do you think it simplifies things a lot to remove the remote setting of gate voltage and remote monitoring of drain current? I'd like to keep these to allow us to change the amplifier biassing (as discussed above). But, if they add a lot of extra complexity to the design, or if they increase the cost noticeably then I'm open to removing them...

hartytp, 2018-01-29:
@gkasprow one other thing about this: if it's easy and inexpensive, we should consider replacing the potential dividers for the hardware forwards and reverse power interlocks that the board currently has with cheap I2C DACs. My thinking here is that if we do decide to lower the PA bias current to save power consumption then we may also want to adjust the maximum forward/reverse power cut-offs correspondingly.

I imagine a software setting that allows one to select between a few different output ranges, each with a different gain and P1dB. The software would then adjust the PA bias current as and set the interlock DACs appropriately.

NB: the circuit performance (matching elements) should be optimized for the output power level/biasing. We can accept a degradation in matching/gain flatness etc for other bias levels (the user gets to trade off performance for power efficiency).

closed 2018-05-26 by hartytp