sinara-hw/sinara #460: Booster interlock

hartytp, 2018-01-11:
@gkasprow the booster interlock is required to prevent damage to the amplifier itself as well as the load (which can be very expensive!), so it needs to work very robustly. It also needs to have a guaranteed maximum response time that is short enough to protect sensitive loads from damage.

If we do this in software (current plan) then the software needs to be extremely robust. e.g. needs interrupt-base watchdogs to ensure that if the software crashes the interlock will still function.

Another option would be to put some DACs and comparators on the control board and add extra circuitry to do the interlocking in analog rather than digital (we'd still want the current ADCs for logging/remote monitoring).

I don't mind which approach you take, so long as you're confident that it's really robust. I'm just curious to hear your thoughts on this!

hartytp, 2018-01-25:
@gkasprow This has been discussed a bit elsewhere, but I wanted to summarise here:
1. Booster has a hardware interlock that shuts down the input when the input power exceeds a certain (fixed) value. This value is chosen by the circuit designer to prevent damage to the amplifier, not the load.
2. Booster has a hardware interlock that shuts down the input when the input power exceeds a certain (fixed) value. This value is chosen by the circuit designer to prevent damage to the amplifier, not the load.
3. Booster has a software interlock that shuts down the input when the *output* power exceeds a certain programmable value. This value is chosen by the user to prevent damage to the load.
4. As far as I can tell (but correct me if I'm wrong) No software interlock is required to shut down the amplifier if the reverse power or VSWR exceed a certain value. The reason for this is that the amplifier is already protected from damage by reverse power by (2) and reverse power should not damage the load (only the forward power can do that). I want to keep the interlock firmware as simple and fast as possible, so I don't want to add extra VSWR interlocking in unless there is a case where it actually helps!
5. In the current firmware, all channels start off disabled (interlocks tripped) and must have the interlock manually reset to function. Please change this so that all channels start off enabled without requiring user intervention.

Please confirm that you're happy with the above plan and pass it on to the firmware designer. Thank you!

gkasprow, 2018-01-25:
I'm happy with this. This simplifies the software a lot. We already implemented mechanism that tracks the VSWR but it was not fully tested.

gkasprow, 2018-01-25:
This is essential information for @wizath (Jakub)

hartytp, 2018-01-25:
@gkasprow Great! I'm *very* keen to keep the interlock firmware as simple as possible since it's critical that this works robustly. The simpler it is, the more likely it is to work, and the easier it will be for us to check it.

So long as you're happy that the above plan is sufficient to protect the amplifier and the load then let's do it.

hartytp, 2018-01-26:
@gkasprow On my list above, I missed something off :
6. Booster has a software interlock that shuts down the interlock when either the temperature or PA voltage/current exceed a certain (fixed in firmware) value. This value is chosen by the circuit designer to prevent damage to the amplifier, not the load.

hartytp, 2018-01-26:
@gkasprow I get why we need to have an interlock on the PA the temperature (e.g. in case of fan failure or air vent blockage). But, why do we need an interlock on the PA current? What fault condition could cause the PA current/voltage to increase other than:
a. excessive input power
b. excessive reverse power
c. excessive temperature
d. component failure (should be very rare due to interlocks)

But, (a), (b) and (c) should all be prevented by the other interlocks. (d) is potentially an issue, but all the interlock does is shut the input off using the RF switch, and it's not clear that this helps if the PA chip fails.

So, I was wondering if it would be easier to scrap the software PA V/I interlock (again, the simpler we make this the less likely there are to be software bugs)? Anyway, I'm happy either way, just looking for your feedback.

hartytp, 2018-01-26:
One final thought on this: what would you think about doing interlock (3) in hardware. My thinking is this:
- We already have I2C on the control board, so adding a small, low-cost I2C DAC onto the control board is very simple and cheap. We can then use this DAC and a cheap comparator to do the output power interlock in hardware.
- This will be more robust than the software interlock (it still works even if the firmware crashes) and response faster, helping to protect delicate loads.
- Taking this interlock out of software allows us to simplify the software quite a bit. If we do this, and also remove the PA V/I interlock (see comment above) then the only interlock left in the software is the temperature interlock, which should be the least crucial.

Anyway, just an idea, but I'd be interested to hear your thoughts on the subject...

hartytp, 2018-01-26:
@gkasprow the more I think about this the more I wonder if we're not doing this correctly...
- currently we route the forward / reverse power / temperature to the main on as analog signals.
- wouldn't it make more sense to put small cheap I2C ADCs and DACs (Maybe use quads?) On the control on in the amplifier modules?
- that way all interlocking is done in hw simplifying the firmware (and being more robust and faster to reapond). The firmware just needs to program the DACs and read the ADCs for slow status reporting 
- that way we don't need to route differential analogy signals to in amps on the large control on (the one that the amplifier modules plug into). This could really help to reduce the cross talk.

What do you think?

dhslichter, 2018-01-26:
@hartytp @gkasprow I think the sensible thing is to put the comparator and an I2C DAC in each module and just have the main board be in charge of programming the I2C DAC.  No long-distance analog signal routing, and no software back-and-forth for the interlocks -- all one does is set the interlock level from the software and the rest happens in hardware on each module.  Seems to me to be the most robust.  It does require a respin, but could be done using DAC + comparator on a little board that plugs into the pin headers.  That would be my vote here...

hartytp, 2018-01-26:
@dhslichter That's my thought too. The amp modules  already have 2 pcbs. This only affects the cheap non-rf pcb so shouldn't be too much work I hope...anyway will leave decision to Greg.

dhslichter, 2018-01-26:
@hartytp I thought it was 3 PCBs...anyway, @gkasprow if you're willing I think it would be a nice, clean modification to make.  Lots of great I2C DAC options -- something like the AD5311R would provide everything you need, for example.

gkasprow, 2018-01-26:
@hartytp we wanted to have the VSWR measurement quick to be able to save the amplifier if reflected power was too high. With I2C ADCs it takes far too long to scan 8 ADCs. Reaction time would be an order of second. According to Tomasz experience it was too high.
By routing them to the CPU we cause very little delay. And from SW point of view it' simpler - the embedded ADC samples the voltage itself and transfers samples to desired place in memory. And such ADC come for free. Then these samples are used to calculate VSWR with little delay and once the threshold is exceeded, the amplifier input is disconnected.
That's a pitty that Tomasz didn't wanted to do proper discussion on github, he prefers oral conversation.
We can scrap the current monitoring hardware - it will simplify the things a lot. The original idea was to use the output MOSFET gate voltage control to stabilise the working point of amplifier by adjusting the gate negative level ad measurement of drain current. I don't think we will use this feature.
Since it was a prototype I installed plenty of options to  test. It's easier to get rid of them then to solder something that was not foreseen originally.
Another role of all these diagnostic hardware was to remotely detect what is going on when on of the channels fails.  On the other hand even if we discover that something is broken, we cannot repair it remotely :)
So I'd get rid of these electronic fuses, current monitoring and leave only channel LDOs + polymer fuses.
VSWR monitoring already is implemented. Threshold I2C DAC solution would react only to absolute reflected power, but I'm not sure if we care about VSWR at low power...
I don't think we need to stabilise the MOSFET current, single trimpot in the RF channel should do the job.

hartytp, 2018-01-27:
@gkasprow thanks for the response!

> @hartytp we wanted to have the VSWR measurement quick to be able to save the amplifier if reflected power was too high. 
> VSWR monitoring already is implemented. Threshold I2C DAC solution would react only to absolute reflected power, but I'm not sure if we care about VSWR at low power...

Okay. I'm not an expert on this, but I didn't think that it was necessary to limit the VWSR in order to protect the amplifier. I thought that independently limiting the absolute forwards and reverse power would be enough. After all, if the forward power is low then the VSWR can be very bad without any problem. 

Do you think it helps to limit the VSWR as well as the forward and reverse power? Or, is just limiting the forward and reverse power enough (this seems simpler to do).

> With I2C ADCs it takes far too long to scan 8 ADCs. Reaction time would be an order of second. According to Tomasz experience it was too high.
> By routing them to the CPU we cause very little delay. And from SW point of view it' simpler - the embedded ADC samples the voltage itself and transfers samples to desired place in memory. And such ADC comes for free. Then these samples are used to calculate VSWR with little delay and once the threshold is exceeded, the amplifier input is disconnected.

I definitely agree that one doesn't want to use the I2C ADCs as part of a software interlock. My thinking was that the I2C DACs + comparators do the interlock and the ADCs are just there for slow monitoring/status reporting (controlling the LEDs and reporting powers over ethernet).

I guess the question is what's easier/cheaper: route the powers from the amplifier modules to the control PCB as analog signals and do the interlock in software; or, do the interlock with I2C DACs and comparators on the amplifier modules (but then also add some cheap I2C ADCs for remote monitoring). I'll leave that decision to you.

hartytp, 2018-01-27:
> Another role of all these diagnostic hardware was to remotely detect what is going on when on of the channels fails. On the other hand even if we discover that something is broken, we cannot repair [...]

That's my thinking as well: why bother remotely detecting that the amplifier has failed when there is generally nothing one can do about it remotely.

dhslichter, 2018-01-29:
I agree that the VSWR is not the important number, it's the absolute reflected power you worry about.  The I2C ADCs is just so if your amplifier throws an interlock error, you can look at see what the issue is, but certainly they are not part of the loop.  You just do it in hardware thresholding relative to the I2C DAC voltage.  I think all the simplification ideas you have proposed @gkasprow sound good to me.

gkasprow, 2018-03-13:
I used AD5627R for reference setting and MCP3221A5 for INPWR readout. I also added unique ID chip.

hartytp, 2018-03-14:
@gkasprow Thanks!

To confirm, the plan for Booster is to do the interlocking in HW using the DACs and comparators, and use the ADCs for slow readout of forwards/reverse power via I2C?

That sounds good! (Reduces the amount of firmware development we need).

gkasprow, 2018-03-14:
Yes, this is what we do:
the reflected/forward power signal is already routed to the CPU so we leave as it is because the code and HW works. 
The input power is sampled by the I2C ADC
The thresholds for input power and reflected power are set by I2C DACs.

hartytp, 2018-03-14:
@gkasprow sounds good. That way, we can use either HW or SW interlocks.

hartytp, 2018-04-23:
@gkasprow please close this issue if this has all been implemented.

gkasprow, 2018-04-24:
@hartytp I managed to recreate the interlock issue.
I had to take the pcbs out of the shielded module and simulate the amplifier circuits
[image]
The yellow plot is RF switch signal generated by onboard logic
Blue on is RF module output
Violet is reset signal I apply from external generator
Green is interlock signal indicating that reflected power was exceeded. There is reflected power because I did not connect any load.
The detection circuit is damn fast - after 6ns it detects the reflected signal and after 10ns the input switch gets deactivated.
The setup:
[image]

gkasprow, 2018-04-24:
The fix is trivial  increase R1 to 10k and add 1nF 0603 capacitor to GND.
Now the sequence is correct - the reset is generated with slight delay.
[image]

gkasprow, 2018-04-24:
Here is how to apply the fix
[image]

hartytp, 2018-04-24:
@gkasprow Can you explain the issue a bit more, I'm not sure I understand.

Is the first plot showing that when you open the RF switch you get some noise due on the RF output due to the control signals? And that noise is enough to trip the interlock? Is that a problem in terms of possibly damaging sensitive loads?

I don't think I have the latest design files, so I'm not sure which resistors you've changed. What exactly are you delaying?

Anyway, great that this all works. I'll make the changes to my boards. If you can confirm about the firmware (see the other issue) then I'll check the interlock on my boards.

Thanks!

hartytp, 2018-04-24:
> The pulse I get is roughly 10V. It is probably caused by some DC jump during RF switch toggling.

If this transient is present on the RF output, should we be concerned about possible damage to sensitive loads?

> So by adding this delay I slightly prolong the time in which the latch is kept in reset state.

gkasprow, 2018-04-24:
That's very good question. It is extremely short - only a few ns. But may be enough to break down sensitive circuit, i.e. MSOFET gate. The glitch is produced by reflective switch. It has some internal capacitances that when charged produce the glitch
[image]
So either we mount reflective switch to mitigate it, use another non-reflective switch with external biasing or leave it as it is.

hartytp, 2018-04-24:
Okay, thanks for the explanation Greg!

How is the switch connected in the current design? Is RFC the output or the input? If RFC is the output, would these glitches go away?

If the glitch is only a few ns, then it should be okay. But, let's make sure it's in the documentation.

How long is the delay you added? Am I correct in thinking that this delay sets the minimum time it takes the interlock to re-trip after being reset?

hartytp, 2018-04-24:
Actually, on second thoughts:
- these glitches are nasty, let's get rid of them
- if flipping the switch around will work then let's do that
- let's not consider any switches that need external biasing as it's too much of a pain
- reflective switches are fine IMHO. The amp has P1dB=37dBm and G=40dB, so input for P1dB is -3dBm. Any source should be able to take full reverse power at that level. In any case, we need to decrease the overall gain by something like 6dB, so if we put the 6dB attenuator before the switch then the return loss is already -12dB with a reflective switch, which is pretty good!

gkasprow, 2018-04-25:
I talked with Maciej.
He will replace the switch with reflective one and add the attenuator at the input. This should solve the issue.

hartytp, 2018-04-25:
Cool, thanks Greg.

I'll add the RC to the prototype boards I have.

gkasprow, 2018-05-09:
One more thing - to enable this protection, you have to remove one jumper from upper board that at the moment keeps reset in high state.

hartytp, 2018-05-26:
Closing, as there is now an issue in the booster repository that covers all of this.

closed 2018-05-26 by hartytp