sinara-hw/sinara #462: touching MMC UART locks Sayma into LPC bootloader mode

sbourdeauducq, 2018-01-11:
Write characters on the MMC UART (e.g. E or P) then power cycle the board without unplugging the USB cable. The MMC will get stuck into bootloader mode, no matter how many times the board is power-cycled, until the USB cable is physically unplugged and replugged. This is causing quite a serious problem with our development cycles, especially when tracking down Ethernet and 1V8 issues that require the MMC UART.

gkasprow, 2018-01-11:
@sbourdeauducq what do you do with flow control lines?
I do not observe such behaviour on Windows machine.
UART control lines are used as:
- LPC CPU reset
- LPC CPU bootloader mode.
So the behaviour you describe is correct. 
Once you send some character, you toggle the control lines and next time after power cycle CPU enters bootloader mode.
This is feature that lets the user upload the firmware via UART.
I can add a DIP switch that will disable it in next revision.

jbqubit, 2018-01-11:
@sbourdeauducq Please add this type of Issue to [Milestone Sayma 1.0](https://github.com/m-labs/sinara/milestone/26).

jbqubit, 2018-01-18:
>Write characters on the MMC UART (e.g. E or P) then power cycle the board without unplugging the USB cable. The MMC will get stuck into bootloader mode

Isn't this a manifestation of https://github.com/m-labs/artiq/issues/873? If @gkasprow shorted pins 1 and 2 on T24 he may not see what you're seeing @sbourdeauducq.

gkasprow, 2018-01-19:
This is exactly what I recreated.

jbqubit, 2018-01-20:
> Write characters on the MMC UART (e.g. E or P)

Do you keep the terminal open while rebooting?

sbourdeauducq, 2018-01-20:
No. Anyway Greg's explanation makes sense, I'll do more testing later. Thanks @gkasprow.

sbourdeauducq, 2018-01-20:
You can exit the bootloader with:
```
#include <sys/ioctl.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
main()
{
  int fd;
  int RTS_flag;
  fd = open("/dev/ttyUSB_sayma-1_2", O_RDWR | O_NOCTTY);
  RTS_flag = TIOCM_RTS;
  ioctl(fd,TIOCMBIC,&RTS_flag);
  close(fd);
}
```
So yes, that's definitely what is happening.

closed 2018-01-20 by sbourdeauducq