sinara-hw/sinara #463: intermittent Sayma JTAG failure

sbourdeauducq, 2018-01-11:
Sometimes, the JTAG on Sayma locks up, even when 1.8V has not failed. This persists until the board is power-cycled, and does not require unplugging and re-plugging the USB connector.
```
Open On-Chip Debugger 0.10.0-00007-g977d9e41 (2017-11-01-19:49)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x3f not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
loaded file artiq_drtio/artiq_sayma_rtm/top.bit to pld device 0 in 3s 624329us
```

gkasprow, 2018-01-11:
It seems that it is not fault of FTDI chip because it is bus-powered.
When such condition occurs, please plug and unplug the OCD header. This will disable and enable the SCANSTA chip. Are you able to talk to the MMC in such state?

gkasprow, 2018-01-16:
@sbourdeauducq I found why it works like this.
The procedure to have JTAG working is following:
- plug the USB
- run the OCD, it will set EN_USB_JTAG line high
- turn on the Sayma. It will load SCANSTA configuration

If you disconnect USB with Sayma powered, the configuration will be kept until the next power cycle.
Generally speaking, one must run OCD before power supply is on.
I can fix this in MMC firmware - I can add a command that will enable USB JTAG on request

sbourdeauducq, 2018-01-16:
> run the OCD

What does that mean?

sbourdeauducq, 2018-01-16:
Why can't we just have USB JTAG enabled at all times? This is what we use for debugging those boards, and bugs like this are wasting our time, so better keep things as simple as possible to limit their occurence. We do not want to use the JTAG from µTCA for the time being.

hartytp, 2018-01-16:
@gkasprow Good catch. Glad we're starting to get to the bottom of these JTAG issues. Right now Sayma basically works beautifully, but there are enough small issues that each waste time that it is hard to work on.

marmeladapk, 2018-01-16:
> What does that mean? Try using JTAG with the FPGA off?

Yes, I ran your script while power supply was off.

gkasprow, 2018-01-16:
@sbourdeauducq 
You can short pins 13 and 11 of JTAG header. 2mm jumper is recommended
This will permanently enable USB JTAG mode.

hartytp, 2018-01-16:
> You can short pins 13 and 11 of JTAG header. 2mm jumper is recommended
This will permanently enable USB JTAG mode.

Nice. I'll do that on my board for the time being.

sbourdeauducq, 2018-01-17:
OK, good. Which JTAG header, the small one used to program the MMC?

gkasprow, 2018-01-17:
nope, the bigger Xilinx one.

jbqubit, 2018-01-18:
@sbourdeauducq Did this fix the problem you saw?

hartytp, 2018-01-30:
@sbourdeauducq did applying Greg's fix solve this problem? Or, does it still occur with those pins shorted?

sbourdeauducq, 2018-01-30:
> @sbourdeauducq did applying Greg's fix solve this problem?

I don't know, can you try it if you have time?

hartytp, 2018-01-30:
I haven't had any intermittent problems recently, so can't really test this :)

AFAICT, it only needs a single jumper so it should be quick for you to apply Greg's fix next time you're in the lab...

sbourdeauducq, 2018-01-30:
Occurence of this bug is pretty rare here as well. The problem is not the jumper, but power-cycling the board enough times that it becomes statistically significant.

hartytp, 2018-01-30:
Fair enough. Have you added the jumper though? If you do that, it would be interesting to know if you see any occurrences of this bug in the next few weeks of work on Sayma.

sbourdeauducq, 2018-01-30:
Ordered it, will arrive soon.

gkasprow, 2018-01-30:
@sbourdeauducq the "jumper issue" can be resolved by power cycling as well :)

hartytp, 2018-01-31:
@gkasprow 
> The procedure to have JTAG working is following:
>
>     plug the USB
>     run the OCD, it will set EN_USB_JTAG line high
>     turn on the Sayma. It will load SCANSTA configuration
>
> If you disconnect USB [...]

If we're going to keep the SCANSTA chip for Sayma v2.0 then we should make sure it actually works -- if we end up adding a jumper to enable USB JTAG permanently then we could go for a much simpler arrangement?

Did you have a chance to produce a version of the MMC firmware that doesn't have this bug yet? If not, is this something you'll be able to do soon?

gkasprow, 2018-01-31:
In next revision I will add additional control line that will load SCANSTA configuration after power on or configuration change. One can do it now but needs additional piece of wire to solder.

jbqubit, 2018-01-31:
OK. I'm moving this Issue to Sayma v2.0 since its resolved for Sayma 1.0 with jumper. @gkasprow please close when you've made the change to the hardware.

hartytp, 2018-01-31:
@gkasprow what exactly is the fix for this? Is it worth @sbourdeauducq applying it to his board instead of using the jumper? That way we can check that this will actually be fixed in Sayma v2.0

gkasprow, 2018-01-31:
I will fix it on my board to be 100% sure. MMC will have dedicated pin which is now routed as a test point. On other board it would be optional fix instead of using the jumper.
The MTCA mini chassis has xilinx virtual cable running on its MMC. We are currently testing this implementation. 
Here is the project [repository](https://github.com/helipiotr/tiva-xvc)

hartytp, 2018-01-31:
Thanks Greg!

sbourdeauducq, 2018-02-01:
> its resolved for Sayma 1.0 with jumper

Have you tested it?

sbourdeauducq, 2018-02-07:
Jumpers installed on all HKG boards.

sbourdeauducq, 2018-02-26:
There has been a good hundred power cycles since I installed the jumpers, and no unexplained JTAG failures. Closing.

hartytp, 2018-02-26:
@sbourdeauducq I'd like to keep this open for now.

IIRC, the jumper is a bit of a hack and not the long-term plan for operation of Sayma. @gkasprow correct me if I'm wrong about this, but in the new MMC firmware version which contains the Exar fixes, the JTAG should also be fixed. That should mean that the jumper is no longer required. I'd like to test that before we close the issue, otherwise this will surface again for Sayma v2.0

gkasprow, 2018-02-26:
I confirm. I added it to todolist. To make this fix happen, one needs to add piece of wire that lets control SCANSTA chip by MMC.

hartytp, 2018-05-03:
@gkasprow Can you remind me what the status of this is? Did you add the extra wire to the boards you sent out, and is the SCANSTA control added to the new MMC revision? Or is the plan to leave this until Sayma v2.0?

gkasprow, 2018-05-03:
Let's leave it for the next revision. This is not critical since it is resolved by the jumper.

hartytp, 2018-05-03:
Sounds, good.

hartytp, 2018-06-04:
Running `flterm` before `artiq_flash` seems to kill the JTAG on my board pretty reliably.

sbourdeauducq, 2018-06-04:
flterm on which port?

hartytp, 2018-06-04:
UART. Will check.

sbourdeauducq, 2018-06-04:
Well this was obviously done many times before... seems to be a problem specific to your board, or introduced by https://github.com/m-labs/misoc/commit/ba3c5b478488dcb3a086fec0f0dd8769d565d39a (though I did not notice issues on my board).

hartytp, 2018-06-04:
Misoc is py35_15+git7f63aff5

jbqubit, 2018-06-04:
I can consistently run the following two command in succession while leaving ```flterm``` connected to ARTIQ UART. No probolems. 

- ~/github/jbqubit/sinara-testing/sayma$ artiq_flash -t sayma bootloader firmware  --srcbuild ~/artiq-builds/artiq_sayma_20180530_HMC830_SAWG_updatedJESD204B_38b51282226f9
- ~/github/jbqubit/sinara-testing/sayma$ source tools/load_amcrtm.bash ~/artiq-builds/artiq_sayma_20180530_HMC830_SAWG_updatedJESD204B_38b51282226f9

hartytp, 2018-06-04:
No, I'm having an issue which only happens when I first power the board on.

jbqubit, 2018-06-04:
Just did a test
- flterm is connected to UART
- power off Sayma
- wait 10 seconds
- power on Sayma
- MiSoC Bootloader chatter appears on UART
- artiq_flash ... ; load_amcrtm.bash ...
- MiSoC Bootloader chatter appears on UART

hartytp, 2018-06-04:
Good to know. My board is playing up, and I don't have great statistics on this yet, so may just be a red herring...

hartytp, 2018-06-04:
Finally got round to soldering those two pins together and my board now seems much happier...

gkasprow, 2018-08-10:
can we close it?

sbourdeauducq, 2018-08-11:
I haven't seen JTAG issues after the fixes, so I think so.

jbqubit, 2018-08-16:
I haven't seen this in a long while. I'll close.

closed 2018-08-16 by jbqubit