sinara-hw/sinara #465: HMC830 supply

hartytp, 2018-01-11:
@gkasprow on my board, IC55 (supply for the first clock buffer in the chain) only outputs about 600mV. If I start the RTM with a clock applied then the output of this regulator is minus a few hundred mV.

As a result I'm not getting a 100MHz clock at my HMC830. This is at least one reason the HMC830 doesn't lock...

hartytp, 2018-01-11:
The regulator still only produces about 600mV when L54.

However, ... using a different RTM, the HMC830 does lock!

hartytp, 2018-01-11:
Hmm the quiescent current for the working RTM is quite different to the broken one. Not sure what the issue is here. Dodgy solder joint?

gkasprow, 2018-01-11:
What voltage do you have at its input?
I tested all boards with clock applied to the SMP inputs...

hartytp, 2018-01-11:
4V as expected.

hartytp, 2018-01-11:
The set pin was below 3V thought (2.something IIRC).

gkasprow, 2018-01-11:
And the chip is cold? If set point is OK it looks like short circuit or missing solder on output pad. Did you try to push it a bit ?

hartytp, 2018-01-11:
Yes, the chip is cold. Setpoint is not OK: it should be 3V3, but is around 2V.

My guess is a bad solder joint, I'll try to look at this tomorrow.

Sadly, the HMC830 worked the first time I ran the board, but then stopped locking. I'll investigate tomorrow.

hartytp, 2018-01-12:
Seems to be an issue with one board rather than the design, so closing. Will investigate what happened when I have time.

closed 2018-01-12 by hartytp

jbqubit, 2018-01-12:
@hartytp Can you elaborate? Did you reflow the solder and get 3V3 as expected? Any luck debugging broader HMC830 issue?

hartytp, 2018-01-12:
> Can you elaborate? Did you reflow the solder and get 3V3 as expected? 

No, not yet. This board needs to go back to WUT anyway (we are only supposed to have 1 RTM) so I'll let Greg look at it when we return it (he has better reworking tools than we have).

Point is that it hasn't occurred on any other boards, and there is no indication so far that it's an issue with the design rather than a bad solder joint on one board (may have been damaged in shipping).

> Any luck debugging broader HMC830 issue?

enjoy-digital, 2018-01-13:
@hartytp: to use 1.2GHz clock directly you just use `0*DAC_CLK_SRC_SEL` instead of `1*DAC_CLK_SRC_SEL` here: https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard_artiq/hmc830_7043.rs#L25

hartytp, 2018-01-13:
@enjoy-digital Thanks!

jbqubit, 2018-01-18:
In https://github.com/m-labs/sinara/issues/472#issuecomment-357968420 @hartytp did tests on another Sayma and observed that power was OK.

> Probed the HMC830 5V and 3V3 supplies. No nasty turn on transients (turn on cleanly over a second or two), correct voltage, and no obvious glitches when I load the RTM firmware.

> So, correct input clock, no power supply issues. I will verify the register set we're loading using an eval board (or Greg can do this using his Sayma and forth CPU). If no issues turn up there, this has to be some issue with the way we're programming the SPI from the ARTIQ firmware.

gkasprow, 2018-01-18:
Guys, are you sure that you load anything to the HMC830 chip? Can you read the ID?
It has nasty feature that when incorrectly initialised, alternative interface is enabled.
The state of the control lines during power on determine what serial interface protocol is enabled.
And once selected, you cannot change it until next power cycle.