sinara-hw/sinara #468: running ARTIQ Sayma hardware at WUT

jbqubit, 2018-01-12:
From #358

@hartytp said
> @gkasprow If you give me remote access to one of your PCs that's connected to a Sayma then I'm happy to install ARTIQ and load it onto your board for tests.

@gkasprow said
> @marmeladapk please help @hartytp to get access to our ARTIQ computer in our lab

@marmeladapk, @hartytp please do installation discussion in this Issue (vs eg email).

hartytp, 2018-01-12:
Joe, we're on it. Don't worry.

This doesn't need an issue, I'll take care of it.

jbqubit, 2018-01-12:
I know you'll get it done @hartytp. I'd like to make sure that the Q&A that goes along with bootstrapping @marmeladapk and @gkasprow understanding of how to install ARTIQ is visible. This will help M-Labs understand where the hangups are in installing the system, gaps in documentation.

jbqubit, 2018-01-12:
@marmeladapk Were you able to build .bit for the Sayma?

marmeladapk, 2018-01-14:
Yeah, although I thought that @hartytp offered to do this.

hartytp, 2018-01-14:
Will do once i get remote access to a pc

marmeladapk, 2018-01-14:
Soo, I may have jumped the gun after @jbqubit message and installed it myself. Steps I did:

1. Updated Conda <sub>(I think this point should be not recommended in [manual](https://m-labs.hk/artiq/manual-release-3/installing.html) but required for users like me - 1. I thought conda would be updated along with other system updates (TBH I'm not sure why I thought that) 2. Installation proceeded with older version of conda with message I treated as a warning, not error, but some dependencies weren't pulled).</sub>

2. Clone artiq-dev repo: `git clone --recursive https://github.com/m-labs/artiq ~/artiq-dev/artiq`

3. `cd ~/artiq-dev/artiq`

3. Created conda environment: `conda env create -f conda/artiq-dev.yaml`

4. Activated environment `source activate artiq-dev`

5. `pip install -e .`

6. (Vivado and OpenOCD were installed earlier, but I had to change path to Vivado in `miniconda3/envs/artiq-dev/lib/python3.5/site-packages/migen/build/xilinx/vivado.py` line 185)

7. `ls ./gateware/targets` to see available targets

7. `python3 -m artiq.gateware.targets.sayma_rtm`
`python3 -m artiq.gateware.targets.sayma_amc_drtio_master`

I didn't flash the board yet, I may do it on Monday if @gkasprow won't use it.

@hartytp @jordens Is this, more or less, what I should do to run Artiq and to be able to synthesize latest gateware?

Notes about installation process - I think it's quite easy, however I would save some time if I were to update conda right away at the beginning - perhaps including `source deactivate` (if you're in some env) and `conda update conda` commands to installation manual could help some users and make manual more idiotproof.

Another note: in [this part of the manual](https://m-labs.hk/artiq/manual-release-3/developing.html#artiq-anaconda-development-environment) I think that points 6 to 9 shouldn't link to "Installing ARTIQ from source" but just be copy-pasted from there. I wasn't sure, for example, if I had to export path to clang or not.

jordens, 2018-01-14:
@marmeladapk 
* We have `--gateware-toolchain-path` to specify the xilinx toolchain path. The fact that that doesn't work for the RTM bitstream is a bug.
* `openocd` should also be pulled in by `artiq-dev` that will override your other openocd. 
* Is a sentence like `Install an up-to-date conda package` be sufficient? Could you file an issue (or better a pull request with the change that would have helped you)?
* You are correct. The content of the items 6-9 should be referenced in such a way that it is absolutely cler to what section they refer. Could you file an issue?

marmeladapk, 2018-01-14:
@jordens I didn't specify `--gateware-toolchain-path`, it's not mentioned in the manual so that's not a bug.

hartytp, 2018-01-14:
> Soo, I may have jumped the gun after @jbqubit message and installed it myself. Steps I did:

@marmeladapk Great! Good for you. In the long run, it's much better if you can do this yourselves.

marmeladapk, 2018-01-15:
Should I flash sayma_amc_drtio_master and sayma_rtm to work with Artiq?

marmeladapk, 2018-01-15:
@jordens sayma_rtm target does not have --gateware-toolchain-path, should I file a bug?

jordens, 2018-01-15:
@marmeladapk The `drtio_master` bitstream is for something else. Ignore it. The `sayma_rtm` bitstream needs to be loaded manually over JTAG. @sbourdeauducq can guide you through it.

You can file an issue for the missing `--gateware-toolchain-path` and other options. Those should be there (or `sayma_rtm` shoudl be better integrated).

jordens, 2018-01-15:
@marmeladapk Or you could just give it a try and add an `ArgumentParser` using the code from the other targets as baseline. Do a pull request when you'd like us to look at it.
And if you come to our IRC channel (#m-labs @ freenode) we can help you with faster response times.

hartytp, 2018-01-16:
@marmeladapk Could you try running Sayma RTM with an external 1.2GHz clock, please?

Configuration I'm using is:
- 1.2GHz clock attached to DAC_CLK SMP of clock mezzanine (attach clock source to one SMP, terminate the other with 50R to ground)
- Change [this](https://github.com/m-labs/artiq/blob/fcc438524c38d1ac336aa8d2b857e4c2338705f3/artiq/firmware/libboard_artiq/hmc830_7043.rs#L25) line to `0*DAC_CLK_SRC_SEL);`. There should now be a 1.2GHz clock fed to the HMC7043, and a 600MHz clock fed from the HMC7043 to the DACs. (Please verify that the clock looks okay using the UFL test points and a fast scope or spectrum analyzer)
- Now, look at the DAC's outputs on a scope. There should be a sawtooth ramp on each DAC port (see the photo I posted on another issue).

My finding is that the DACs all produce a sawtooth briefly, which then cuts out presumably due to a loss of the JSED link. If that happens for you, can you check the DAC power supplies etc for glitches and do some general sanity checking to see if there are any obvious hardware reasons for this error (otherwise, it is a M-Labs software/gateware issue).

Thanks!

hartytp, 2018-01-16:
Also, @marmeladapk @gkasprow IIRC, you previously saw RF out of Sayma using a bitstream that @enjoy-digital sent you. Please can you remind me what your configuration was, and whether you saw any issues. e.g. did you get nice RF out and did the JSED links stay up? If so, this is less likely to be a hardware issue...

gkasprow, 2018-01-16:
I used Florent  binaries to test all Saymas working with 1GHz external clock applied to the SMP connectors.
The link was stable, but there is still unsolved issue with mux control line that caused some boards to loose the link. Florent attched this FPGA pin to GND and on some boards this particular pin was high, while on some of them was low.
I asked him to route it to the register and managed to control it from the software, this way I managed to make all boards working with Allaki. Jakub wrote little Python script that let me control the Allaki gain and RF switch.

enjoy-digital, 2018-01-16:
@hartytp @gkasprow: from what i remember, i was able to also get the JESD link stable after we found that one calibration pin was floating on the AD9154 and fixed it. But this was with the designs from my sayma_test repository. I'm at M-Labs office tomorrow and will test if JESD is also stable with ARTIQ.

gkasprow, 2018-01-16:
I applied this fix to all boards prior shipment

jbqubit, 2018-01-16:
> Updated Conda (I think this point should be not recommended in manual but required for users like me - 1. I thought conda would be updated along with other system updates (TBH I'm not sure why I thought that) 2. Installation proceeded with older version of conda with message I treated as a warning, not error, but some dependencies weren't pulled).

That's why the [manual says](http://m-labs.hk/artiq/manual-master/installing.html?highlight=pre%20existing#installing-the-artiq-packages) to update conda (cf https://github.com/m-labs/artiq/issues/785). 

> I didn't specify --gateware-toolchain-path, it's not mentioned in the manual so that's not a bug.

It wouldn't hurt if the manual indicated that this path is commonly needed. Just to grease the skids. 

> python3 -m artiq.gateware.targets.sayma_rtm
python3 -m artiq.gateware.targets.sayma_amc_drtio_master

That's great progress! 

> The sayma_rtm bitstream needs to be loaded manually over JTAG.

artiq_flash seems to handle flashing sayma_rtm when I checked last week. 

> Do a pull request when you'd like us to look at it.

@jordens Please let @marmeladapk focus on getting up to speed with ARTIQ on Sayma and test the hardware. Learning to edit ARTIQ source and push patches is desirable but maybe in a couple weeks. :) 

> I'm at M-Labs office tomorrow and will test if JESD is also stable with ARTIQ.

@enjoy-digital Glad that you'll be working next to @sbourdeauducq in HK. :) How long will your visit last?

hartytp, 2018-01-16:
> The link was stable, but there is still unsolved issue with mux control line that caused some boards to loose the link. Florent attched this FPGA pin to GND and on some boards this particular pin was high, while on some of them was low.

You mean the clock mux? I didn't see any issues with the clock, which was still present after the JESD link went down. While, it's possible that there was a brief glitch that I didn't catch, I don't think that was the issue in my case.

gkasprow, 2018-01-16:
@hartytp some boards had this issue, some didn't

hartytp, 2018-01-16:
Okay. Well, I'm not sure why my JESD links lost lock. Will be interesting to hear what @enjoy-digital finds in HK...

marmeladapk, 2018-01-19:
@jbqubit Right now finishing Kasli v1.1 took precedence.

@hartytp Did you get login details?

hartytp, 2018-01-19:
@marmeladapk no, not yet. FWIW, you've already done most of what I planned to do!

gkasprow, 2018-01-19:
I used [this](https://www.digikey.com/product-detail/en/advanced-thermal-solutions-inc/ATS-50350B-C1-R0/ATS1011-ND/1284725). @marmeladapk if you find suitable, let's try it.

marmeladapk, 2018-01-19:
@gkasprow Wrong issue.

jbqubit, 2018-01-19:
Try to flash sinara. Right now there's a bug in artiq_flash but follow here to see if its fixed https://github.com/m-labs/artiq/issues/898.

marmeladapk, 2018-01-29:
Crossposted from https://github.com/m-labs/artiq/issues/854, it fits better here.

@sbourdeauducq I flashed it now:

```artiq_flash --srcbuild ./misoc_standalone_sayma_amc -t sayma_amc```

```                                                                                            
MiSoC Bootloader                                                                                                 
Copyright (c) 2017 M-Labs Limited                                                                                
                                                                                                                 
Bootloader CRC passed                                                                                            
Initializing SDRAM...
Write leveling: 56 91 80 91 53 84 52 52 done
Read delays: 7:02-18 6:09-27 5:48-64 4:71-90 3:109-125 2:99-115 1:130-146 0:139-160 done
SDRAM initialized
Memory test failed (639392/1114624 words incorrect)
Halting.
```
Second time:
```
MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 58 90 81 90 56 82 52 53 done
Read delays: 7:00-20 6:04-20 5:47-63 4:69-86 3:107-129 2:99-115 1:124-141 0:153-178 done
SDRAM initialized
Memory test failed (635976/1114624 words incorrect)
Halting.
```

I have build from 25.01.

jbqubit, 2018-01-29:
I see something similar. Discuss in Issue https://github.com/m-labs/artiq/issues/908.

marmeladapk, 2018-01-30:
Crossposted from https://github.com/m-labs/artiq/issues/908#issuecomment-361462516 so I have everything in one place:

sbourdeauducq:

> Here is everything I built from the current master (with RTM bridge, RTIO and other things disabled to save compilation and RTM yak-shaving time):
> http://dl.free.fr/lAFdh3oQV
> With those binaries, I verified that SDRAM works fine on both Florent's board and Sayma-1.
> Can you try those binaries on your boards?
> @marmeladapk You can use the Ethernet TX clock phase adjustement script I posted in the RGMII issue on those binaries.
> @marmeladapk If the problem persists, can you use Sayma-2 that I shipped to you to debug Ethernet, since I didn't have SDRAM problems on that one?

marmeladapk:

> sbourdeauducq I loaded it to check if memory tests are passed:
> ```
> Bootloader CRC passed
> Initializing SDRAM...
> Write leveling: 95 130 118 133 96 125 87 88 done
> Read delays: 7:37-249 6:61-267 5:110-316 4:131-336 3:167-369 2:173-377 1:195-40e
> SDRAM initialized
> Memory test passed
> 
> Booting from flash...
> Starting firmware.
> [     0.000005s]  INFO(runtime): ARTIQ runtime starting...
> [     0.003864s]  INFO(runtime): software version 4.0.dev+516.g0edc34a9
> [     0.010126s]  INFO(runtime): gateware version 4.0.dev+516.g0edc34a9.dirty
> [     0.016910s]  INFO(runtime): log level set to INFO by default
> [     0.022630s]  INFO(runtime): UART log level set to INFO by default
> [     0.028790s]  INFO(runtime): press 'e' to erase startup and idle kernels...
> [     1.028006s]  INFO(runtime): continuing boot
> [     1.030975s]  WARN(runtime): using default MAC address 02-00-00-00-76-01; ct
> [     1.039568s]  INFO(runtime): using default IP address 192.168.1.60
> [     1.054501s]  INFO(runtime::session): accepting network sessions
> [     1.059438s]  INFO(runtime::session): running startup kernel
> [     1.064959s]  INFO(runtime::session): no startup kernel found
> [     1.070665s]  INFO(runtime::session): no connection, starting idle kernel
> [     1.077527s]  INFO(runtime::session): no idle kernel found
> [     1.084122s]  INFO(runtime::mgmt): management interface active
> [     6.274350s]  WARN(runtime): ethernet mac: rx preamble errors: 2
> [     7.357698s]  WARN(runtime): ethernet mac: rx preamble errors: 3
> [    19.820658s]  WARN(runtime): ethernet mac: rx preamble errors: 4
> [    20.128752s]  WARN(runtime): ethernet mac: rx preamble errors: 5
> [    20.888642s]  WARN(runtime): ethernet mac: rx preamble errors: 6
> ```
> So it works. I'll try the script you mentioned later.

marmeladapk:

> sbourdeauducq Should I change it in xdc in artiq/artiq_sayma/gateware/top.xdc and rebuild? Will latest artiq pass memory tests?

sbourdeauducq:

> No. Please follow the instruction in my comment: https://github.com/m-labs/artiq/issues/854#issuecomment-360497764 - you just save the script as edit_pll.tcl and run the mentioned vivado command.
> There is no bitstream rebuilding and it is a rather quick process. Nothing in the design other than the PLL phase will be changed, the routes etc. will be exactly as before, so yes memory test should be unaffected.

marmeladapk, 2018-02-01:
@sbourdeauducq How can I use artiq_flash without RTM?

sbourdeauducq, 2018-02-01:
It's not supported, but you can hack-patch it by removing the RTM FPGA detection in the OpenOCD script and adjusting the PLD device numbers.

marmeladapk, 2018-02-01:
I currently invoke it this way: `artiq_flash --srcbuild /home/pawel/Pobrane/artiq_sayma -t sayma`

sbourdeauducq, 2018-02-01:
You need to connect the RTM or edit the source of artiq_flash to hack-patch it.

marmeladapk, 2018-02-01:
@sbourdeauducq Where is this script located?

sbourdeauducq, 2018-02-01:
``python3 -c "import artiq; print(artiq.__path__[0])"``, then it's in ``frontend``.

jbqubit, 2018-02-02:
cf https://github.com/m-labs/artiq/issues/854#issuecomment-362616089
> Oops, presumably Greg's test wasn't done with Artiq. That should be the next step, I guess...

@marmeladapk Would be best to just retrieve a RTM from Creotech/Techonosystem so you can duplicate the AMC+RTM that the rest of us are using.

hartytp, 2018-02-02:
@marmeladapk I'll post you the second RTM that I have on Monday (the one with the 3V3 regulator issue).

marmeladapk, 2018-02-02:
I don't know what's the scope of @wizath 's work, however when he finishes then I could work on Sayma in MTCA crate.

marmeladapk, 2018-02-02:
> @marmeladapk Would be best to just retrieve a RTM from Creotech/Techonosystem so you can duplicate the AMC+RTM that the rest of us are using.

@gkasprow ?

gkasprow, 2018-02-02:
I will check what is easier to do. Guys at CTI also use RTM and AMC intensively...

sbourdeauducq, 2018-02-03:
@marmeladapk Just do something like this (not tested so you may have to finish it yourself):
```
--- a/artiq/frontend/artiq_flash.py
+++ b/artiq/frontend/artiq_flash.py
@@ -232,19 +232,15 @@ class ProgrammerSayma(Programmer):
             "reset_config none",
             "adapter_khz 5000",
             "transport select jtag",
-            # tap 0, pld 0
-            "source {}".format(self._transfer_script("cpld/xilinx-xc7.cfg")),
-            # tap 1, pld 1
             "set CHIP XCKU040",
             "source {}".format(self._transfer_script("cpld/xilinx-xcu.cfg")))
         self.add_flash_bank("spi0", "xcu", index=0)
         self.add_flash_bank("spi1", "xcu", index=1)
 
-        add_commands(self._script, "echo \"RTM FPGA XADC:\"", "xadc_report xc7.tap")
         add_commands(self._script, "echo \"AMC FPGA XADC:\"", "xadc_report xcu.tap")
 
     def load_proxy(self):
-        self.load(find_proxy_bitfile("bscan_spi_xcku040-sayma.bit"), pld=1)
+        self.load(find_proxy_bitfile("bscan_spi_xcku040-sayma.bit"), pld=0)
 
     def start(self):
         add_commands(self._script, "xcu_program xcu.tap")
@@ -337,10 +333,8 @@ def main():
                 programmer.write_binary(*config["firmware"], firmware_fbi)
             elif action == "load":
                 if args.target == "sayma":
-                    rtm_gateware_bit = artifact_path("rtm_gateware", "rtm.bit")
-                    programmer.load(rtm_gateware_bit, 0)
                     gateware_bit = artifact_path("gateware", "top.bit")
-                    programmer.load(gateware_bit, 1)
+                    programmer.load(gateware_bit, 0)
                 else:
                     gateware_bit = artifact_path("gateware", "top.bit")
                     programmer.load(gateware_bit, 0)
```
If your JTAG breaks completely with the RTM disconnected, you need a newer MMC firmware with the bug fixed.

marmeladapk, 2018-02-06:
@sbourdeauducq Thanks I missed pld # in load_proxy. Yesterday I tried all gatewares generated by your script, but it didn't help, still get preamble errors. Also seems that we have recreated 1,8 V bug overnight.

sbourdeauducq, 2018-02-06:
> Yesterday I tried all gatewares generated by your script, but it didn't help, still get preamble errors. 

Did you edit the script so it changes the RX, not TX phase? As it is, it does nothing to the RX.

> Also seems that we have recreated 1,8 V bug overnight.

Good. So at least we can make some progress with that bug despite inane EU customs bureaucrats (which are one of the reasons I moved to Hong Kong).

marmeladapk, 2018-02-06:
@sbourdeauducq EU customs are nothing compared to our post service ;) That's where I expect the main delay will come from.

> Did you edit the script so it changes the RX, not TX phase?

No, I missed your comment how to do it. I'll try that today.

sbourdeauducq, 2018-02-06:
> EU customs are nothing compared to our post service ;) That's where I expect the main delay will come from.

Didn't know EMS went through the regular post when sending to Poland. I'll avoid it next time. Is DHL working correctly?

>     Did you edit the script so it changes the RX, not TX phase?
> 
> No, I missed your comment how to do it. I'll try that today.

OK, then it is expected that you saw no change in the preamble errors.

hartytp, 2018-02-06:
> inane EU customs bureaucrats (which are one of the reasons I moved to Hong Kong).

You emigrated in search of laxer customs regulations?

sbourdeauducq, 2018-02-06:
That's one of many reasons, yes; wasting 2+ hours per package containing something as basic as a few KF valves or a FPGA development kit isn't fun. Anyway, back to the point.

marmeladapk, 2018-02-22:
I had some time to play with Kasli and Artiq. (notes for me to trace what I have done)

`artiq_mkfs flash_storage.img -s mac 02:02:02:02:02:01 -s ip 192.168.95.174`
`artiq_flash -f flash_storage.img -t kasli storage start`
`/artiq/examples/kasli_opticlock$ artiq_compile idle_kernel.py`

Changed device address in artiq/examples/kasli_opticlock/device_db.py

`artiq_coreconfig write -f idle_kernel idle_kernel.elf`

`artiq_flash -t kasli start` to restart board.

Played a bit with artiq_run. Will continue tommorow, after I test new batch of Kaslis. Eventually I want to be able to test most functionalities with Artiq and use Vivado only for IBERT and platform cable. I'll start with writing and reading from DIOs.

jordens, 2018-02-22:
Well done!

hartytp, 2018-02-22:
@marmeladapk Nice! You should be able to use this to test Zotino v1.1 when it arrives. Might be quicker than the dev kit.

marmeladapk, 2018-02-23:
https://github.com/marmeladapk/kasli-i2c

marmeladapk, 2018-03-05:
@jordens Did you test AD9910 on Kasli?

Currently with `"refclk": 240e6` and `"pll_n": 15` on all channels I get `PLL lock timeout`

sbourdeauducq, 2018-03-05:
@marmeladapk I did and it worked fine. https://github.com/m-labs/artiq/tree/master/artiq/examples/kasli_sysu

marmeladapk, 2018-03-05:
@sbourdeauducq `python3 -m artiq.gateware.targets.kasli -V sysu`
```
make: Entering '/home/pawel/artiq-dev/artiq/artiq_kasli/sysu/software/bootloader'
 CARGO    cargo/or1k-unknown-none/debug/libbootloader.a
env: \u201ecargo\u201d: No such file or directory
```

marmeladapk, 2018-03-05:
I added I2C device to coredevice/i2c.py (practically a copy-pasted version of PCA9548 driver):
```
class PCF8574:
    """Driver for the PCF8574 I2C bus extender.

    I2C transactions not real-time, and are performed by the CPU without
    involving RTIO.

    On the RJ45, this chip is used for selecting direction.
    """
    def __init__(self, dmgr, busno=0, address=0x7c, core_device="core"):
        self.core = dmgr.get(core_device)
        self.busno = busno
        self.address = address

    @kernel
    def set(self, directions):
        """Set direction.

        High - output, low - input

        :param directions: 2 bits of direction for all channels (0x00 - 0xFF)
        """
        i2c_start(self.busno)
        try:
            if not i2c_write(self.busno, self.address):
                raise I2CError("PCF8574 failed to ack address")
            if not i2c_write(self.busno, directions):
                raise I2CError("PCF8574 failed to ack control word")
        finally:
            i2c_stop(self.busno)

    @kernel
    def readback(self):
        i2c_start(self.busno)
        r = 0
        try:
            if not i2c_write(self.busno, self.address | 1):
                raise I2CError("PCF8574 failed to ack address")
            r = i2c_read(self.busno, False)
        finally:
            i2c_stop(self.busno)
        return r
```
Now with 
```
self.i2c_switch0.set(7)
self.i2c_rj45_dir.set(0x00)
self.i2c_switch0.set(5)
self.i2c_rj45_dir.set(0x01)
```
I'm able to set direction on RJ45 DIO.

jordens, 2018-03-05:
And check the DIP switch settings.

jordens, 2018-03-05:
@marmeladapk The PCF8574 driver looks good. It would be great if you could add that to the API documentation and submit a pull request.

hartytp, 2018-03-06:
Closing this as WUT have ARTIQ up and running (well done and thanks again @marmeladapk).

PCF8574 driver should have a separate issue if needed....

closed 2018-03-06 by hartytp