sinara-hw/sinara #470: serial numbers

jbqubit, 2018-01-12:
Sayma_AMC and Sayma_RTM v1 didn't have consistent factory-supplied serial numbers. This makes it hard to keep track of which boards have particular problems or what modifications have been made to them. 

Do we want Sinara hardware to have serial numbers at all? If so here's a proposed format __AABBBCCCC__.
- __AA__ is manufacturer number eg 01 for Technosystem
- __BBB__ is board type eg 001 for Sayma_AMC
- __CCCC__ is board serial number

The point of adding __AA__ is that if there's another manufacturer other than Technosystem that manufacturer can keep track of its own serial numbers.

jordens, 2018-01-12:
Yet Another Naming System. Great!
Just use the EUI48s or have the guts to get rid of them if you don't.

jbqubit, 2018-01-12:
Good point. I didn't realize that MAC addresses are really part of a larger name space. EIU-48 looks like it's superseded by EIU-64. IEEE issues 24-bit Organizationally Unique Identifier ([OUI](https://standards.ieee.org/develop/regauth/cid/index.html
)) for a nominal fee. That would leave 40 bits in the ARTIQ s/n space.

gkasprow, 2018-01-13:
Almost all our boards, Sayma AMC, RTM and all EEMs have EIU-48 numer assigned - there is dedicated I2C chip. Only Kasli does not have one. Such approach would need to organize database which would assign release version to the EUI number.

jbqubit, 2018-01-18:
Here's an example database. It could be populated by technosystem prior to shipment. 

https://docs.google.com/forms/d/e/1FAIpQLSf7Y3nuaTHbF7lu8rs6N51hA4LGiMhyXESg4ULpbs6Zv-qAbw/viewform?c=0&w=1

jbqubit, 2018-01-18:
Also, manufacturer ought to affix the s/n to the PCB on a sticker for human readout.

gkasprow, 2018-01-18:
@jbqubit good idea!

jbqubit, 2018-01-29:
@gkasprow Are you (and technosystem) OK with using the google form for keeping track of hardware serial numbers? I'd really like to see all hardware including prototype hardware numbered. This would entail reading out the EIU-38 on the I2C chip and printing a custom s/n per board. I guess a computer-operated label maker is needed to prevent transcription errors of these long numbers.

gkasprow, 2018-01-29:
This is standard thing that is already implemented in Creotech. Every board that leaves production is tracked and entered to the ERP system. I think we can generate report which can then later be copied to Gogole sheet. TS would have to do it manually or they would have to implement it.

jbqubit, 2018-01-29:
They dropped the ball on consistently assigning serial numbers to the Sayma boards. For the boards which did get a serial number, the Technosystem-supplied number doesn't appear to be EIU-48. To my view it's fine if Technosystem uses their existing ERP system provided they export the Sinara number's when requested.

hartytp, 2018-01-31:
@jbqubit Let's not track the serial number ourselves. If TechnoSystem are tracking it internally, that's good enough for me.