sinara-hw/sinara #472: HMC830 issues

hartytp, 2018-01-12:
@gkasprow I'm trying to get the HMC830 on Kasli up and running.

I modified the firmware to dump all registers (apart from the VCO subsystem) 3 times: once after power on reset; once after configuration; and finally after waiting 2 seconds for the VCO to lock. Result is [here](https://hastebin.com/rijuxupoke.css).

Comments:
- The VCO does not lock
- The VCO output is stuck at about 750MHz. Note that we're using the div 2 output divider, so this is consistent with the 1.5GHz minimum VCO frequency of the HMC830
- Register 0x10 always reads 0x80, which indicates that the VCO AutoCal has not completed, so it appears to be stuck for some reason.
- I tried a few sets of register values, but the main ones I tried were @enjoy-digital's ones here
- Putting in delays in a few places was the only thing that seemed to make a difference, so I wonder if there is some timing thing here (just speculation at this point)
- The board is a bit of a pain to probe due to the small vias and lack of test points. So I haven't done that.

Suggested paths forward:
1. So that this doesn't hold up getting Sayma working, @sbourdeauducq can you use an external 1.2GHz source? If you don't have one then I believe that we have a PLL eval board that we can post you (program if from a microprocessor or whatever). AFAICT, that should work fine. If you want, I can test this on Monday with out RTM.
2. @gkasprow Can you look at your RTM with the forth CPU that you have and see if the autocal also gets stuck for you. From your previous posts, it seems that it didn't (IIRC your VCO had roughly the correct output frequency). If that works, I can load your register set into my code and see if it works. That would confirm that it's not the register values themselves that are the problem.

Thanks!

hartytp, 2018-01-12:
PS Checked that the input clock looked good, but forgot to check the supply rails. Will do that once @gkasprow has checked that his fourth CPU doesn't have the same problem.

gkasprow, 2018-01-12:
I will check it on Tuesday, all HW is at the Unviersity.
The values I used are here:
0 $0 $20 spi_hmc830 .
 \ 0 $0 $00 spi_hmc830 .
0 $1 $2 spi_hmc830 .  \ chip enable by SPI 
0 $2 $2 spi_hmc830 .  \ divider = 2 
0 $6 $303CA spi_hmc830 . 
 \ 0 $9 $153FFF spi_hmc830 . 
0 $5 $1628 spi_hmc830 . 
0 $5 $60A0 spi_hmc830 . 
0 $5 $E090 spi_hmc830 . 
0 $5 $2818 spi_hmc830 . 
0 $5 $F88  spi_hmc830 . \ magic word enables PLL out
0 $5 $0 spi_hmc830 . 
0 $7 $14d spi_hmc830 . 
0 $8 $c1beff spi_hmc830 . 
0 $a $2046 spi_hmc830 . 
0 $b $7c061 spi_hmc830 . 
0 $f $81 spi_hmc830 . 
\ 0 $3 38 spi_hmc830 . 
\ 0 $4 3851ec spi_hmc830 . 
;

hartytp, 2018-01-12:
Note to self: the other thing I should do on Monday is modify my code to also dump the VCO subsystem registers. IIRC, that's controlled by a second SPI bus internal to the HMC which runs off a slower clock than the main one, so I should check that it has been programmed correctly...

gkasprow, 2018-01-12:
@hartytp I never said that my PLL was locked. I managed to get something with more or less right frequency but did not check the lock bit.

hartytp, 2018-01-12:
Sorry, that's what I meant: my PLL VCO is currently stuck at the bottom of its range (from the register dump, it seems that the autocal hangs) and outputs 750MHz. Once we have an output at roughly the right frequency, then we can worry about getting it locked.

hartytp, 2018-01-12:
Sorry, that's what I meant: my PLL VCO is stuck at the bottom of its range (from the register dump, it seems that the autocal hangs) and outputs 750MHz. Once we have an output at roughly the right frequency, then we can worry about getting it locked.

sbourdeauducq, 2018-01-13:
> So that this doesn't hold up getting Sayma working, @sbourdeauducq can you use an external 1.2GHz source? If you don't have one then I believe that we have a PLL eval board that we can post you (program if from a microprocessor or whatever).

I have two generators that can do 1.2GHz (SynthNV and a ADF4351-based Chinese gizmo). But it's not that straightforward, we actually tried that before (as soon as we noticed HMC830 issues) and could not get it to work; it would seem the clock switches have some problem. We didn't investigate in depth as we focused on the HMC830. I can have another look as the HMC830 won't cooperate.

sbourdeauducq, 2018-01-13:
> @hartytp I never said that my PLL was locked.

You said you managed to get RF out of a full Sayma system, including BaseMod/Allaki. Was that with the HMC830 bypassed?

enjoy-digital, 2018-01-13:
Just for info on the tests i did and results:
- when setting up the HMC830 in July, i was able to get the correct frequency out of the HMC830 (without checking the lock), but since JESD was not working and we were not sure that clocks were clean, i then used an external 1.2GHz clock. (on the sayma amc/rtm i have providing an external dac clock works fine).
- we later figured out with @gkasprow  that JESD was not working due to calibration pins that were floating. After fixing that i tried to test JESD with the HMC830 but we no longer able to get correct frequency out of it and don't explain why.

hartytp, 2018-01-13:
@sbourdeauducq Can you try your 1.2GHz generator again, please? If you have problems then I'll see if I can get it working on my system -- if not, I'll have a go at debugging it here.

hartytp, 2018-01-13:
@enjoy-digital Yes, I'm not sure why it stopped behaving. Mine locked the first time I tried it, but never again :(

From the register dumps I have, it looks like the autocal routine has gone into a lockup state. I've poked all registers relevant to it and can't see an obvious reason for it, which is why my suspicions are leaning towards timing issues or some kind of SPI glitch.

As I said, getting Greg to see if he can reproduce his earlier results with his fourth CPU would be a good next step.

hartytp, 2018-01-13:
@enjoy-digital one other thing I will try on Monday: disable the auto-cal and sweep through the VCO ranges in the firmware, looking for lock at each point. Hopefully, that way I can get the HMC830 to lock. One we know which VCO range to use, we can hard-code it as a temporary fix. If that works, it gives us a temporary solution, if it doesn't work it will give us some more info about exactly what's going wrong here.

hartytp, 2018-01-14:
@gkasprow Can you sanity check something for me, please?

I don't have Allaki, and I want so see some RF from the DAC. Their outputs are current sources, so don't work without a load. The appropriate load is just a 100Ohm resistor between them, right? So, I can solder this between the two SMP connectors, can't I (IIRC there are tabs on those connectors which I can solder to). I can then probe one end of the resistor with a scope to look for a signal?

jordens, 2018-01-14:
Don't bother with source side termination. You'll get RF output despite that. And Greg adds the 50r source side termination manually iirc. There is an erratum.

hartytp, 2018-01-14:
I didn't mean source-side termination, I meant that as a load -- I was planning to use a high-Z scope probe so the 100R resistor is my load.

gkasprow, 2018-01-14:
The outputs are terminated with 50Ohm close to the DAC, so you will clearly see the signal at the SMP connectors. You can connect the scope probe and U will see the signal.

hartytp, 2018-01-14:
> And Greg adds the 50r source side termination manually iirc. There is an erratum.

Sorry, I'm misunderstood you.

> The outputs are terminated with 50Ohm close to the DAC, so you will clearly see the signal at the SMP connectors. You can connect the scope probe and U will see the signal.

aah yes, now I remember. Great, I'll just stick a scope on the SMPs.

hartytp, 2018-01-15:
> Note to self: the other thing I should do on Monday is modify my code to also dump the VCO subsystem registers. IIRC, that's controlled by a second SPI bus internal to the HMC which runs off a slower clock than the main one, so I should check that it has been programmed correctly...

Can't do that. Not supported by the HMC830. :(

hartytp, 2018-01-15:
Greg,

above you posted the register set:

0 $0 $20 spi_hmc830 .
0 $1 $2 spi_hmc830 .  \ chip enable by SPI 
0 $2 $2 spi_hmc830 .  \ divider = 2 
0 $6 $303CA spi_hmc830 . 
0 $5 $1628 spi_hmc830 . 
0 $5 $60A0 spi_hmc830 . 
0 $5 $E090 spi_hmc830 . 
0 $5 $2818 spi_hmc830 . 
0 $5 $F88  spi_hmc830 . \ magic word enables PLL out
0 $5 $0 spi_hmc830 . 
0 $7 $14d spi_hmc830 . 
0 $8 $c1beff spi_hmc830 . 
0 $a $2046 spi_hmc830 . 
0 $b $7c061 spi_hmc830 . 
0 $f $81 spi_hmc830 . 

But, in #216 you posted

: config_hmc_830
0 0 $20 spi_hmc830 .
0 1 $2 spi_hmc830 .
0 2 $2 spi_hmc830 .
0 5 $1628 spi_hmc830 .
0 5 $60a0 spi_hmc830 .
0 5 $e110 spi_hmc830 .
0 5 $2818 spi_hmc830 .
0 5 $0 spi_hmc830 .
0 6 $303ca spi_hmc830 .
0 7 $14d spi_hmc830 .
0 8 $c1beff spi_hmc830 .
0 9 $153fff spi_hmc830 .
0 a $2046 spi_hmc830 .
0 b $7c061 spi_hmc830 .
0 f $81 spi_hmc830 . ;

Which should I use? FWIW, the one posted above in this issue seems to disable all output bufers for the HMC830, so I assume that's not correct?

hartytp, 2018-01-15:
Anyway, I'm going to leave this for now. Let me know when you've had another go with your forth CPU and can get RF out at approximately the correct frequency (even if it's not in lock).

Thanks!

PS the routing on the HMC830 loop filter looks pretty long, could be an issue with pickup. On the next revision we should try to make that more compact.

hartytp, 2018-01-15:
Okay, using an external 1.2GHz source, I do get a ramp out of the DACs on Sayma. It looses lock after about 1s, but it's a start!

hartytp, 2018-01-15:
@gkasprow My HMC830 eval board has arrived, so if you have problems getting your HMC830 on Sayma_RTM up and running I'll have a look at that. Just let me know!

hartytp, 2018-01-16:
After one of those frustrating JTAG bugs cost me quite a while (JTAG become non-responsive, but a lot of randomly plugging and unplugging things and power cycling everything a few times eventually fixed it) my Sayma is back up and running. So...drum roll...

RF from Sayma DAC channel 6 (zero-indexed) using ARTIQ:
[image]

gkasprow, 2018-01-16:
@hartytp do you use external JTAG cable or USB-JTAG?

hartytp, 2018-01-16:
USB

gkasprow, 2018-01-16:
Any idea how to recreate this issue?

hartytp, 2018-01-16:
Nope, sorry (if I did I would have filed an issue).

Might well have been one of the other known issues (perhaps I hit ctrl-C during an OpenOCD run or something like that). Right now, there are quite a few known issues with USB-JTAG that aren't fixed on my board (like plugging and unplugging cables causing issues). I think it will be hard to track down new issues until those ones are fixed.

hartytp, 2018-01-16:
@gkasprow For the HMC830, the datasheet says that the NC pins were connected to ground for all characterization tests. We leave them floating. But, on the eval board, they are connected to all kinds of things see [here](http://www.analog.com/media/en/technical-documentation/evaluation-documentation/hmc830lp6ge_eval_pcb_schematic.pdf)

What should we do? I vote that in the next iteration we copy the eval board.

gkasprow, 2018-01-16:
the datasheet also says: The pins are not connected internally; however, all data shown herein 
was measured with these pins connected to RF/DC ground externally

gkasprow, 2018-01-16:
I will follow the datasheet.

hartytp, 2018-01-16:
@gkasprow Right, that's my point.

But, if that's the case, why aren't they connected to ground in the eval board? Surely that's used for testing.

Personally, I would prefer to follow the eval board instead of the datasheet. That datasheet is a bit ambiguous in places and if we have issues then it's hard to understand why. If we follow the eval board then we can always check the performance of our circuit against the eval board.

hartytp, 2018-01-16:
@gkasprow I think that unplugging the USB cable with the board powered and a terminal open may be the source of my JTAG issues.

gkasprow, 2018-01-16:
USB is bus powered. The question is if the USB port on PC side locks.
I observed such behaviour on our Linux machine and had to reboot it to resume operation.

hartytp, 2018-01-16:
Rebooting didn't help in this case...

hartytp, 2018-01-16:
Probed the HMC830 5V and 3V3 supplies. No nasty turn on transients (turn on cleanly over a second or two), correct voltage, and no obvious glitches when I load the RTM firmware.

So, correct input clock, no power supply issues. I will verify the register set we're loading using an eval board (or Greg can do this using his Sayma and forth CPU). If no issues turn up there, this has to be some issue with the way we're programming the SPI from the ARTIQ firmware.

sbourdeauducq, 2018-01-16:
Intermittent JTAG breakage is https://github.com/m-labs/sinara/issues/463

hartytp, 2018-01-16:
This seems like a different issue, since it was not fixed by power cycling the board (which 463 was AFAICT).

jbqubit, 2018-01-18:
@hartytp Thanks for your pounding on this! Circling the wagons there's some tests @hartytp asked for. 

- [ ] @sbourdeauducq Can you try your 1.2GHz generator again, please? If you have problems then I'll see if I can get it working on my system -- if not, I'll have a go at debugging it here.

- [ ] @gkasprow As I said, getting Greg to see if he can reproduce his earlier results with his fourth CPU would be a good next step. [look for VCO AutoCal lock up]

- [ ] @enjoy-digital One other thing I will try on Monday: disable the auto-cal and sweep through the VCO ranges in the firmware, looking for lock at each point. Hopefully, that way I can get the HMC830 to lock. One we know which VCO range to use, we can hard-code it as a temporary fix. If that works, it gives us a temporary solution, if it doesn't work it will give us some more info about exactly what's going wrong here.

jbqubit, 2018-01-18:
@gkasprow I moved your comment here from #465 which has to do with power supply. Since there's no problem with HMC830 power supply that Issue is closed. 

> Guys, are you sure that you load anything to the HMC830 chip? Can you read the ID?
It has nasty feature that when incorrectly initialised, alternative interface is enabled.
The state of the control lines during power on determine what serial interface protocol is enabled.
And once selected, you cannot change it until next power cycle.

jbqubit, 2018-01-18:
@gkasprow Did you look for VCO AutoCal lock up?

hartytp, 2018-01-19:
@gkasprow 

> Guys, are you sure that you load anything to the HMC830 chip? Can you read the ID?
It has nasty feature that when incorrectly initialised, alternative interface is enabled.
The state of the control lines during power on determine what serial interface protocol is enabled.
And once selected, you cannot change it until next power cycle.

Yes. See the register dump I posted above in this issue. I did the following:
- Software reset of PLL
- Dump all PLL registers to the UART. Check that the values we read out match the data sheet defaults
- Program PLL with our register set
- Dump registers. Check that they match the programmed values
- Wait 2s for PLL to lock (it does not lock)
- Dump registers again. See that the VCO is stuck on the lowest range setting and that the auto cal is still in progress (see register 0x10).
- Check the PLL output with a scope and confirm that it is at 750MHz=1.5GHz/2 which is consistent with the VCO being on its lowest range setting.

From the above, it seems that the SPI transactions work.

sbourdeauducq, 2018-01-19:
> They didn't dump the PLL registers

Let's just add some code that dumps PLL registers to the log when the HMC830 fails to lock. @hartytp do you have a Rust routine that does such a dump already?

hartytp, 2018-01-19:
@sbourdeauducq the current code I have is a bit of a hacked mess. I'll produce something and submit a PR.

hartytp, 2018-01-19:
m-labs/artiq#901

hartytp, 2018-01-30:
Updates:
- On the RTM I have, the first time I loaded ARTIQ the firmware reported lock via the UART although I didn't have a chance to verify that with a scope. I have not seen this board report lock subsequently using any firmware. This seems to be different to @sbourdeauducq's experience where his boards lock sometimes.
- Weida has written a simple SPI core and used it to program an HMC830 eval board from an FPGA eval board. That worked 100% out of 10+ attempts. We loaded this same firmware onto Sayma RTM and it didn't lock once. We did some basic sanity checking (check that there was an SPI clock at the HMC830 etc) and didn't see anything obviously wrong.

Potential next steps:
- @gkasprow if you have code that gets the HMC830 to lock, could you send it to me please? If that doesn't work here then maybe there is an issue with my hardware (I have two RTMs atm and one is not usable since one of the 3V3 regs doesn't seem to work, breaking the clock path. I haven't investigated why or tried replacing components yet).
- We will try re-routing the SPI pins to a digital connector on Sayma RTM and hooking that up to the HMC830 eval board. If that locks successfully, it rules out anything to do with the FPGA as being the problem.
- @gkasprow @sbourdeauducq if we send you Weida's binaries that lock the HMC830 would you like to try them out? NB there is no UART etc (it's really basic code) so you need to check the LD_SDO pin with a scope/DMM (again, I'm kicking myself for a lack of test points here).

sbourdeauducq, 2018-01-30:
> NB there is no UART etc (it's really basic code) so you need to check the LD_SDO pin with a scope/DMM (again, I'm kicking myself for a lack of test points here).

Was Weida's design written with Migen? If so, it would be straightforward to add an UART and microscope to it to check LD_SDO status (including on remote boards). If not, it is possible to encapsulate it, but it's a bit more work.

hartytp, 2018-01-30:
@sbourdeauducq no, it wasn't. Verilog IIRC.

gkasprow, 2018-01-30:
@hartytp The code I use relies on Python enviorement that loads the code to FORTH CPU via UART.
Are you sure that register content I pulished is not sufficient?
The FPGA binary is here
[sayma_rtm_tester.zip](https://github.com/m-labs/sinara/files/1677359/sayma_rtm_tester.zip)
to access RTM from UART, you need to load the AMC FPGA binary that simply redirects UART to the RTM FPGA 
[sayma_amc_tester.zip](https://github.com/m-labs/sinara/files/1677361/sayma_amc_tester.zip)
And then unzip the Python  enviorement
[Sayma_RTM_tester.zip](https://github.com/m-labs/sinara/files/1677367/Sayma_RTM_tester.zip)

In the script start_FORTH27.bat change COM number that matches FTDI and run it. In my case it was port COM27
then paste this code to the terminal
16 base ! \ switches to hex mode
 800  OUT0_SET_REG io! \ ref_clk_src_sel_o high
1000  OUT0_CLR_REG io! \ ref_lo_clk_sel_o low
2000  OUT0_CLR_REG io! \ dac_clk_src_sel_o low
4000 OUT0_SET_REG io! \ ref_clk_src_ext_sel_o high
and run script
config_hmc_830

gkasprow, 2018-01-30:
are you sure you switched it to base 16 ?

gkasprow, 2018-01-30:
the FORTH machine is both in AMC and RTM :) Try with other COM port

hartytp, 2018-01-30:
yes.

> '>16 base ! \ switches to hex mode'
>  'ok'
>'>800 OUT0_SET_REG io! \ ref_clk_src_sel_o high'
> 'error: FFFFFFF3 +'

hartytp, 2018-01-30:
@gkasprow yes, I loaded both FPGAs. Which UART do you want me to use? I'm using the second one atm. When it first boots, I see...

 . start_FORTH.sh
hi
Contacting... established
'\\ The code below is written by Wojciech M. Zabolotny'
'\\ ( wzab01<at>gmail.com or wzab<at>ise.pw.edu.pl )'
'\\ It is available as PUBLIC DOMAIN or under Creative Commons CC0 License'
'\\'
'\\ In routines below I mark double cell opperands as x0 (LSW) x1 (MSW)'
'\\ The quadruple cell values are marked as x0 x1 x2 x3'
'\\ In case of calculated double cell results, they are described as (e.g. for a0*b1):'
'\\ a0*b1 (MSW) . (LSW)'
'\\ All values are unsigned'
'decimal'
''
'4 cells buffer: UDres'
''
'\\ Find the value of the most significant bit in the cell'
'-1 dup 1 rshift xor'
''
'constant cell_msb'
'\\ Unfortunately we have only d< operator, which cant be used in the multiplication'
'\\ Below is the definition of the unsigned double compare'
': ud< ( a0 a1 b0 b1 -- flag )'
'    \\ If MSW are equal, check the LSW'
'    rot ( a0 b0 b1 a1 )'
'    over over = if'
'        \\ MSWs are equal, so compare LSWs'
'\tdrop drop u<'
'    else'
'        \\ MSWs are not equal, so their comparison produces the result'
'\tu> >r'
'\tdrop drop r>'
'    then'
';'
''
'\\ Procedure below compares multiple precision buffers'
': un< ( adr1 adr2 n )'
'    0 swap ( adr1 adr2 flag n ) \\ Added 0 on stack to improve loop organization'
'    0 do'
'\t( adr1 adr2 flag)'
'\tdrop ( adr1 adr2 )'
'\tover @ over @ u> if'
'\t    0 leave'
'\tthen\t  '
'\tover @ over @ u< dup'
'\tif'
'\t    leave'
'\tthen'
'\t( adr1 adr2 flag)'
'\trot 1 cells +'
'\trot 1 cells +'
'\trot'
'\t\\ Addresses are updated'
'    loop'
'    -rot 2drop \\ Drop addresses'
';'
'\\ Procedure below subtracts multiple precision buffer pointed by adr2 from buffer pointed by adr1'
'\\ To avoid problems with lack of borrow bit, we use double precision operations'
': un- ( adr1 adr2 n )'
'    \\ Adjust buffers to point to LSW'
'    swap over 1- cells + ( adr1 n adr2[LSW])'
'    -rot swap over 1- cells + ( adr2[LSW] n adr1[LSW])'
'    -rot ( adr1[LSW] adr2[LSW] n)'
'    1 swap ( adr1[LSW] adr2[LSW] borrow_flag n)'
'    0 do'
'\t( adr1 adr2 borrow_flag)'
'\tover @ -1 xor 0  ( adr1 adr2 borrow_flag -a1 .)'
'\trot 0 ( adr1 adr2 a1 . borrow_flag .)'
[... 238 lines cut ...]
'\tthen'
'\tI2C_REGS 4 + io!'
'\tbegin '
'\t    I2C_REGS 4 + io@'
'\t    dup 2 and'
'\twhile'
'\t\tdrop '
'\trepeat'
'\t( dtaptr num stat )'
'\t\\ We have the status on the stack, but first we store the number of received bytes and the bytes'
'\trot ( stat num dtaptr )'
'\ti over c! '
'        dup i +  ( stat cnt dtaptr dest )'
'\tI2C_REGS 3 + io@  swap c!'
'\t-rot'
'\t128 and if'
'\t    \\ NACK - no more data'
'\t    leave'
'\tthen'
'    loop'
'    drop'
'    drop'
';'
'    '
 in ['.', '../', '../../common']

hartytp, 2018-01-30:
I'm running this on linux, and just ran 'python shell.py -h /dev/ttyUSB9 -p ../ -p ../../common  qmath.fs i2c.fs sayma.fs'

hartytp, 2018-01-30:
Okay, UART3 works...

hartytp, 2018-01-30:
>and run script
config_hmc_830

How?

I tried:

Loaded 435 words
>16 base ! \ switches to hex mode
  ok
>800 OUT0_SET_REG io! \ ref_clk_src_sel_o high
  ok
>1000 OUT0_CLR_REG io! \ ref_lo_clk_sel_o low
  ok
>2000 OUT0_CLR_REG io! \ dac_clk_src_sel_o low
  ok
>4000 OUT0_SET_REG io! \ ref_clk_src_ext_sel_o high
  ok
>config_hmc_830
Traceback (most recent call last):
  File "shell.py", line 63, in <module>
    swapforth.main(TetheredJ1b)
  File "../original/shell/swapforth.py", line 385, in main
    r.shell()
  File "../original/shell/swapforth.py", line 339, in shell
    self.shellcmd(cmd)
  File "../original/shell/swapforth.py", line 294, in shellcmd
    self.interactive_command(cmd)
  File "../original/shell/swapforth.py", line 120, in interactive_command
    c = ser.read(max(1, ser.inWaiting())).decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 0: invalid start byte

hartytp, 2018-01-30:
hmm...tried again and now it just hangs after `config_hmc_830` anything else I should be doing @gkasprow ?

gkasprow, 2018-01-30:
oh, it seems that sayma script was not loaded byc the *bat script

gkasprow, 2018-01-30:
sayma.fs

gkasprow, 2018-01-30:
U need at least i2c.fs sayma.fs

gkasprow, 2018-01-30:
i2c.fs was loaded but sayma.fs wasnt

[227 more replies]

closed 2018-06-20 by marmeladapk