sinara-hw/sinara #474: Sayma: clock signal integrity

hartytp, 2018-01-15:
1. Loop filter layout for the HMC830 forms quite a big loop, this is likely to be bad from an EMI perspective.

hartytp, 2018-01-15:
2. My seutp is:
- Synth at +6dBm, 1.2GHz attached to one of the clock mezzanine DAC_CLOCK SMPs. The other one is 50R terminated.
- Set mux switches to feed this signal into the HMC7043.
- Look at j59 on spectrum analyser. Probing with a short SMA pig tail that's soldered on to the PCB. 
- Clock signal looks worse than I would have expected...

[image]
my setup

[image]
my synth on a spectrum analyzer

[image]
J59 on same SA.

@gkasprow Thoughts?

hartytp, 2018-01-15:
I suspect that's not enough to prevent the JSED links from working, but it's not ideal!

hartytp, 2018-01-15:
> Did you rip away the shielding can to get to the SPI bus?

...yes...and some of the ground plane underneath....

> The proximity of your inner conductor to the large power supply islands is might also be a problem.

Yes, this might just be pickup in the test point. 1 sec

hartytp, 2018-01-15:
Okay, this is the same measurement, but looking at J60 (HMC7043 output). NB I left this also connected to GTP_CLK_IN, but AFAICT the HMC7043 can drive 25R okay.

[image]

That seems pretty good, so maybe the above was just a bit of pickup in my pigtail.

hartytp, 2018-01-15:
So, when using an external 1.2GHz source, the HMC7043 produces a good 600MHz clock for the DACs. No major problems here AFAICT!

hartytp, 2018-01-15:
@gkasprow Something to consider for Sayma v2.0: if we do end up removing the clock mezzanine and simplifying the clocking, there may be room to move the HMC7043 + HMC830 onto the top side of the board. That could be really helpful IMHO: clock chips on same side of PCB as connectors makes it easier to work on the board etc.

gkasprow, 2018-01-17:
@hartytp to get rid of the shield can, just use hot air gun, it is really easy to desolder such massive component :)

hartytp, 2018-01-17:
That approach lacks therapeutic value.

hartytp, 2018-01-17:
:)

jbqubit, 2018-01-18:
-49 dBc ought not to cause locking problems. @hartytp Did you look very far from carrier?