sinara-hw/sinara #475: Sayma in uTCA: loss of JTAG

jbqubit, 2018-01-16:
@sbourdeauducq and I both see a problem with Sayma in uTCA rack. Take a board that works fine on the desktop powered via external supply and put it into uTCA rack. JTAG over USB stops working. @gkasprow do you see this?

gkasprow, 2018-01-16:
@hartytp is the blue LED blinking on the front panel? Did you enable the power supply in the crate?

hartytp, 2018-01-16:
I think you mean @jbqubit

jbqubit, 2018-01-16:
@gkasprow asked
> is the blue LED blinking on the front panel

The power for the AMC slot is enabled and the MCH power led is blinking green. When I run Sayma_AMC and Sayma_RTM on the desktop in my lab using stand-alone power supply I observe 3.3 A current. In uTCA crate the current is only 1.85 A ([as reported by MCH](https://github.com/m-labs/sinara/issues/371#issuecomment-358113664)).

jbqubit, 2018-01-16:
And the blue LED is not blinking.

gkasprow, 2018-01-16:
@jbqubit can you check if LEDs on Sayma AMC, close to the DC/DC converter are ON?
The same with RTM LEDs.
It looks like the RTM is not enabled. Maybe it is not fully plugged?

gkasprow, 2018-01-16:
anyway, not initialised Sayma should consume roughly 1.9..2A

jbqubit, 2018-01-17:
- Sayma AMC power LEDs are not on. 
- Sayma RTM power LEDs are not on.

gkasprow, 2018-01-17:
so if the power LEDs are not on, so who consumes that much power...I will check it with my crate.

jbqubit, 2018-01-17:
Were you able to reproduce @gkasprow?

jbqubit, 2018-01-18:
@gkasprow Were you able to reproduce this?

jbqubit, 2018-01-18:
Changed name to "Sayma in uTCA: no power" as this better reflects the underlying problem.

jbqubit, 2018-01-19:
I've shorted pins 13 and 11 of JTAG header as in https://github.com/m-labs/sinara/issues/463. But no improvement.

sbourdeauducq, 2018-01-23:
@gkasprow Have you reproduced/looked into this issue? Ken Brown is asking when the boards will work in µTCA.

gkasprow, 2018-01-24:
We finally managed to collect all required HW in the lab. NAT MCH and the supply were lent to other group. Jakub is working on it right now trying to make OpenMMC running on Sayma. So expect progress very soon.
[image]

sbourdeauducq, 2018-01-24:
Porting OpenMMC in a short time sounds a bit ambitious. Maybe there is a quick hack that can be done to get the power supply running?

gkasprow, 2018-01-24:
We port only necessary things, power supply and sensor. All remaining stuff is not really useful for ARTIQ.
The original processor for which it was developed differs mainly by package.

gkasprow, 2018-01-24:
Jakub ported the MMC to new CPU, tomorrow we will try to make it running.

gkasprow, 2018-01-25:
MMC works, we still need to resolve some minor issues, but the MMC gets power. Hot swap also works. Tested with NAT and Vadatech MCH.

jbqubit, 2018-01-25:
Does Sayma AMC + RTM also power up? Can you see startup ARTIQ/MiSoC UART messages on AMC?

gkasprow, 2018-01-25:
Didn't check yet with RTM. The AMC wakes up, all supplies are on, start-up messages are on the USB UART.

gkasprow, 2018-01-25:
RTM wakes up as well.

jbqubit, 2018-01-26:
Congrats on getting the OpenOCD up and running on Sayma AMC! That's a big step toward the OpenMMC [milestone](https://github.com/m-labs/sinara/milestone/27). 

As of this Issue now there's two versions of MMC software. One contains many months of configuration tweaks that got Sayma working (Forth). The second version now only contains these uTCA-power configuration tweaks (OpenMMC). Can you back port the uTCA-power tweaks to the Forth code so we can commence with testing with rack? I agree with @sbourdeauducq that it's better to reduce dependency on OpenOCD port for purposes of [Sayma v1 milestone](https://github.com/m-labs/sinara/milestone/26).

jbqubit, 2018-01-26:
Once added to Forth MMC firmware and confirmed to work by @sbourdeauducq let's close this Issue.

sbourdeauducq, 2018-01-26:
OpenOCD? Forth? This has close to nothing to do with the MMC.

gkasprow, 2018-01-26:
Forth was used to play with FPG peripherals.
Jakub already integrated my init procedures with OpenMMC, now he is testing them

jbqubit, 2018-01-26:
> OpenOCD? Forth?
Typo. I mean OpenOCD. 

OK. Please use OpenMMC [milestone](https://github.com/m-labs/sinara/milestone/27) and related Issues to track progress of Jakub's work.

dhslichter, 2018-01-26:
>OpenOCD? Forth?
Typo. I mean OpenOCD.

You mean OpenMMC? :)

jbqubit, 2018-01-29:
OK. To test I need to know how to program OpenMMC firmware. cf https://github.com/m-labs/sinara/issues/428

gkasprow, 2018-01-29:
@jbqubit are you able to get the FlashMagic software running on your local machine?

jbqubit, 2018-01-29:
Yes, I have FlashMagic on a Windows machine.

gkasprow, 2018-01-29:
OK.
Short pin 2 of T23 with pin 2 of T24. This will enable reset over USB
Please connect USB to the computer.
In device manager/ Ports (COM&LPT) check which USB port is FTDI port D
Run FlashMagic tool, select correct COM port.
I will place *hex file here so you could test if it works.

gkasprow, 2018-01-29:
set LPC1776, 8MHz oscillator, select hex file and press start.

jbqubit, 2018-01-29:
Which .hex file?

gkasprow, 2018-01-29:
I'm trying to generate it. I have it configured on another computer. I need to get hex file from axf file.

gkasprow, 2018-01-29:
OK, this should work:
to convert axf to FlashMagic compatible format I used this command. I post it for further reference.
```
arm-none-eabi-objcopy -O binary "${BuildArtifactFileName}" "${BuildArtifactFileBaseName}.bin"
checksum -v "${BuildArtifactFileBaseName}.bin"
arm-none-eabi-objcopy -I binary "${BuildArtifactFileBaseName}.bin" -O ihex "${BuildArtifactFileBaseName}.hex"
```
Please use this file

sbourdeauducq, 2018-01-30:
You need only one command. https://github.com/m-labs/sinara/issues/448#issuecomment-359141733

jbqubit, 2018-03-30:
I received replacement AMC from @gkasprow this week. Board powers on fine on bench top when connected to RTM. The Blue front panel LED is not blinking. @gkasprow says this is proper behavior of board in uTCA crate -- MMC firmware on this AMC is now more standards compliant. OK good. 

@gkasprow @wizath But...... For this AMC + RTM in uTCA rack... I see fast flashing of blue, red, green lights on front panel. No power LEDS are illuminated on RTM or AMC. ​

jbqubit, 2018-04-02:
@gkasprow @wizath On your board can you ping Sayma using MCH as switch?

gkasprow, 2018-04-05:
@jbqubit we have troubles getting ARTIQ running on the only AMC board we have. On Tuesday I will receive your board so we will have more chances to debug it.
To diagnose Ethernet the list of the things to check are:
- ping
- LINK UP LED
- GMIICR register content in the PHY dump (E character)
If the GMIICR register is zero or all ones, it means that it was not initialised properly and there is no chance for Ethernet to work.
@wizath has new MMC with some issues solved so try it.

If the power doesn't work in the crate, just check what MMC and MCH says

sbourdeauducq, 2018-04-30:
I flashed your openMMC_exar.axf and updated the Exar chips. The Sayma appears to work fine outside the µTCA crate.
But now the whole µTCA crate dies (no more MCH response to TCP/IP) when a Sayma is inserted and the hotswap switch closed, with the following final MCH messages:
```
ScanPM: ...
no PM for fru_id=51
AMC2(6): Handle=0x01 - closed
no PM for fru_id=52
LSHM(0): FRU 6 sensor 38 LUN 0 'HS 006 AMC2' hotswap M1->M2
LSHM(0): FRU 6 sensor 38 LUN 0 'HS 006 AMC2' hotswap M1->M2
Activation: modules are ready
Activation: all modules ready, Allowance Period (20 sec) stopped - continue with module startup !
Activation: starting AMC2
bp_fru: power chan=6, current limit=8.0 A
PwrAllocate(fru 6 chan 6): allocating 8.0 A granted (available 44.5 A)
pm_EnablePayloadPwr(50,6):
PM(50,6):Enable PP (primSite:1 secSite:0)
pm_PP_good(6): Failure: power channel state=0x0b
pm_PP_good(6): Failure: power channel state=0x0b
```

sbourdeauducq, 2018-04-30:
And of course another board produces different behavior:
```
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): REQ(I2C=0x74) failed on bus 2 - no ACK
R(6,2,2)pm_processEvent PM(50): channel(1) new state=0x5b
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
ipmiMsgSender(6): RSP(I2C=0x74) failed on bus 2 result -8
```
...and no MCH dying.

sbourdeauducq, 2018-04-30:
Sometimes I get yet another error:
```
Activation: modules are ready
Activation: all modules ready, Allowance Period (20 sec) stopped - continue with module startup !
Activation: starting AMC3
bp_fru: power chan=7, current limit=8.0 A
PwrAllocate(fru 7 chan 7): allocating 8.0 A granted (available 45.5 A)
pm_EnablePayloadPwr(50,7):
.CU1(40): FRU active (state M4)
PM(50,7):Enable PP (primSite:1 secSite:0)
pm_PP_good(7): Failure: power channel state=0x0b
pm_PP_good(7): Failure: power channel state=0x0b
CU1(40)  fan speed properties:
  minimum speed level:    0x00
  maximum speed level:    0x0f
  normal operating level: 0x03
  fan tray properties:    0x80
PM(50,7): new state=0x1b
pm_processEvent PM(50): channel(3) new state=0x1b
pm_processEvent PM(50): channel(7) new state=0x1b
...LSHM(0): FRU 7 sensor 38 LUN 0 'HS 007 AMC3' hotswap M2->M3
LSHM(0): CU1 FRU 40 added
LSHM(0): FRU 7 sensor 38 LUN 0 'HS 007 AMC3' hotswap M2->M3
LSHM(0): FRU 40 sensor 55 LUN 0 'Hot Swap' hotswap M3->M4
LSHM(0): FRU 0 sensor 32 LUN 0 'HS 000 CM' hotswap M3->M4
........
pm_PwrFree(3)
bp_fru: power chan=1, current limit=8.0 A
PwrAllocate(fru 3 chan 1): allocating 1.0 A granted (available 37.5 A)
pm_EnablePayloadPwr(50,1):
PM(50): payload already ON for FRU 3
PM(50,1):Enable PP (primSite:1 secSite:0)
WARN - LSHM(0): ignore version change sensor
LSHM(0): CM sensor 59 LUN 0 <unknown> hotswap M1->M2
LSHM(0): FRU 3 sensor 66 LUN 0 'HotSwap' hotswap M2->M3
LSHM(0): FRU 3 sensor 66 LUN 0 'HotSwap' hotswap M3->M4
pm_processEvent PM(50): channel(1) new state=0x5b
LSHM(0): FRU 7 sensor 38 LUN 0 'HS 007 AMC3' hotswap M3->M4
LSHM(0): FRU 7 sensor 38 LUN 0 'HS 007 AMC3' hotswap M3->M4
ScanPM: ...
no PM for fru_id=51
no PM for fru_id=52
no PM for fru_id=53
LSHM(0): FRU 40 sensor 52 LUN 0 '+12V' voltage 'lower non-critical go low' -assertion 
LSHM(0): FRU 40 sensor 52 LUN 0 '+12V' voltage 'lower critical go low' -assertion 
LSHM(0): FRU 40 sensor 52 LUN 0 '+12V' voltage 'lower non-recoverable go low' -assertion 
LSHM(0): FRU 40 sensor 51 LUN 0 '+12V_1' voltage 'lower non-critical go low' -assertion 
LSHM(0): FRU 40 sensor 51 LUN 0 '+12V_1' voltage 'lower critical go low' -assertion 
LSHM(0): FRU 40 sensor 51 LUN 0 '+12V_1' voltage 'lower non-recoverable go low' -assertion 
```

But, whatever happens, Sayma never gets powered.

sbourdeauducq, 2018-04-30:
This later error turns off the fans completely (and then parts of the super-reliable µTCA crate overheat), but the MCH still responds to TCP/IP.

vdirksen, 2018-04-30:
@sbourdeauducq : Please start the web interface of the NAT-MCH. If you have not changed the IP address, then you can use "firefox 192.168.1.41". Login if not changed is "root" without quotes and password "nat" without quotes. Scroll down to "System information". Click on this and then let it create a text file. Please send that system file to support@nateurope.com. We can see then the history and also all components in the system. We will try to support you here.

hartytp, 2018-05-24:
@gkasprow I gather this is fixed now, but please reopen if it's not.

sbourdeauducq, 2018-05-25:
@hartytp No commits to https://github.com/m-labs/mmc-firmware so obviously this isn't fixed.

hartytp, 2018-05-25:
Hmm Greg reported having BP Ethernet working on Sayma in a uTCA rack. So I think the issue is fixed in his code. Not having the code uploaded to the repo sounds like a separate issue to me.

gkasprow, 2018-05-30:
@wizath where is the code?

sbourdeauducq, 2018-06-10:
@gkasprow @wizath Did you get Sayma running without power supply or fan-related bugs in the µTCA crate? Did you test it very carefully, as the behavior I observed was non-deterministic?
If so, can we get updated OpenMMC firmware images?

jbqubit, 2018-06-11:
I received two Sayma_AMC + Sayma_RTM from Warsaw this morning. When I put them into my uTCA crate and power it on the AMC lights flash synchronously green, red and blue. No power LEDs are illuminated on AMC PCB. @gkasprow @wizath Did this work in your crate prior to shipping to UMD? 

See attached for diagnostic info for my NAT Native R5 crate. 
[nat_mch_sysinfo.txt](https://github.com/sinara-hw/sinara/files/2090999/nat_mch_sysinfo.txt)
[nat_mch_running_cfg (3).txt](https://github.com/sinara-hw/sinara/files/2091000/nat_mch_running_cfg.3.txt)

gkasprow, 2018-06-11:
Can you try with just one AMC board (without RTM) ?

wizath, 2018-06-11:
> @gkasprow @wizath Did you get Sayma running without power supply or fan-related bugs in the µTCA crate? Did you test it very carefully, as the behavior I observed was non-deterministic?
> If so, can we get updated OpenMMC firmware images?

I've tested Sayma in two configurations:

- Vadatech crate & Vadatech MCH
- Vadatech crate & NAT MCH

On both MCH's board seems to work fine, it acquires power and hotswaps without any problems. 
I had fan issues, but these were related to Vadatech MCH incompatibility with their own FANs - they go crazy and full speed after about 2 mins of uptime.

Another issue is that sometimes RTM fails to get power rails up. This AFAIK is related to Sayma Duo current draw at initialization and because we're using only one PSU atm - although this is rare.

I've also encountered several events of whole uTCA crate dying unexpectedly. I think that is related to this strange transition between Bootloader mode and normal appliaction, since it occured while I was plugging the USB cable in - but that pattern is not really deterministic.

marmeladapk, 2018-06-11:
@jbqubit Does it work not in crate? Perhaps something is wrong with the board itself.

jbqubit, 2018-06-11:
In case of AMC alone in crate the front panel led flashing is different. Only green led flashes. 

I'll try on bench top next.

jbqubit, 2018-06-11:
Power systems on both Sayma are fine when running on the bench top.

jbqubit, 2018-06-11:
openMMC Starting!
build date Apr 19 2018 14:12:54
CPU clock = 80 MHz
Selected board: Sayma
Could not find FRU information in EEPROM, building a runtime info...
>AMC FRU Information:
	-Board info area:
		-Language Code: 0
		-Manuf time: 10463040
		-Manufacturer: Creotech
		-Name: Sayma AMC
		-Serial Number: 1
		-Part Number: SAYMA
		-File ID: AFCFRU
No Chassis info area
No internal use area
	-Product info area:
		-Language Code: 0
		-Manufacturer: Creotech
		-Name: Sayma AMC
		-Part Number: 00001
		-Version: v1.0
		-Asset Tag: No tag
		-Serial Number: 1
		-File ID: AFCFRU
	-Multirecord Area: 
		-Module Current: 8 A
		-Zone3 Compatibility code: 0x11223344
>AMC FRU total size: 220 bytes
RTM Board disconnected!
AMC Hotswap handle pressed!
openMMC Starting!
build date Apr 19 2018 14:12:54
CPU clock = 80 MHz
Selected board: Sayma
Could not find FRU information in EEPROM, building a runtime info...
>AMC FRU Information:
	-Board info area:
		-Language Code: 0
		-Manuf time: 10463040
		-Manufacturer: Creotech
		-Name: Sayma AMC
		-Serial Number: 1
		-Part Number: SAYMA
		-File ID: AFCFRU
No Chassis info area
No internal use area
	-Product info area:
		-Language Code: 0
		-Manufacturer: Creotech
		-Name: Sayma AMC
		-Part Number: 00001
		-Version: v1.0
		-Asset Tag: No tag
		-Serial Number: 1
		-File ID: AFCFRU
	-Multirecord Area: 
		-Module Current: 8 A
		-Zone3 Compatibility code: 0x11223344
>AMC FRU total size: 220 bytes
RTM Board disconnected!
AMC Hotswap handle pressed!

------------ PHY Dump ----------
Register, ADDR DATA
BMCR 0 0x0
BMSR 1 0x0
ID1 2 0x0
ID2 3 0x0
AN_ADV 4 0x0
AN_RX 5 0x0
AN_EXP 6 0x0
EXT_STAT 15 0x0
page0
JIT_DIAG 16 0x0
PCSCR 17 0x0
GMIICR 18 0x0
CR 19 0x0
IR 20 0x0
page1
ID 16 0x0
GPIOCR1 17 0x0
GPIOCR2 18 0x0
GPIOSR 19 0x0
PTPCR1 20 0x0
page2
PTPCR1 16 0x0
?? 17 0x0
?? 18 0x0
?? 19 0x0
?? 20 0x0
page3
?? 16 0x0
?? 17 0x0
?? 18 0x0
?? 19 0x0
?? 20 0x0
------------ Exar Dump ----------
GET_HOST_STS 0x2 0x0 0x0
GET_FAULT_STS 0x5 0x0 0x0
PWR_GET_STATUS 0x9 0x0 0x0
PWR_READ_VOLTAGE_CH1 0x10 0 0 V
PWR_READ_VOLTAGE_CH2 0x11 0 0 V
PWR_READ_VOLTAGE_CH3 0x12 0 0 V
PWR_READ_VOLTAGE_CH4 0x13 0 0 V
PWR_READ_VOLTAGE_IN 0x14 0 0 V

gkasprow, 2018-06-11:
What is the MCH configuration? maybe the current limit is too low per AMC slot?

jbqubit, 2018-06-11:
[nat_mch_running_cfg (4).txt](https://github.com/sinara-hw/sinara/files/2091647/nat_mch_running_cfg.4.txt)
[nat_mch_sysinfo.txt](https://github.com/sinara-hw/sinara/files/2091648/nat_mch_sysinfo.txt)

jbqubit, 2018-06-11:
Also observe that the fans shutdown on the crate but the power remains applied to PSU and AMC. So they get hot. This seems very bad behavior. SB observed something similar on his Native R5. 

I got email from NAT inviting me to upgrade MCH firmware to 2.20.1. Should I do this?

gkasprow, 2018-06-11:
Yes, do it please.

[46 more replies]

closed 2019-03-15 by gkasprow