sinara-hw/sinara #477: NAT MCH configuration for Sayma setup

jbqubit, 2018-01-16:
Users need to know how to do basic configuration of NAT MCH so it plays well with Sayma. Here's a rough outline. 

1. Using putty to configure MCH over a USB cable.
- On Linux the device is /dev/ttyACM0
- 19200, 81N with XON/XOFF flow control
	
2. Using MCH console interface set IP configuration for web interface (GbE1)
- type 'ip' to configure IP address of management web interface. 
- type 'reboot'
	
3. Access web interface on GbE 1  
- URL is typically http://192.168.1.80
- credentials: root, nat

4. MCH default configuration script 
- Does things like https://github.com/m-labs/sinara/issues/476

jbqubit, 2018-02-07:
Users need to know how to configure MCH in order to power on board in uTCA crate. @filipswit Please add a section on this to your Sayma AMC documentation then close.

gkasprow, 2018-02-07:
@filipswit please ask @wizath about it.

wizath, 2018-02-08:
It's pretty much default configuration

>************* MCH Configuration **********************
>INFO - MCH configuration CFG V2.0 for V2.16.1 Final (17:14:43 Jan  8 2015) ok
>
>MCH global parameter:
>---------------------
>remote interfaces:
>  RMCP access:                         enabled
>  telnet access:                       enabled
>  WEB access:                          enabled
>IP address source Mgmt:                DHCP
>RMCP session activity timeout minutes: 0 min
>RMCP session activity timeout seconds: 60 sec
>default fan level:                     5 percent
>MCH configuration flags:
>  Enable backward compatibility V2.4:  yes
>  Enable alternative cooling scheme:   no
>  Control rear transition module fans: yes
>  Enable Telnet inact timeout  (180s): no
>  Set PM Assignment strategy:          strict
>
>Shelf manager parameter:
>------------------------
>ShM configuration flags:
>  allow shelf FRU invalid:             no
>  temperature management:              enabled
>  emergency shutdown:          disabled
>  send SEND_MSG confirmation to SMS:   no
>  use external shelf manager:          no
>
>Carrier manager parameter:
>--------------------------
>carrier number default:                0
>quiesced event timeout:                10 sec
>CM configuration flags:
>  allow carrier FRU invalid:           yes
>  overrule carrier FRU:                no
>  shutdown system if MCH goes down:    no
>  enable Clock E-keying:               no
>
>CM debug:
>  IPMI:                                disabled
>  FRU:                                 disabled
>  E-keying:                            disabled
>  sensor:                              disabled
>  event:                               disabled
>  power module:                        enabled
>  cooling unit:                        disabled
>  CM/ShM interface:                    disabled
>  debugging FRU:                       0
>
>SEL parameter:
>--------------
>SEL configuration flags:
>  'keep on read':                      disabled
>  allocate SEL in non-volatile RAM:    yes
>  ignore 'version change' sensor:      yes
>
>Ethernet switch parameter:
>--------------------------
> configuration source:  load from FLASH
> ignore backplane FRU info:            no
>
>PCIe parameter:
>---------------
>PCIe Virtual Switch configuration
>change via web-interface
> VS # | Host   | NT-Host | Members
>  0     AMC01_4  none      AMC01_4 AMC02_4 AMC03_4 AMC04_4 AMC05_4 AMC06_4 AMC07_4 AMC08_4 AMC09_4 AMC10_4 AMC11_4 AMC12_4 
>  1     
>  2     
>  3     
>  4     
>  5     
>Upstream slot power up delay:          5 sec
>PCIe hot plug delay for AMCs:          0 sec
>PCIe configuration flags:
>  100 MHz spread spectrum:             disabled
>  hot plug support:                    disabled
>  PCIe early ekey (before payload):    disabled
>  'no ekey' for PCIe:                  disabled
>  PCIe clustering:                     enabled
>
>Time Protocol/SNTP parameter:
>--------------
>Time server IP:                        0.0.0.0
>'Check for Time' delay minutes:        0 min
>'Check for Time' delay hours:          0 h
>local time offset:                     0 h
>Configuration flags:
>  Time client:                         disabled
>  Time client protocol:                Time Protocol
>
>DHCP Configuration:
>-------------------
>host name:                             nat-mch
>flags:                                 0x00

jbqubit, 2018-02-15:
@wizath The full configuration dump has lots of extraneous lines. Better if the user is instructed to 
1. reset MCH to its factory default configuration
2. apply short list of changes

closed 2019-05-28 by jbqubit