sinara-hw/sinara #478: new Ethernet chipset for Sayma 2.0?

jbqubit, 2018-01-17:
From #854, need to decide between a) keep existing Ethernet solution which gives FPGA Ethernet access but no Ethernet for MMC b) switch to different chip and give Ethernet to both FPGA and MMC. 

@gkasprow said

> On other AMC boards I use single-chip Ethernet switch that has RGMII on two ports, RMII or MII on one port, 2x RJ45 and 2x 1000BASE-X/SGMII. It does not need any configuration, just pin-straps. It costs 17$.

Which chip is this?

gkasprow, 2018-01-17:
[88E6321](https://origin-www.marvell.com/switching/assets/LinkStreet_88E6341_Product_Brief.pdf)
I have a devkit and I checked that it works well without any SW nor configuration memory.
I didn't wanted to propose it for Sayma due to NDA constraints.

hartytp, 2018-01-17:
@gkasprow what's the use-case for ethernet for the MMC? This isn't something I can see ARTIQ using, but is it something that non-ARTIQ users or Sayma are likely to value enough for us to bother with it?

gkasprow, 2018-01-17:
Original idea was to provide FLASH upgrade via Ethernet but we can do it over JTAG providing that we have NAT MCH + JTAG switch module.

gkasprow, 2018-01-17:
Of course we can do it via USB :)

jbqubit, 2018-01-17:
Since so many aspects of Sayma are controlled by MMC it's desirable to have some chassis-level support for flashing, configuring and reading parameters from MMCs.

gkasprow, 2018-01-17:
by default, there is I2C available via MCH user interface.

hartytp, 2018-01-17:
Would that mean that Metlino would need to support that (ie would we have to write extra firmware for that approach)?

gkasprow, 2018-01-17:
Metlino uses same MMC as Sayma.

hartytp, 2018-02-26:
Sounds like the plan is to use the NAT JTAG switch module (or similar) for MMC update rather than putting a complex, NDA-requiring Ethernet switch on Sayma.

closed 2018-02-26 by hartytp

gkasprow, 2018-02-26:
When somebody needs Ethernet MMC access, one can consider using one of new MCUs with RGMII, but it's probably not worth the effort.