sinara-hw/sinara #479: Sayma 2.0: support for JTAG from uTCA backplane

jbqubit, 2018-01-17:
For development of Sayma 1.0 JTAG via FTDI is always-on; see https://github.com/m-labs/sinara/issues/463#issuecomment-358029840.  Decide if we want to support extension of JTAG chain to uTCA backplane for Sayma 2.0.

gkasprow, 2018-01-17:
It does not have to be always on, it can be set to USB  by default. I use JTAG via backplane regularly. NAT offers remote debugging and FPGA programming via JTAG port.

hartytp, 2018-01-17:
AFIACT (but @gkasprow should correct me on this if I'm wrong) it's probably worth keeping the uTCA BP JTAG since it's likely to be helpful for non-ARTIQ users. However, if we are going to do that then @gkasprow should demonstrate that the USB-JTAG setup works really reliably, without any of the issues we've encountered so far (but, AFAICT, he's getting pretty close to that).

Also, if all issues with the current JTAG arrangement are solved by adding a single jumper (I haven't tested this myself yet) then this doesn't seem to be a major issue for us anyway.

jbqubit, 2018-01-17:
@gkasprow Have you tested this on Sayma? I'm waiting for a firmware update from NAT to get JTAG switching enabled for my MCH.

----

"JTAG Switch Module Configuration" option on MCH is not available. NAT web interface reports "Sorry! NAT-MCH BASE PCB V3.5 FPGA V1.9 doesn't support NAT-JSM. Minimal requirements: PCB V3.1 FPGA V1.10"

My MCH version info:
- My PCB PCB Version is V3.5. 
- Firmware VersionV2.18.8 Final (r14042) (Mar 31 2017 - 11:29)
- FPGA VersionV1.9
- Microcontroller VersionV1.2

vdirksen, 2018-02-05:
Hi Joe,

if not already done, please update to the NAT-MCH firmware 2.19.5.

This version should also update your FPGA and you should see JSM working.

The differences of the firmware versions can be found in the release.txt file.

Best regards
Vollrath

hartytp, 2018-05-24:
Closing, as I don't think this issue offers anything new on top of the other JTAG issue.

closed 2018-05-24 by hartytp

gkasprow, 2018-05-24:
This was tested some time ago. I managed to get access to the FPGA from my AMC mini backplane.
It was not tested with JSM