sinara-hw/sinara #481: Sayma HMC830 ground "NC pins"

jbqubit, 2018-01-18:
Discussed here. 

https://github.com/m-labs/sinara/issues/472#issuecomment-357932888

hartytp, 2018-01-19:
Well, as I said, I generally prefer to follow the eval board (which doesn't ground all these pins) to the datasheet. Data sheets are often misleading/ambiguous, but we know the eval board works (we have one and are using it)!

gkasprow, 2018-08-10:
so let's keep it as it is now.

hartytp, 2018-08-11:
@gkasprow let's follow the connections on the eval board, instead of leaving these pins floating.

gkasprow, 2018-11-27:
fixed 
[image]

hartytp, 2019-01-09:
> project plan states that hmc830 should be primary clock source. Is this correct?

Could we leave this until the end of the design review, please? I'd like to finish testing the ADF PLL before we make a final call on this (working atm, but have some other strict deadlines to meet this week as well, so can't spend as much time as I'd like).

Ideally we'd have finished doing design studies with existing hardware before drawing up the project plan, allowing us to treat the project plan as totally fixed. However, that's not how things worked out and we're still doing prototyping and design studies during the design review. It would be a mistake to stick to the plan blindly, even if new evidence comes to light that suggests major improvements can be made to the design with small tweaks.

hartytp, 2019-01-09:
If we're going to stick to the plan in all cases regardless of any results of the tests we're doing atm then I won't bother doing the tests.

marmeladapk, 2019-01-09:
@hartytp It was not a call to stick to the plan, plan can, and has been revised  few times during the review.

Anyway, that's not the point of my comment, with the way caps are now positioned you can either have ufl or adf clock source, where instead it should be hmc or adf, right? See C423 and C424, C432 and C433 below. 

[image]

hartytp, 2019-01-09:
> @hartytp It was not a call to stick to the plan, just to see which one is write and modify schematics or the plan accordingly. Plan can, and has been revised few times during the review.

@marmeladapk I know. Not a response to you, but rather to anticipated feedback from other quarters. 

Anyway, agreed about your point (although, this will all need a bit of tweaking in light of the discussion about ADF PLL connections).

gkasprow, 2019-03-11:
I think we can close it now. HMC pins are connected as on a devkit.

closed 2019-03-11 by gkasprow