sinara-hw/sinara #482: Sayma_AMC renaming

jbqubit, 2018-01-18:
In https://github.com/m-labs/sinara/issues/253#issuecomment-357508979 @hartytp suggested 

> we should probably rename Sayma_AMC to something else. Motivation for this -- totally independent from the Russian names thing -- is that Sayma was originally intended to be an RF generation board. I think it gets confusing to also name the AMC "Sayma" once it becomes a general purpose carrier that is used for other boards like Shuttler etc.

> My preference would be to keep the Sayma name for the RTM (or give it a new non-russian name if anyone still cares enough to bother) and find a new name for the AMC.

I agree with @hartytp that Sayma_AMC may serve as a useful FMC/RTM interface for a variety applications beyond Sayma_RTM. A distinct name would disambiguate things going forward. I propose we hold off on this discussion until Sayma v2.0 is finalized.

jbqubit, 2018-01-18:
Copying from https://github.com/m-labs/sinara/issues/253 .... 

@sbourdeauducq responded to @hartytp suggestion above

> Not so confusing. That's why we have opaque names...

@hartytp responded

>his is nothing to do with opacity or russian names.

The AMC v RTM distinction is easily lost on users (who mainly don't know much about uTCA). So, Sayma_AMC and Sayma_RTM sound a lot like the same product. So, it's confusing to explain to people that Sayma is an RF generating board, but it's also a general purpose FPGA card that can be used by shuttler and other projects.

> IMHO it's better to have clearly distinct names (opaque or whatever) for conceptually different components. Just makes things less confusing for users (which is the whole point of giving things names in the first place).

> @sbourdeauducq To be clear, this is also not meant as an attack on the name "Sayma" (I don't like that name, but that's a separate conversation!).

> I just think that users will find things easier to understand if "Sayma" refers to one of the AMC or RTM, and we find a different name for the other one.

jbqubit, 2018-01-18:
I agree with @hartytp that Sayma_AMC may serve as a useful FMC/RTM interface for a variety applications beyond Sayma_RTM. A distinct name would disambiguate things going forward. I propose we hold off on this discussion until Sayma v2.0 is finalized.

gkasprow, 2018-01-18:
I call my open source AMC FMC boards in following way
AFC - Artix based AMC FMC Carrier
AFCK - Kintex based AMF FMC Carrier
AFCZ - Zynq based AMC FMC Carrier.
This one could be AFCU :)
I know this is not opaque name.

hartytp, 2018-01-21:
Okay, so let's keep "Sayma" as a name for Sayma_RTM (at least for the time being), and come up with a new name for Sayma_AMC.

Generally, I'd be in favour of something semi-opaque along the lines of the new EEM names. Something that hints at its role as the digital/DSP half of these boards. Suggestions? (Shutdown sweepstake for @dhslichter while you can't do anything productive at work )

closed 2018-03-28 by jbqubit