sinara-hw/sinara #483: RGMII Ethernet on Sayma_AMC

jbqubit, 2018-01-18:
Tracking https://github.com/m-labs/artiq/issues/854

hartytp, 2018-02-26:
This now appears to be fixed.

@gkasprow While this is fresh in all of our minds, would you mind summarising the changes that need to be applied to Sayma v2.0 to fix this properly? Thanks!

gkasprow, 2018-02-26:
- [ ] better RGMII nets equalisation
- [ ] removal of the RGMII bus mux
- [ ] optionally Tx cloc termination. I'm going to test it in a moment
- [ ] placement of the PHY closer to the FPGA if clock termination won't help

hartytp, 2018-02-26:
Thanks @gkasprow. Were you also planning to change the phy (was there an issue with the MMC not supporting GbE?).

gkasprow, 2018-02-26:
There is no PHY that supports speed translation. I'd have to use Marvell switch IC I use it in other project.

gkasprow, 2018-02-26:
Such chip adds some latency, and its documentation is under NDA...

hartytp, 2018-02-26:
Well, I'd definitely prefer to avoid that if possible. The question is just whether you need it for anything? (I can't remember the argument for it. Something about MMC updates? Can you do that another way easily?)

gkasprow, 2018-02-26:
Actually I don't need it that much. It was for remote FLASH update only, but recently you can do it via JSM

hartytp, 2018-02-26:
Ace. Let's scrap it then and stick with the current phy.

jbqubit, 2018-02-26:
> Ace. Let's scrap it then and stick with the current phy.

Agreed.

jbqubit, 2018-02-27:
Is there anything to add to [this](https://github.com/m-labs/sinara/issues/484) summary of the required white wire fix? Figure it's good to check before cutting into my Sayma_AMC.

gkasprow, 2018-02-28:
Just follow the instruction. If you feel that more details are needed let
me know

jbqubit, 2018-04-26:
@gkasprow Good find!

hartytp, 2018-04-27:
@gkasprow So, the big question then: does ethernet on Sayma now work on all combination of AMCs + media adapters with that fix?

gkasprow, 2018-04-27:
@hartytp I will respond to this question once I test all AMC boards Joe sent me and make sure they work :)

jbqubit, 2018-05-01:
@gkasprow Please also check PORT0 Ethernet with boards inserted into uTCA crate.

gkasprow, 2018-05-01:
@jbqubit I will ship you all the stuff back after I make sure all works in the MTCA crate.

gkasprow, 2018-05-16:
Finally I made Ethernet working on all AMC boards.
The last one has broken connection between DIO0 and BGA ball so the Rx clock did not arrived at the FPGA. I switched it to DIO3 and it works. I will give this board to Technosystem for X-ray inspection.
Next step is to check if they work in the MTCA crate.

hartytp, 2018-05-17:
@whats the plan for testing that ethernet works on Sayms using ARTIQ?

gkasprow, 2018-05-17:
@we are trying to make it working in the MTCA crate
MCH reports the link established so far.

gkasprow, 2018-05-17:
@jbqubit @hartytp it works in MTCa crate.
Both in #1 and #12th slot.
At lease we can ping the boards.
@marmeladapk is changing the IP adress of one of them to check operation of multiple boards in same crate.

marmeladapk, 2018-05-17:
It works.

hartytp, 2018-05-17:
Nice! Is that running ARTIQ?

marmeladapk, 2018-05-17:
@hartytp Yes.

hartytp, 2018-05-17:
Great. Glad that's solved then.

gkasprow, 2018-05-17:
The PSU in my crate is not strong enough to supply more Saymas, I will
bring another one from Creotech and try.

jbqubit, 2018-05-17:
I wasn't necessarily worried about testing multiple Sayma_AMC at the same time. Rather, testing that all the Sayma_AMC that you have on hand independently work with uTCA Ethernet.