sinara-hw/sinara #484: RGMII white wire patch HOWTO

jbqubit, 2018-01-19:
Thanks to the hard work of @gkasprow and @sbourdeauducq RGMII is more or less [working](https://github.com/m-labs/artiq/issues/854#issuecomment-358842622) on Sayma. This path requires a [white-wire modification](https://github.com/m-labs/artiq/issues/854#issuecomment-352244364) to Sayma_AMC. So far its been done in two boards. 

@gkasprow Please update your instructions on how to do this patch in this Issue. Do you have photos of what this looks like once complete? Are there any pitfalls we should avoid?

gkasprow, 2018-01-19:
I'm not sure what white wire you are talking about.
I applied PHY clock connection on all boards

sbourdeauducq, 2018-01-19:
RGMII RXCLK to clock-capable GPIO0.

sbourdeauducq, 2018-01-19:
Note: On future PCB version, the RGMII RXCLK should be placed in the same bank as the other RGMII signals. The FPGA still doesn't like it too much when you are clocking a bank with another (or at least it doesn't like this particular combination of banks), and it still needs a special Vivado command (``CLOCK_DEDICATED_ROUTE BACKBONE``) to complete the routing.

jbqubit, 2018-01-19:
@gkasprow The modification that will allow me to duplicate what you and @sbourdeauducq did to get RGMII working.

gkasprow, 2018-01-19:
I'll update my board and make  microscopic photos with instructions.

gkasprow, 2018-01-25:
@jbqubit 
The steps:
- cut the trace with sharp blade

[image]

- solder thin insulated wire
[image]

- solder it to pin 1 of GPIO connector
[image]

- secure the wire with thermal glue
[image]

closed 2018-01-31 by hartytp