sinara-hw/sinara #486: fix bloat in Sinara repository

jbqubit, 2018-01-19:
New Issue prompted by discussion in https://github.com/m-labs/sinara/issues/436. 

@jordens said

> This repository is poisoned.

@jordens has discussed problem in depth and and proposed standards to improve repository hygen in these Issues #39, #46, #328. As of 2/19/2018 size of Sinara repository is 3.0 G. 

@hartytp said

> Also, having cloned Sinara a few times recently on slow WiFi connections, I think I can guess what RJ means by "poisoned". Sinar repo is needlessly huge due to all the PDFs that have been uploaded.

> I'm half tempted to suggest that we just burn the repository and start afresh but with much stricter rules (gitignore all pdfs, "Project output" dirs etc) and use releases or whatever for PDFs (or even just attach them to the relevant GitHub issues).

@dhslichter said
> Ack @jordens, new repository just for Sinara documentation sounds good to me. I think we can keep the current repository for the design stuff, though, and achieve the "burning" that @hartytp wants by having ONE designated person go through and clean out the Augean stables, then have everyone operate on a much stricter basis going forward regarding generated files etc. Aggressive use of .gitignore is clearly called for.

jbqubit, 2018-01-19:
As we discuss how to fix the repository please have the grace to not give @gkasprow (and others) a hard time for past practices. Without knowing git or the github tracking system @gkasprow and most of us jumped in began using these tools. git is complicated and our development has progressed at a blistering pace. It's unfortunate that the repository was messed but there's often collateral damage like that when learning new tools. 

I'm proud of the collaborative culture that we've cultivated in Sinara. And, I'm confident that in time we can work out the repository bloat problem and adopt improved git hygiene. But for the present I strongly advocate that we hold off dealing with the repository bloat and keep on with the status quo until after Sayma 2.0 and Kasli are finished.

marmeladapk, 2018-01-20:
We can easily separate each Altium project to separate repository, they're self-contained. However Xpedition folder has similar size and houses only 3 boards. Also, as far as @gkasprow told me how Xpedition works, it's hard to separate those projects, because they share schematics and database between each other.

And even with Altium, we constantly upload binary data that is already compressed, so with each commit, I guess, we store completely new copies of files, so sooner or later repo will grow in size.

jbqubit, 2018-02-28:
Due to the mess of the git repository itself, most of the value of Sinara on github is in metadata like the wiki and Issues. github doesn't provide a clean migration for metadata when creating a new repository. I think it would be much better to clean up existing sinara vs creating a new repository from scratch. 

Since there wasn't much thought put into most of the commits that contribute to bloat, they're not of much value anyway. I propose we scrap all prior revisions of PCB designs in the repository. Here's how I think this could work. 

1) clone sinara repository
2) copy all the files to another directory on disk
    - [clean up](https://github.com/m-labs/sinara/issues/46) the files to get rid of gunk 
3) figure out hash of first commit 
```
$ git rev-list --max-parents=0 --abbrev-commit HEAD
```
4) reset to that hash 
```
$ git reset --hard b618697
```
5) copy a single PCB design back into the repository 
6) ```git add``` then ```git commit``` then ```git tag```
    - use tag names that reflect current release status of the boards
    - @jordens has laid out an [approach](https://github.com/m-labs/sinara/issues/328)  
7) repeat 5 and 6 for each Sinara PCB
8) publish to a temporary github repository for others to inspect
9) if there is consensus that it's peachy make the changes permenant on sinara
``` git push -f```

If all goes well we're left with slim sinara and with discipline we can keep it smallish. 

-----------
Aside:  While googling around on this I also found a nifty script that shows large files in a repository. 

```
git rev-list --objects --all \
| git cat-file --batch-check='%(objecttype) %(objectname) %(objectsize) %(rest)' \
| awk '/^blob/ {print substr($0,6)}' \
| sort --numeric-sort --key=2 \
| cut --complement --characters=13-40 \
| numfmt --field=2 --to=iec-i --suffix=B --padding=7 --round=nearest
```

Right now it is 2.7 GB in size. Biggest commit is 79 MB.

closed 2018-05-03 by hartytp