sinara-hw/sinara #487: Urukul float SMAs

hartytp, 2018-01-20:
- [ ] For Urukul, should we float the SMAs (clock input and RF outputs) by changing 0R resistors to 100k resistors.

AFAICT, this is what one generally wants. IIRC, our only only reason for not doing it originally was that we didn't want to bother fitting insulating washers. As we're now doing that, it seems that we should just float the grounds. This only requires a minor alteration to the BOM.

hartytp, 2018-01-25:
Any objections to doing this?

@gkasprow Since this just requires changing a single resistor value, can we do this for the current production round (I assume that population hasn't started yet so this kind of change won't cause issues)?

gkasprow, 2018-01-25:
We can still change it. I need to generate new P&P  and BOM files

hartytp, 2018-01-25:
Well, unless @jordens objects, let's do it! I'd like to try to make this the definitive Urukul v1.x, so it would be nice to close all outstanding issues if possible.

jordens, 2018-01-25:
OK. Do it.

hartytp, 2018-01-25:
Whoo!

Well, FWIW, we may want another slight BOM change after we've looked at the AD9910 loop filter. But, hopefully the current design is close enough to optimal that we can leave it.

hartytp, 2018-01-25:
@gkasprow @WeiDaZhang will test the AD9910 loop filter on Monday. Hopefully we will find that the performance is close to the simulations and no changes are required. However, if we can make a significant improvement by changing the loop filter passives, would it be possible to include those changes to v1.1? We should have finished this work by the end of Tuesday and will be able to tell you what changes if any are needed by then. Sorry for not getting this done sooner, but there has been so much to do recently.

gkasprow, 2018-01-25:
We still wait for PCBs so this should not be a problem.

hartytp, 2018-01-25:
Good. Let's not hold up the manufacturing for this, but hopefully we can finish in time to make any changes that are needed.

hartytp, 2018-02-02:
@gkasprow please can you confirm that this will be done for the production round of Urkul v1.1 (and the schematic updated) and then close the issue.

AFAICT, no other changes are required for this round.

Thanks!

hartytp, 2018-03-06:
AFAICT, this has been done for V1.1, so we can close this issue.

closed 2018-03-06 by hartytp