sinara-hw/sinara #488: Clock synchronization using Sinara

werty37, 2018-01-23:
Sorry for the ignorance, is Sinara related to White Rabbit?
https://www.ohwr.org/projects/white-rabbit

I am looking at open source time/clock synchronization hardware with nano second accuracy for indoor positioning system for an academic project.

jordens, 2018-01-23:
Related in the sense that both involve open source, open hardware, physics, real time controls, accurate synchronization, distributed devices. The protocols are different however.

jordens, 2018-01-23:
But you can do emission and time tagging of signals to nanoseconds with artiq/sinara.

werty37, 2018-01-24:
Thank you Jordens. How can i learn about Artiq/Sinara? On a high level, I understand that Kasli hardware generates the clock and is distributed over Fiber to slave devices. 

Would you please point me to the tutorial sections for someone who is a beginner in clock sync and electronics in general?

jordens, 2018-01-24:
For clock synchronization and electronics in general, please refer to a good book. One text that is sometimes recommended is https://artofelectronics.net/ But there are many other textbooks.
And then there are a lot of good documents on white rabbit that will get you started with the general topic, https://www.ohwr.org/projects/white-rabbit/documents
After that, maybe the ARTIQ manual, https://m-labs.hk/artiq/manual-release-3/

closed 2018-01-24 by jordens