sinara-hw/sinara #489: Sampler v2.0 Errata and tests

gkasprow, 2018-01-24:
- [  ] R29 = 20k instead of 120k

gkasprow, 2018-01-24:
Interesting thing. I forgotten to add damping tantalium capacitor at the output of the LC filter which is also input of the LDO regulator. And I observed 2Vpp, 76kHz beautiful sine wave at that point. It was caused by 12V LDO.
Once I added 10uF  tantalium capacitor, the oscillations stopped. I did not expect oscillations with such high amplitude. It seems that 10uH together with 3uF ceramic capacitor formed LC circuit.
It also means that 3x 0603 1uF capacitors do not have 3uF. To form 76kHz oscillator with 10uH coil, the capacitor must have much smaller value than that.
X7R/X5R capacitance degrades with voltage and this is the case. The capacitor has roughly 500nF which is consistent withX7R characteristics. Caps are 25V rated and operate at 13V so have only fraction of initial capacitance.

gkasprow, 2018-01-24:
@hartytp the power supply works.

hartytp, 2018-01-24:
@gkasprow Good catch on the regulators. Wonder if that had something to do with the DAC deaths we were seeing before?

Are there any other boards we need to double check with this (e.g. Zotino/Urukul?)

hartytp, 2018-01-24:
Also, are there any other places that we should add some tantalums? e.g. between the SMPS outputs and the inductors?

Also, is it worth using X7Rs with a larger package and a higher voltage rating to help with this?

gkasprow, 2018-01-24:
Zotino has these caps in place.

gkasprow, 2018-01-24:
Previous version of Sampler has them as well...

gkasprow, 2018-01-24:
I connected dual channel generator to CLK and CNV input via 3U TTL tester module.
CNV is 1kHz 10% duty cycle pulse, CLK is 10MHz clock. I observe signal patterns at the ADC output, as well as at the 3U TTL tester module output. When I touch the ADC inputs, the pattern changes. So it seems to convert the input data.

gkasprow, 2018-01-24:
There is one bizzare behaviour of the ADC OVDD.
It has value of 3.9V which is far too much for this ADC. Max ratings specify it at 3V.
The LDO is supplied with 3.4V so cannot be the source.
2.5V rail is connected to the LVDS signal receiver that generates CNV pulse. LVDS inputs are 0.6 and1.2V. So this receiver cannot be the source neither.
It looks like the ADC generates such high voltage via its internal circuits. I have same situation on another board.
When I disable both +/- 12V rails the situation remains the same. All input signals are 2.5V.
When I source 30mA current from this pin, the voltage drops a few mV..

gkasprow, 2018-01-24:
I checked also DDR config pin. The configuration gets updated immediately, without need of power cycle.
This is what I get in SDR mode:
[image]

a-shafir, 2018-01-24:
@gkasprow "It also means that 3x 0603 1uF capacitors do not have 3uF"
It is not widely known but it is kind of "normal". The story is following: initially MLCC technology was quite inaccurate and it ends to industrial standard "-20%,+80%@ tolerance. Since it is still most common tolerance type. So the quality manufacturers with technology capable for +-5% just producing -15% by default for the cost reduction.
So if the manual saying "3uF minimum" it will be safe to put 2.2uF * 2 or so but certainly not 3.3uF. 
Btw, the similar story with the tantalum and aluminium caps.

gkasprow, 2018-01-24:
@a-shafir -20%,+80% tolerance is yet another story. This is specified at 0V. C=f(V) characteristic is also widely known, almost every datasheet specifies that. Even if you take +/-5% X5R capacitor, it will have 5% tolerance at 0V, but its capacitance will degrade with voltage reaching even 10% of initial capacitance. It strongly depends on case size. 1210 are least sensitive, 0402s are the most.
Aluminium and tantalium ones do not have such behaviour.
The aluminium ones degrades with time when operated at voltage different than nominal.
Let's say you install 100uF capacitor rated at 100V and connect to 10V. With time, the oxide layer will get thinner and thinner leading to increase of the capacitance, even by the factor of two. You can revert it to nominal capacitance by connecting to 100V but via large resistance to limit the current. After weeks, the oxide layer will form again and capacitor will get its original capacitance.

hartytp, 2018-01-24:
@gkasprow What's the plan then? Do you think you've seen enough to be happy to ship these boards to us and @jordens? Or, do you want to finish investigating the OVDD issue?

hartytp, 2018-01-24:
> I checked also DDR config pin. The configuration gets updated immediately, without need of power cycle.
This is what I get in SDR mode:

Okay, that's good.

@jordens in that case, would it be better to DNP the switch in a future revision and allow the FPGA to control this pin instead. That way the user doesn't have to remember to set the switch to the right place to get this up and running. (Make it so that if the FPGA doesn't program it then it's set to SDR by default).

gkasprow, 2018-01-24:
@hartytp I wrote to Analog Devices about this issue.
This happened after I measured 2.5V instead of 5V . It was caused by wrong resistor (120k instead of 20k). But this 2.5V was not delivered by the LDO but by some other means.
So the ADC was switched on without the main power supply, but the datasheet says that it does not require any power sequencing.
 I fixed this issue on remaining 3 boards but still the problem persists. But it also means that ADC chip was not broken by missing the main power supply.

gkasprow, 2018-01-24:
@hartytp I can ship you 3 pieces and keep one at WUT for investigation. Once I finally get the devkit will do more measurements.

hartytp, 2018-01-24:
@gkasprow 

> @hartytp I wrote to Analog Devices about this issue.

It's odd seeing LTC being referred to as ADI...

>  It was caused by wrong resistor (120k instead of 20k) in LDO.

Okay, so that incorrect resistor would have put the 5V supply rail at 1.85V instead of 5V.

> So the ADC was switched on without the main power supply, but the datasheet says that it does not require any power sequencing.

Really? Looking [here](http://cds.linear.com/docs/en/datasheet/232016fb.pdf) on page 2 there are some requirements.

The most relevant is that the analog inputs and reference must not exceed the supply voltage. So, the reference will have driven 20mA through its ADC pin, while the OpAmps will have driven (2.5V-1.8V-0.3)/50R=8mA through each AIN+- pin. Not sure if that's enough to damage the ADC though.

> I fixed this issue on remaining 3 boards but still the problem persists. But it also means that ADC chip was not broken by missing the main power supply.

So, this problem with the incorrect OVDD is present on all boards you have, even ones which were not powered up with an incorrect VDD?

hartytp, 2018-01-24:
Why are we pulling SDR/DDR up to P5V0A instead of 0VDD?

gkasprow, 2018-01-24:
@hartytp the board has power sequencing.
There was yet another problem with R22 so +/- 12 were not powered on. But the reference was...
But it was a case only on 1 board, I fixed it on others and switched the power on after that.
Yes, the problem is on all boards.

gkasprow, 2018-01-24:
@hartytp I've noticed that so on all boards I keep this pin low.
Anyway, the input current is too low to cause latchup

gkasprow, 2018-01-24:
on page 24 there is a chapter about power sequencing.
Each such circuit contains input protection diodes, but once you don't exceed 20mA, it won't cause a latchup and chip damage.

hartytp, 2018-01-24:
@gkasprow am I misreading the datasheet for that 2V5 regulator, or is it incapable of sinking current from its output?

gkasprow, 2018-01-24:
Usually LDOs can only source the current. Only ones dedicated for DDR termination can sink/source

hartytp, 2018-01-24:
> Usually LDOs can only source the current. Only ones dedicated for DDR termination can sink/source

Okay. Anyway, I guess that the OVDD pin has to source quite a bit of current into those LVDS outputs, so whatever leakage is causing this problem needs to be quite high (>20mA). That narrows down the number of possible sources quite a bit!

I suspect that this must be the 5V or 6V supply getting connected to the 2V5 supply through a few diodes, but I can't see how that could happen!

Probably best not to ship any boards until this issue is tracked down and resolved.

gkasprow, 2018-01-24:
I agree. This can lead to catastrophic failure of the chip.

hartytp, 2018-01-24:
Aaah, I think it's pin 25. CMOS/LVDS should be tied to OVDD, not VDD.

hartytp, 2018-01-24:
That's connected directly to the power supply rail with no current limiting resistor.

hartytp, 2018-01-24:
For now, just make R11 100k or so, but let's fix that in the next revision.

May also be worth replacing the ADCs before shipping ;)

gkasprow, 2018-01-24:
Good point @hartytp  I noticed possible problem with pin 23, but not with 25 which is obvious.
the 5V gets via protection diode of pin 25...

gkasprow, 2018-01-24:
Actually, it was 10k in some earlier revision and then I forgotten about it and changed to 0R....

hartytp, 2018-01-24:
:) should be easy to fix. Well, if those are the only mistakes in this design then I'm impressed!

Once that's done (and the ADCs are replaced if you think that's a good idea) I'm happy for you to ship these boards to us and @jordens so long as there is nothing else that looks suspicious.

gkasprow, 2018-01-25:
I ordered new ADCs.

hartytp, 2018-01-25:
@gkasprow Great! out of curiosity, when you change that resistor to 100k did that solve the OVDD issue?

Are there any other components that should also be replaced at the same time as the ADC, such as the VLDS driver? I don't want to waste time when we test these boards because of strange behaviour caused by damaged components, so I'd prefer to replaced anything that has been taken out of its abs max range.

gkasprow, 2018-01-25:
@hartytp the issue disappeared. One LVDS driver got 3.9V but it is able to survive 4V.

hartytp, 2018-01-25:
@gkasprow Great, so no known bugs left in this version ;)

jordens, 2018-02-01:
@gkasprow AFAICT, the shield clips around the SMPS bridge over the slot in the ground plane (at least on the left) and allow return currents to flow where you didn't want them (modulo the shield resisitivity). Not dramatic but probably not optimal.

[image]

gkasprow, 2018-02-01:
indeed, I didn't notice that.
At least it's worth to check the return current influence just by shorting them with shielding can:)

gkasprow, 2018-02-01:
I've received the devkit today so will continue with detailed tests.

hartytp, 2018-02-02:
@gkasprow No need to redo all the detailed tests we did on Sampler v1.0 (I'll do that with the M-Labs ARTIQ driver). Just check you get some samples out of it and ship it, please.

Let's focus on Sayma for the time being.

jbqubit, 2018-02-02:
> Let's focus on Sayma for the time being.

Yes!! :)

hartytp, 2018-02-08:
@gkasprow What's the status of this? Obviously Sayma still takes priority, but it would be good to ship these boards as soon as you've replaced the ADCs and checked that you can get samples out with the DevKit (no need for any detailed testing before shipping).

Thanks!

gkasprow, 2018-02-08:
well, I managed to assemble the setup, but tomorrow I'm leaving abroad for one week holidays. So iwll continue once I get back to the lab.
[image]

hartytp, 2018-02-08:
Have a good, well-earned break!

jordens, 2018-03-21:
Works well with ARTIQ 97918447a3426c39f23d3766049c7fb644c0a33e

gkasprow, 2018-03-21:
Moved release 2.1 to new, dedicated repository
https://github.com/sinara-foundation/Sampler.git

closed 2018-03-29 by hartytp

hartytp, 2018-03-29:
@marmeladapk Nice! Good job. Thanks for doing that!

To check I understand you, is the dB number including the gains? (i.e. does it reflect the input voltage one would have to apply to a given channel to produce that level of cross-talk, or is it neglecting the gains)?

If you have time, could you also do that measurement for G=1 on all channels (determines how the cross-talk scales with gain) and post the results to the Wiki?

marmeladapk, 2018-03-29:
> is the dB number including the gains

@hartytp Yes, I use adc_mu_to_volt function with respective gains.

hartytp, 2018-03-29:
Thanks for confirming.

So, the ADC cross-talk should be -95dBc at 300kHz. If we were limited by ADC cross-talk, the effective (input) cross-talk would be -95dBc + 20*log10(gain) = -155dBc for g=1000.

So, the conclusion seems to:
- the g=1 cross-talk should be limited to -95dBc by the adc (needs confirming)
- The g=1k cross-talk is limited to -155dBc by the ADC. In fact, we seem to outperform this for some channels, so either my maths is wrong, or the ADC outperforms the specification (see datasheet p9)
- the g=1k cross-talk is limited to -110dBc for signals on the same stacked BNC connector
- For other channels/other gains, the cross-talk lies in between.

Anyway, IMHO, that data looks good! Nice to see that the layout is clean enough to give high isolation even at the limits of the ADC BW.

marmeladapk, 2018-03-29:
Same setup as before, just 1x gain on all channels:

Crosstalk [dBc]:
```
0.00, -84.36, -100.65, -100.16, -102.72, -93.51, -96.23, -105.70
-91.95, 0.00, -87.47, -104.87, -115.80, -99.91, -101.55, -106.71
-109.04, -86.28, 0.00, -88.78, -96.81, -95.41, -108.53, -109.23
-101.31, -97.47, -92.72, 0.00, -88.65, -96.58, -100.80, -97.46
-101.27, -95.18, -97.16, -88.29, 0.00, -87.26, -99.11, -100.12
-103.41, -102.10, -101.54, -104.59, -99.87, 0.00, -89.34, -102.49
-104.62, -104.64, -103.39, -101.73, -104.08, -87.61, 0.00, -88.34
-100.67, -99.20, -97.34, -95.48, -102.93, -113.76, -92.80, 0.00
```
Sample spectra:
[image]

All measured spectra and my script (to check if my measurements are correct.

[sampler_measurements.zip](https://github.com/sinara-hw/sinara/files/1860976/sampler_measurements.zip)

hartytp, 2018-03-29:
Thanks for the data.

Okay so the adc cross-talk varies by something like +-10dB between channel pairs. Gòd to know!