sinara-hw/sinara #490: Sayma documentation: \input LaTeX derived from github wiki markdown

jbqubit, 2018-01-26:
Some elements of Sayma documentation already exist on Sinara wiki. It would be nice to make some elements of Sayma documentation dependent on wiki to prevent duplication. In schematic... 

1. clone wiki locally 

```
$ cd ~/github
$ git clone https://github.com/m-labs/sinara.wiki.git
```

2. Convert github markdown to LaTeX. [markdown-latex](https://github.com/cebe/markdown-latex) is nice. 
To install: 
```
$ git clone cebe/markdown-latex
$ cd markdown-latex
$ sudo apt-get install php5-cgi
$ php -r "copy('https://getcomposer.org/installer', 'composer-setup.php');"
$ php composer-setup.php
$ echo { "require": {"cebe/markdown-latex": "*" } } > composer.json
$ php composer.phar update
```
To convert: 
```
$ bin/markdown-latex --flavor=gfm ../sinara.wiki/Kasli.md > Kasli.tex 
```

3. Include in body of existing LaTeX documentation using ```\input{filename}```

jbqubit, 2018-01-26:
If you want the ability to do more atomic import of wiki topics restructure wiki by creating new pages. For example, break up [SaymaAFE](https://github.com/m-labs/sinara/wiki/SaymaAFE) into [TestMod](https://github.com/m-labs/sinara/wiki/TestMod), [BaseMod](https://github.com/m-labs/sinara/wiki/BaseMod), [MixMod](https://github.com/m-labs/sinara/wiki/MixMod).

filipswit, 2018-01-27:
I added [https://github.com/m-labs/sinara/wiki/uTCA](url) as a first pargraph, then this: [https://github.com/m-labs/sinara/wiki/Sayma](url) as an intro of a project, but in my opinion if we want to use wiki as a base of paragraph, the Configuartion section should be moved to another page.

jbqubit, 2018-01-29:
> Configuartion section should be moved to another page. 

Sounds good. Maybe call it sayma_configuration in wiki so its not confused with configuration of other sinara devices.

filipswit, 2018-01-30:
Ok. I've splited it. I hope I haven't done any mess.

jbqubit, 2018-01-31:
@filipswit I see you've created the following wiki pages. Looks good! 
- https://github.com/m-labs/sinara/wiki/Sayma_configuration 
- https://github.com/m-labs/sinara/wiki/Sayma-gateware
Please be sure to include links to the new pages from the main Sayma wiki page.

closed 2018-02-07 by jbqubit