sinara-hw/sinara #491: baseMod documentation

jbqubit, 2018-01-26:
baseMod is the-board-formerly-known-as-Allaki. 

### AFE Interface 

Discuss broader AFE interface concept. baseMod is one manifestation of an analog front end. 

### baseMod details
[Wiki](https://github.com/m-labs/sinara/wiki/SaymaAFE) gives a head start. 
- [ ] discussion of use case (frequency range, output impedance, etc)
- [ ] AFE interface (photo, names of pins) 
- [ ] power dissipated

### test results
- [ ] output chain phase noise
- [ ] switch isolation 
- [ ] attenuator isolation
- [ ] power supply rejection
- [ ] AM noise
- [ ] switch settling time
- [ ] attenuator settling time
- frequency-dependent plots (10MHz - 4GHz) 
    - [ ] output chain S parameters vs frequency
    - [ ] cross-talk vs frequency
    - [ ] IIP3 vs power and frequency (2D)

jbqubit, 2018-02-07:
@filipswit can @sbouhabib help you with this section of documentation?

filipswit, 2018-02-08:
Yes sure, that would be great.

jbqubit, 2018-02-14:
@sbouhabib @gkasprow @filipswit any word on this?

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/BaseMod#1

closed 2018-08-18 by gkasprow