sinara-hw/sinara #492: Zotino v1.1 errata

hartytp, 2018-01-29:
- [ ] +12V supply to TEC IDC should have a separate ground return trace and not return the current through the main PCB ground. There should also be an electrolytic decoupling cap between +12V and gnd near the IDC.
- [ ] Depending on what we decide to do with the Thermostat EEM, consider populating the IDC with a smaller one, which does not expose the I2C, +12V and +3V_mp
- [ ] The GND trace between the thermistor and the TEC should be connected to Zotino's ground through a 0R resistor that is DNPd by default. The idea here is that we want to try to avoid creating a ground loop between Zotino and Thermostat if we can, so this should only be grounded at one end, not both. In general, it's best to ground this kind of thing at the Thermostat end rather than at the load end.

a-shafir, 2018-01-29:
@hartytp I am not sure how it is possible to set the TEC without i2c with the constrains discussed above. It can be a CPU in the TEC regulator that will decide the target temperature but even in this case i2c interface to the host will help a lot.

Also in the view of low PPM what will be the target load of the board? There is 500r 1% serial to the output. So if the load will be even 1M 5% it is not so precise.

For the BOM. There are 0.1% resistors in the opamp filter BUT the same circuit uses 5% capacitors. I have not simulated it but it looks odd.

gkasprow, 2018-01-29:
The TEC should be connected to GND by capacitor and common mode choke.. Otherwise we will get ground bouncing caused by possible external GND loop which we cannot control right now.

hartytp, 2018-01-29:
> For the BOM. There are 0.1% resistors in the opamp filter BUT the same circuit uses 5% capacitors. I have not simulated it but it looks odd.

Why is that odd? 0.1% resistors are pretty standard at most assembly houses and are very cheap. 0.1% (or even 1%) capacitors are not.

For this circuit, 5% is fine (we don't need the pole frequency to be set more accurately than that). Remember that the load will often be 100pF or more due to cabling etc, which is non-negligible.

hartytp, 2018-01-29:
> @hartytp I am not sure how it is possible to set the TEC without i2c with the constrains discussed above. It can be a CPU in the TEC regulator that will decide the target temperature but even in this case i2c interface to the host will help a lot.

Current setup is fine. We have a microprocessor in the temperature regulator which can do all that (the control loop is all digital). a second I2C temperature sensor is not required.

> According the opamp specs it has short circuit protection and stable witc C load. So I think the output filter (if need) can be optimized to minimize R (50r or so, simulation need most likely). For the best DC precision better to remove the RC at all.

No! That's not a good idea. Lowering that too much increases the dynamic current when the output value changes, leading to nasty self-heating effects that will make the stability worse.

> Also in the view of low PPM what will be the target load of the board? There is 500r 1% serial to the output. So if the load will be even 1M 5% it is not so precise. Even the leakage of bav99 can't be ignored because it can be from uA to tens uA! bav199 has much better specs.

The load for this will generally be capacitive with >1MOhm of DC resistance. Remember that we don't care about the absolute accuracy of the voltage as much as its stability.

Those diodes were selected for very low leakage. Looking at the datasheet, even at 85C, the leakage is <1nA at for reverse bias between a few V and 10V (the range we're in) with very low temperature dependence. even 1nA*500R = 500nV. 1LSB=20V/2^16=300uV. So the leakage is completely negligible.

hartytp, 2018-01-29:
> The TEC should be connected to GND by capacitor and common mode choke.. Otherwise we will get ground bouncing caused by possible external GND loop which we cannot control right now.

I see, you want a CMC on the +12V and GND return line for the CMC. Seems sensible.

In any case, let's not worry about this for now as the +12V may not be required in the end if Thermostat ends up using PoE...

gkasprow, 2018-01-29:
I want them for TEC pins to cut common mode switching noise.

hartytp, 2018-01-29:
I see, you're worried about that coupling from the TEC to the ground plane?

FWIW, I think that the MaximIC peltier driver ICs are probably low enough noise that this isn't going to be a problem, but a cheap choke + capacitor is probably not a bad thing to have just in case.

a-shafir, 2018-01-29:
Right, there was BAV199 already. Sorry.

I2C need to communicate with the TEC regulator CPU. Otherwise I see no way to have it working correctly (predictable) depending on the airflow and ambient air conditions.

I think it makes sense to put the note or label about the 500r output impedance.

I understand your concern about the output currents.

I see the cost of 0.1% (purchasing reel) is 23c each and 1% around 0.2c. Both digikey. 5% indeed cost the same as 1% for most of the good brands but 0.1% is a very different story IMHO. For this cost you can put 1% capacitors easily.

For the TEC power filters it makes sense to put it on the load side (regulator extension PCB). In this case you can ensure it matches the module noises etc. Also if the customers will not need TEC in the mass production the cost will be saved.

gkasprow, 2018-02-22:
Zotino 1.1 arrived. TEC heatsink fits well.
[image]

hartytp, 2018-02-22:
Looks really nice! Can you try powering it on. If it doesn't die after 24 hours, can you ship me one for further testing?

hartytp, 2018-03-09:
Greg, just received this. Looks really lovely, well done! Particularly like the TEC mounting.

gkasprow, 2018-03-09:
There is missing thermal paste below the sink and the TEC. Since we will probably dismantle the stuck up I did not place it to not make hands dirty :)

hartytp, 2018-03-18:
Zotino V1.1 seems to work. Nothing feels hot, all else seems as it should be.

Will leave on over the weekend to see if it spontaneously dies again.

Will aim to upload an ARTIQ driver on Monday if anyone else wants to play with it.

Generally, I'm *very* happy with this PCB. Great job @gkasprow.

hartytp, 2018-03-19:
Drum roll...

So, the channel I'm looking at is still putting out voltages correctly after a weekend of operation, and nothing feels particularly warm to touch (although, I've just come in from the "mini beast" and my fingers aren't fully defrosted yet, so sensitivity may not be optimal).

Provisionally, this all seems to work. Phew!

gkasprow, 2018-03-21:
@hartytp On 4 remaining modules I installed TECs in a way that can be removed with use of a screwdriver. I also applied the thermal paste.

gkasprow, 2018-03-21:
to not expose the  I2C interface, I added two resistors that are DNP by default.

closed 2018-03-21 by gkasprow

hartytp, 2018-03-21:
Great! Thanks Greg!

gkasprow, 2018-03-21:
Moved release 1,2 to new, dedicated repository
https://github.com/sinara-foundation/Zotino