sinara-hw/sinara #494: Sayma v1.0 bug fixing meta issue

hartytp, 2018-01-29:
@sbourdeauducq et al. I'm trying to get back up to speed on the current situation wrt Sayma hardware bug fixing. Summarising the other issues, AFAICT, the situation is:
- [ ] **1V8 issues**:
  - @sbourdeauducq sent a failing board to WUT for @gkasprow to look at
  - @gkasprow is waiting for the board?
  - Temporary fix/hack seems to be to add a capacitor to the 1V8 rail on failing boards
- [ ] **Ethernet**:
  - @marmeladapk and @gkasprow  are trying to get ARTIQ up and running on Sayma at WUT. Once that's done they'll be able to investigate this faster.
- [ ] **HMC830**:
  - Still haven't got to the bottom of this. Investigation ongoing

Is that all the issues? IIRC, other than that Sayma and Urukul are basically up and running, right?

jbqubit, 2018-01-29:
Thanks @hartytp. Also for Sayma... 
- [ ] RTM bitstream loading https://github.com/m-labs/artiq/issues/813
- [ ] DDR3 timing errors https://github.com/m-labs/misoc/issues/75
- [ ] MiSoC memory test failed https://github.com/m-labs/artiq/issues/908 
- [ ] Sayma serving as ARTIQ Master, communication via DRTIO

Small bits...
- [ ] serwb working at 625 Mbps, unreliable at 1.25Gbps https://github.com/m-labs/artiq/issues/856
- [ ] occasional ```Unknown flash device (ID 0x00000000)``` #444
- [ ] programming MMC with artiq_flash https://github.com/m-labs/sinara/issues/448

hartytp, 2018-01-30:
@jbqubit Let's keep this limited to issues which have a hardware component, rather than being a gigantic Sayma hardware/software/firmware/gateware catch all. Gateware/firmware/software only issues (like DRTIO) are being tackled by M-Labs alone, without contributions from WUT etc, and I have no reason to want to interfere with their project management of those issues.

hartytp, 2018-01-30:
Also, AFAICT, the flash issues are now well-understood and have known fixes. So this isn't really an outstanding issue anymore. Are you still seeing symptoms of this that aren't fixed by the jumpers Greg recommended?

sbourdeauducq, 2018-01-30:
> Since we can use an external 1.2GHz clock, this isn't crucial for getting Sayma up and running.

@hartytp 
It is still an important problem, when the board is in the µTCA crate, the clock bypass is not easily accessible. And we want to sort this out before the next hardware revision. Have you made any progress?

hartytp, 2018-01-30:
@sbourdeauducq ack. Will update on progress in the other issue.

To check though: are there any other hardware issues that you're aware of that I've missed?

sbourdeauducq, 2018-01-30:
There is, at least: intermittent JTAG breakage not caused by the 1.8V bug (https://github.com/m-labs/sinara/issues/463), problems when using a µTCA crate (https://github.com/m-labs/sinara/issues/475) and I just noticed that sometimes some RTM power rails also fail (I have not opened a separate issue yet).

jordens, 2018-01-30:
https://github.com/m-labs/sinara/issues/495 is a risk as it breaks overtemperature shutdown of the FPGAs.

hartytp, 2018-01-30:
Things @jordens reminded us to make sure to check with the v1.0 hardware in #408:
- [ ] openMMC
- [x] SCANSTA
  - This now seems fairly well tested and debugged. But, let me know if there are any other test you want done.
- [ ] RF backplane operation
- ~high density full scale operation (at least 5 Sayma in a crate)~
  - Let's leave this until the next design revision since I don't think we have enough Sayma to do this atm and there isn't much point manufacturing more Sayma without applying the current bug fixes.
- ~the current ADC (if kept)~: AFAICT this ADC will not be kept in v2.0
- ~DRTIO over the RTM-AMC link~: hopefully just a firmware/gateware issue.

hartytp, 2018-01-30:
Also to test:
- [ ] Check that we can clock the HMC830 from the Si5324 and measure the phase noise when doing this. Ideally do this using Sayma and Metlino both in the same rack.

jbqubit, 2018-01-31:
@hartytp said
> ~RF backplane operation~ let's just drop this instead!

Agreed. 

@hartytp 
> - ~high density full scale operation (at least 5 Sayma in a crate)~

A demo of phase alignment of RF output will consist of 2 Sayma (AMC+RTM) running SAWG and 1 Sayma_AMC serving as ARTIQ master. This many power hungry boards really ought to be run in a crate. Plus with so many boards there's risk of one of the white wires getting bumped.  

@sbourdeauducq I'm using 90-deg connectors that are low enough profile to work inside rack.
https://github.com/m-labs/sinara/wiki/Sayma_configuration#debugging

@hartytp said
> @jbqubit Let's keep this limited to issues which have a hardware component
> interfere with their project management of those issues

Agreed that focus of present Issue Sayma 1.0 hardware is fine. I don't see that having Issues in the Sinara repository track those in the ARTIQ repository is interference. I'll create a new Sinara Issue for M-Labs deliverables for Sayma v1.0.

jbqubit, 2018-01-31:
@hartytp I pulled items discussed into list at top of Issue.

sbourdeauducq, 2018-02-01:
> A demo of phase alignment of RF output will consist of 2 Sayma (AMC+RTM) running SAWG and 1 Sayma_AMC serving as ARTIQ master.

Sayma is only supported with the RTM connected. There would be local SAWG channels too.

jbqubit, 2018-02-02:
So this demo could be done with just two Sayma (AMC+RTM) then? -Joe

sbourdeauducq, 2018-02-02:
Yes.

hartytp, 2018-05-24:
Provisionally, everything on this list seems to be fixed. @enjoy-digital reports that ser-wb works on my board after fixing the HMC830 for reasons unknown.

So, closing this issue for now.

closed 2018-05-24 by hartytp